WouterJD/FortiusANT #238: Is calibration needed?

kwouters1, 2021-02-03:
I've been using FortiusANT for a while now.
Whenever I ride indoors I just start the Windows executable without any extra parameters.
So I get the GUI, and then I just start to ride in Zwift, RGT, Cadesport etc.

But I've been wondering how accurate the transmitted power is.
And if I needed to calibrate something on forehand.
Is there a way to know? I'm using and Tacx Flow with the T1932 headset.

WouterJD, 2021-02-04:
Please refer to #239

closed 2021-02-04 by WouterJD

switchabl, 2021-02-04:
I think #239 is related to motor brakes, while this is about the magnetic brake? #234 is probably more relevant but may not be very helpful too helpful yet.

There is no good solution for calibration of the magnetic brake at the moment.
- You can use the "runoff" button to make sure you are at least in the right ballpark (or at least keep it consistent). The spin-down time should be around 7s. Adjust the knob on the trainer accordingly.
- If you have a power meter you can adjust the calibration value (-c setting) until the power matches.

Do not expect the power readings to be accurate unless you can check with a power meter. I am working on a proper calibration algorithm to improve the situation.