WouterJD/FortiusANT #239: Calibration of T1941 motorbrake

rutteger, 2021-02-03:
Having read the fortiusant manual and online manual for the Tacx fortius not 100% clear on how to calibrate and set the tension on the motorbike
-Assume runoff is for magnetic trainer only
-For the motorbike when starting fortiusant simply kick the pedals will start the cal routine.
However how to know tension is correct on the brake itself as set by the adjuster dial? The manual which came with the trainer gives no indication.
Found a Tacx manual [here](https://www.qbp.com/diagrams/tech_content/pdf/tacx/general_settings.pdf)  > which states
_Calibration Fortius motor brake
After pressing the button Calibrate you will see a new screen. Click on Start calibration and make 1 pedal stroke to make the wheel turn; the brake will automatically calculate the correct calibration value. Do not cycle or brake while calibrating. When you're finished the calibration value will appear in the top left hand corner of the screen. This value should be comprised between 10 and 15. Click on Reset if it did not work out alright and start all over._
But not sure how to ensure this in fortiusant? Does this align to the resistance number somehow?

WouterJD, 2021-02-04:
HI @rutteger 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Thanks for this background information.

It would be great if you would calibrate your Fortius with the Tacx software first (then you know the value) and then repeat with FortiusAnt. This could give an indication what would be a norm.

During calibration, the brake runs at a steady pace and measures how much power it takes to run the wheel.
In the session after, this baseline-power is subtracted from the required power and therefore it equalizes.

If your knob is "tight" then the base-resistance will be high
If your knob is "loose" then the base-resistance will be low

In either case (of course within a reasonable range) loose/tight will be neutralized by the calibration

anyway - I think it's being designed like this.

rutteger, 2021-02-09:
Managed to install the Tacx software and run the calibration. The version installed had a simple red - green - red cart giving a bound of acceptable tension for the motorbrake. 
Adjusted in the Tacx software (turns out I had the brake too tight) then ran the calibration in FortiusAnt. Baseline power / resistance from the command line was around 1100. This has been fairly consistent over a few sessions now.

One interesting point which may have already been considered is the calibration value after a session, tried after a couple of hour long sessions and this seems to be lower at around 900. Suppose some drift is expected. Noted it was mentioned elsewhere but might a more aggressive calibration routine (running the brake quicker) help with this drift?

jujroy, 2021-02-10:
@WouterJD   Hello everyone,  I also have a T1941 Motor Brake and I dont have a working version of TTS.  After reading above and the FORTIUSANT Manual, I am not too sure if I am calibrating my system correctly.  I do NOT normally use the runoff test but I follow all other steps in the book.  Is this sufficent or should I also perform the runoff test having a Motor Brake system ?  Thanks 

PS: I should also mention that I have, on a few occasions, used the runoff test but to achieve the 7 seconds, I need to remove too much tension on the roller thus the tire slips on hills steeper than 5%.  4 seconds seems adaquate for steeper hills.  Cheers

switchabl, 2021-02-10:
@jujroy The tension setting is not nearly as critical with a Fortius (compared to a Flow) because the motor brake has a very large resistance range and can compensate easily as long as you run calibration. As long as 1) it isn't so loose the tyre doesn't slip and 2) it isn't so tight that the tyre gets very hot. You can try to target a calibration value of 1100 like @rutteger has found, if TTS suggests that it is probably reasonable. I don't trust the runoff test much, I also have never gotten 7s on my Fortius. The calibration value should be a much more accurate indicator.

@rutteger That's useful to know. I think we eventually want some scale like TTS has too instead of just displaying (seemingly arbitrary) numbers.

The drift is most likely due to the tyre warming up. This is a problem with all wheel-on trainers. For best accuracy you will usually have to warm up for several minutes before running the calibration. Changing the calibration speed won't help with that. The calibration speed should be close to the typical riding speed for best accuracy. Running it for longer might though. One idea could be to have an optional auto warm-up feature where the wheel spins for 5 minutes. (This has been requested for the Genius before, but unfortunately it is much harder to implement there.)

rutteger, 2021-02-10:
@switchabl the trainer auto warm up prior to calibration was what I was suggesting in a round about way. Think this would be a good idea.

jujroy, 2021-02-10:
@WouterJD @switchabl   Thanks for your answers.  Will try in the next few days.  Enjoying my skis these days.  Cheers

WouterJD, 2021-02-17:
Great, since it's silent I assume issue can be closed. If not please reopen.

closed 2021-02-17 by WouterJD