WouterJD/FortiusANT #240: C++ error?

Ralphat51154, 2021-02-04:
Thanks for your work on this project.  I  have been using the original Antifier code with success for many years.  Here is the message I get when I try to run your windows executable in windows 10.  Thanks in advance for any help you can provide.

Traceback (most recent call last):
  File "multiprocessing\process.py", line 315, in _bootstrap
  File "multiprocessing\process.py", line 108, in run
  File "FortiusAnt.py", line 496, in FortiusAntChild
  File "FortiusAnt.py", line 258, in __init__
  File "FortiusAntGui.py", line 651, in __init__
wx._core.wxAssertionError: C++ assertion ""image.IsOk()"" failed at ..\..\src\msw\bitmap.cpp(841) in wxBitmap::CreateFromImage(): invalid image

I have C++ 2013 and 2015-2019(x64) redistributables running on my PC

WouterJD, 2021-02-04:
Hi @Ralphat51154 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Sorry - sloppy work from my side.
Should be solved by current upload.

Ralphat51154, 2021-02-04:
I have a 1932 head unit and an older T1911 brake, works well enough with
antifier.

Thanks!

closed 2021-02-22 by WouterJD