WouterJD/FortiusANT #241: Vortex is connected but gauges do not show any activity

JoostBart, 2021-02-04:
Hello, 
I'm highly appreciating this piece of software, enabling me to keep my Tacx Vortex 1960 in use. Great job.
It took me a couple of hours to install Python on my Macbook and to figure out how the FortiusANt should be installed. Finally I got it showing the startup screen en coupling it to my trainer. I felt like a true software engineer....
However, the disappointing news is that the gauges in the startup screen do not show anything, when I pedal on my bike. No speed, power or cadence is being showed. only my HRM does show my heartrythm. 
Is there anyone in this community who can provide me some tips how to proceed in coupling my trainer to the computer transferring the data?

WouterJD, 2021-02-17:
Hi @JoostBart 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Sorry for answering earlier; quite busy these days.

Could you provide some more info
For example, run with -d127 flag and attach logfile to this issue

WouterJD, 2021-02-22:
No feedback received; can be reopened when provided

JoostBart, 2021-02-22:
Hello, 
After a couple of times of trial and error and a new ANT+ dongle, I was able to connect and to receive signals back. 
Eventually Zwift and Rouvy were functioning. 
However, after a second trial, It seemed to be good luck or bad luck. In other words I received signals back or not. 
When I was receiving signals to the gauges, Zwift was also functioning. However, After some 10 min. or so, the values were stuck and did not change anymore, even when I completely stopped pedaling. 
So there seems to be some kind of dissynchronisation or discontinuation of the signals from  Vortex back to Fortius ANt. 

I can provide you with a log file in which no signals at all are provided: 

MacBook-Pro-3:pythoncode jbart$ python3 FortiusAnt.py -a -g -H0 -t Vortex -d127
21:15:15,741: FortiusANT started
21:15:15,742: -a
21:15:15,742: -c False
21:15:15,742: -C 0/0
21:15:15,742: -d 127 (0b1111111)
21:15:15,742: -D None
21:15:15,742: -g
21:15:15,742: -H 0
21:15:15,742: -p 1.0
21:15:15,742: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
21:15:15,742: -t Vortex
21:15:15,742: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
21:15:15,742: ------------------
21:15:16,401: FortiusAnt GUI started in child-process
21:15:16,900: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
21:15:16,906: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
21:15:18,560: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
21:15:18,713: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
21:15:18,894: FortiusANT receives data from an ANT Tacx i-Vortex (VTX Controller), any device
21:15:19,073: FortiusANT receives data from an ANT Tacx i-Vortex Headunit (VHU Controller), any device
21:15:19,275: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
21:15:19,430: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
21:15:19,598: FortiusANT receives commands from an ANT+ Generic Remote Control
21:15:19,800: ---------- ANT-devices are activated ----------
21:20:47,743: ---------- ANT-devices are deactivated ----------
21:20:48,356: FortiusAnt GUI ended
21:20:48,441: FortiusAnt ended

WouterJD, 2021-02-22:
Hi Bart

Steps to debug are:
- Test FortiusAntGui.py just to see whether the interface works
- Use FortiusAnt with -m and -s flag in simulation mode; to check the program
- Use FortiusAnt with -m flag and no ANT dongle.

If you're convinced that the software works properly
- Run FortiusAnt with -d127 flag and check logfile what data is being received/transmitted

WouterJD, 2021-03-17:
No additional info received

closed 2021-03-17 by WouterJD