WouterJD/FortiusANT #243: Fortius Brake Stutters

leovanr1, 2021-02-06:
Dear all,
I am new to FortiusANT as well as Github and certainly no programmer. I have set-up FortiusANT on my laptop with for now 1 ANT+ dongle ('hline'). It makes the connection and I can ride in manual mode, but ever few paddle rounds the brake goes very heave for a moment. When I give it another push, it goes smooth for a few (max. 5) strokes and then it brakes again. Can anyone help me how to solve that?
Thanks already,
Leo
Nijmegen, The Netherlands

marcoveeneman, 2021-02-06:
Hi Leo, i suspect it is an issue with the cadence sensor, is it connected properly? The motor brake needs to register the cadence or it will brake hard every few rounds, as you describe.

leovanr1, 2021-02-06:
Thanks. I didn't install the cadence sensor yet, so that might well be the problem. , I will try that! Thanks again!

leovanr1, 2021-02-06:
I stiil have a problem, because I have the sensor but seem to be missing the magnet.

switchabl, 2021-02-06:
@leovanr1 Have a look at #213. Unfortunately, the cadence sensor is required (and there is nothing that we can do about it, this is all handled internally by the Powerback unit). But any sufficiently strong magnet that can be attached to the crank should work (look for "cadence magnet"). In a pinch, even a fridge magnet and some tape might be enough to get you going.

Mickyduck55, 2021-02-06:
Slightly off topic... I know of a man who has a useless fortius just because his cadence sensor wire was broken by catching it on a pedal,,,   he says he cannot find a wired sensor anywhere.. be careful with that innocuous bit of Hardware

leovanr1, 2021-02-07:
Thanks very much. It works like a charm with a small magnet that I put on the paddle bolt. So I might even say it works better than with the original magnet, because I don’t even need tape or whatever.

closed 2021-02-07 by leovanr1