WouterJD/FortiusANT #244: Idea: macOS easy install

marcoveeneman, 2021-02-06:
I see more and more people start to use FortiusANT, also on macOS, which is great! However, with that, also more and more questions and issues arise regarding the installation. I am thinking to create an all-in-one package for FortiusANT to run on macOS, such as FortiusANT.exe for Windows users.

I'm thinking to generate a .dmg file which allows FortiusAnt to be easily installed by double clicking the file and dragging the application to the application folder. This is common installation behaviour on macOS afaik.

Would anyone be interested in this before i spend more time and effort on it? Are there special things we need to take into account? Please share your ideas.

goxmail, 2021-02-09:
Most definitely interested, although not quite sure what would need to be taken into account... ;)

JoostBart, 2021-02-18:
Great idea. It took me a couple of hours to read (and understand) all manuals before I could install Python, nump, brew and so on. And now I hav to startup from my terminal with a couple of manual commands. (And although the software is running and connecting to my Vortex, I still lack the signal from my Tacx coming back to my Mac. )

Aduffy, 2021-03-17:
@marcoveeneman That would be great ! I would be more than happy to test and give feedback .. Installing this manually on mac is defiantly a daunting task..

WouterJD, 2021-06-07:
@all; 

In the windows-environment, the installation is some work, but installing python + dependencies is not the most difficult part.
The most critical task is the USB-part; especially (in many cases) uninstallation of Tacx-usb.

eskimofisherman, 2023-09-28:
Following up on this thread as I would prefer to install an a mac.  
Im in NY, and have a TacxFortius with a T1932 head unit.  I had TTS4.x running on a windows 10 machine perfectly fine, but the TTs install has corrputed with the "migrating videos" error. 
Hence, I figured i might aswell update to the online platfoms and use FortiusAnt as my "interface".  
I have FortiusAnt working on my windows machine fine, but it doesnt natively support BT, and its an older windows machine.  (I could order a USB-BT dongle, but i'd prefer to run this on the mac).
I followed the instructions in the manual but I think there are some conflicts with newer versions of some of the tools that are giving me errors when trying to install on mac. (for example npm install ends with a not ok status).   

Was a dmg file ever developed, and/or a way to get this working using the referenced versions in the manual.

WouterJD, 2023-09-28:
Hi @eskimofisherman 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Great to read that you have the stuff basically working, now going for the last steps.
1. I do not have a Mac and hence cannot do much testing, development and/or improving here.
2. BLE is a difficult subject, being tackled by Marco using NPM but it's always difficult to get that package running. It was the only solution at the time, but an enormous "sub package" for the job.
3. BLE is implemented using libraries but requires a BLE/5 dongle (windows) or native support (raspberry, presumably MacOS) The libraries are not always 100% stable.
4. ANT remains the easiest interface on all platforms but requires an ANT dongle.

Answer to your question whether the packages wqas ever developed: no (sorry for that).
The choice is yours how to move on; I would suggest to buy a BLE/5 dongle and use the windows machine with -bb so you do not need NPM.

closed 2023-12-15 by WouterJD