WouterJD/FortiusANT #245: Problem with calibration T1902

pawskocz, 2021-02-07:
Hello. I have problem with power calibration:( My hardware: Tacx i-flow brake T1902 with 1932 usb head unit. Power meter - powerTap hub sl+. Few days ago I obtain accuracy around 10 % (1 houre training in rouvy). Now diffrence between power meter and tacx is 50 W (power meter show less watt) ! :( So I installed TTS 4.0 and I calibrated tacx. In workout TTS (when slope was -1 tts shows more watts in other hand slope +4 TTS show less watt compare to power meter). I start Fortious with batch file name Rouvy with additional command line (-c6 and -p80) but i still have wrong readings:( What Can i check ?THX !

switchabl, 2021-02-07:
@pawskocz The calibration in TTS is completely independent and should not affect FortiusANT in any way. We don't have automatic calibration yet, it is something I am working on right now. So the way to calibrate manually right now is mainly the -c parameter (I would see how far you get with that before changing -p) _or_ you can leave the default and adjust the knob on the trainer until the power reading matches.

Note that there are multiple factors that can affect calibration:
- tyre temperature: power will drift somewhat at the beginning of the workout as the tyre warms up
- tyre pressure
- knob on the trainer (naturally): if you change the roller pressure, naturally the rolling resistance changes

I usually warm-up for a couple of minutes before doing calibration on any trainer. And check the tyre pressure before every ride to avoid having to re-calibrate.

Another thought: are you using the settings dialog? If you change something there, then the command-line options are ignored. You should use one or the other (most people will probably be better off with the settings dialog).

pawskocz, 2021-02-07:
THX a lot. I will change -c parameter for a start and check tire pressure (7 bar). I use only command line to change parameters. I dont know what is change, 2 days ago i had 10% accuracy:(

WouterJD, 2021-02-11:
Well, for what it's worth.
The 1902 trainer is from the before-power-meter era and no user knew whether that would be constant yes or no.
Today we measure the trainers with modern power-meters and probably detect that these trainers are different one day from the other.

It's like riding outside, same person, same road, same bike. But temperature and personal condition changes....

switchabl, 2021-02-11:
@WouterJD Well, yes, the power accuracy was never great in TTS either and most people never knew but that doesn't mean we can't do better. They sell a Flow Smart now which is essentially the same brake with built-in Bluetooth and claim <5% accuracy for that. It is mostly in the software/firmware. The Vortex is more or less the same, too, but with 2 additional magnets and it matches my Garmin Vector pedals almost perfectly with the built-in spin-down calibration.

The T1901 will never be as good as a NEO: it has limited resistance range, it is noisy and will not have <2% accuracy. But there is no technical reason why it shouldn't be almost as good as new Flow Smart (except for the cables). Unfortunately this will need some more careful work, so don't expect a PR next week. But I have a relatively clear idea now how this works and I am confident we will get there.

pawskocz, 2021-02-11:
Little success ! :) I have accuracy around 5 % :) Tyre pressure 8 bar and parametr -c4

WouterJD, 2021-02-17:
Cool, I'm looking forward to it.

WouterJD, 2021-02-22:
I think we have to live with it as is; let's close this issue

closed 2021-02-22 by WouterJD