WouterJD/FortiusANT #246: Problem connecting Fortius to FortiusAnt

BarryO1, 2021-02-08:
Hello,

I have a Tacx Fortius (without steering) with T1932 display and a laptop with Windows10 64 bit on it. Yesterday I downloaded and installed FortiusAnt following the FortiusAnt manual. To start up FortiusAnt, I use the Windows executable and I installed the Windows prepacked drivers. 

I started FortiusAnt without additional command-line parameters and the FortiusAnt GUI appeared. However, although my Fortius was connected to the computer by the Tacx USB device, FortiusAnt did not recognize my Fortius. I got the message 'No Tacx trainer found'. What could be the problem?

WolfRevo, 2021-02-08:
Can you please check in the device manager (start it before you connect your trainer) which driver the trainer get's assigned too? Do you have installed a TACX trainer? 

Refering to the manual chapter 3.5.2: "On the system where FortiusAnt is running, the easiest is that TTS is not installed, since the two programs require different USB-drivers which may be conflicting."
I for instance installed the new drivers in parallel to TTS. For that I needed to manually change the assigned driver in the device manager (install option 1 in chapter 3.5.2) for the specific USB port I want to use with FortiusANT. Otherwise connecting the trainer will choose the TACX driver which won't work with FortiusANT.

WouterJD, 2021-02-08:
@BarryO1 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

and yes @WolfRevo I agree

BarryO1, 2021-02-08:
At my system, TTS4 is also installed. I did not realize that the USB port I use to connect the Fortius for using TTS4, can not be the same as for using FortiusAnt. I will follow the steps as described in 3.5.2. and will keep you posted.

BarryO1, 2021-02-09:
In Device manager, I found my Fortius under 'Jungo', because TTS4 has also been installed. I do not want to uninstall the Jungo driver because I want to continue using TTS4. I use TTS4 by connecting my Fortius with USB port 1. 

Now, what is the best way to do without losing the use of TTS4? I can imagine that I connect my Fortius with USB port 2 or port 3 and then I install the new drivers in parallel to TTS4 by following install option 1 in chapter 3.5.2. Is that the best way to use FortiusAnt (and Zwift) parallel to TTS4?

switchabl, 2021-02-09:
@BarryO1 You can try using the libusb filter driver. Ideally, this will let TTS and FortiusANT co-exist seamlessly (but it is a little less straightforward and there is a chance it can cause problems).

- reinstall the TTS4 Jungo driver (just to be sure): run `TacxDriversSetup.exe` from `C:\Program Files (x86)\Tacx\TacxTrainersoftware4\TacxDrivers\`
- download the libusbK package https://sourceforge.net/projects/libusbk/files/libusbK-release/3.0.7.0/libusbK-3.0.7.0-bin.7z
- extract `libusbK-3.0.7.0-bin.7z`
- (make sure the T1932 is plugged in)
- run `bin\exe\amd64\install-filter-win.exe` as administrator
- choose "Install a device driver"
- find the T1932 interface (probably "VR-Interface")
- click install

WolfRevo, 2021-02-10:
I have a little bit to disagree @switchabl. I have two laptops (one with TTS3 and one with TTS4) where I use TTS and FortiusANT seamlessly. As TTS does not like the libusb filter (TTS does not recognize the trainier if the device manager assignes the libusb driver to the trainer) the easiest way was to do it as @BarryO1 mentioned.
So my way:

- install TTS4 Jungo driver as @switchabl asked to do so and check that the trainier is connected to the USB port you want to use with TTS
- switch the trainer to the USB port you want to use with FortiusANT, the device manager should automatically assign the Jungo driver by default
- now manually choose the libusb filter driver using option 1 in chapter 3.5.2 of the manual so that the driver is now assigned to the trainer

For me this works perfect on both laptops as Windows remembers which driver to choose for which USB port. So I don't have to fiddle with the drivers anymore. I just marked the USB ports so it is easy to see where to put in the trainer for which use case ;)

switchabl, 2021-02-10:
@WolfRevo Interesting. I have it working fine on my development PC where I occasionally switch to TTS4 for testing (haven't tried TTS3 though). It sounds like the driver installation may have gone wrong for you somehow. For me it still shows up as Jungo in the device manager after installing the filter (unlike with the normal driver package which moves it to libusb devices). It is only when you go to driver details that you can see that "libusb0.sys" has been added.

But it is true that the filter driver is not officially recommended by the libusb-win32 developers and there may be issues with it. Your option seems like a good alternative if it doesn't work.

WouterJD, 2021-02-17:
Since no more communication on this issue, I assume resolved.
Keep going all!

closed 2021-02-17 by WouterJD

graafro, 2022-02-24:
Hi Wouter,

I bought another AND+ adapter from XAND (futurumshop.nl) and still get the message “no tacx trainer found”
Is there a way to find out where things are going wrong?
[image]

WouterJD, 2022-02-24:
This issue is closed
The TACX trainer should be connected using the USBcable

Please raise separate issue if you have issues