WouterJD/FortiusANT #248: Newly encountered problem with connection

bfastmike, 2021-02-08:
Hi, such a great piece of s/w which let me enjoy a brief period on Zwift - however for some reason the trainer is now longer connecting - hoping you can help?

I had everything set up with the libusb 32 driver with a Tacx Fortius T1932 head. Now though when I connect it the FortiusAnt software can't see it. When I check in devices it is showing as a Betronic LCI21XX USB under the Other category?

I tried replacing the driver once again and changing using the inf-wizard with no joy. I also tried a different laptop using libusb_32_T19xx.exe installer and new download of FortiusAnt. No joy.

ps -I'm in Belfast, Northern Ireland training in my freezing cold garage

WolfRevo, 2021-02-09:
Instead of the Inf-Wizard version please try the manual way:

Manually change the assigned driver in the device manager (**install option 1 in chapter 3.5.2**) for the specific USB port you want to use with FortiusANT. Otherwise connecting the trainer will choose the TACX driver which won't work with FortiusANT.

switchabl, 2021-02-09:
Before trying to replace the driver manually, can you
- right-click the Betronic device in the device manager
- click Properties in the menu
- click Driver Details in the Driver tab
- take a screenshot of that
- and in the Details tab choose "Hardware Ids" and take a screenshot of that as well?

I haven't seen "Betronic LCI21XX USB" before and I'm curious what exactly is going on. Normally the driver from the installer package should take precedence.

bfastmike, 2021-02-10:
The weirdest thing. Left it a few days and it's working again??

I'd left the trainer plugged in overnight. That might have caused the issue?

The same issue was happening on two windows 10 laptops so seems to be the trainer. Strange.

bfastmike, 2021-02-10:
Happy to gather info to help investigate

switchabl, 2021-02-11:
@bfastmike It's really hard to tell unless it happens again. The fact that it affected two PCs may point to a hardware issue. Maybe a pin was shorted out and the head-unit went into a bootloader mode (just speculating). Did you get sweat on it? The Tacx head-units don't really like that (which is somewhat ridiculous considering they are meant to go on the handlebars).

I also wouldn't recommend leaving the trainer plugged in (to the mains) when you are not using it. The "powerback" unit is not the most robust piece of electronics ever designed...

oldgearian, 2021-02-11:
Mike
I suspect the answer lies in your very last point of your ‘freezing cold garage’. I am training in a freezing cold cabin in the garden and it took me about a week, 2 new connection cables (usb and motor) 3 or 4 uninstalls and re-installs of fortius software etc before I realised that the T1932 headset just doesn’t want to work after a night in sub zero temperatures.
I now treat it with the respect it deserves and bring it inside at the end of each workout to warm up, along with laptops and any other electrical equipment which doesn’t like very cold weather.

Wouter - like many others I would like to thank you for all your efforts, and those others helping you, in resurrecting old Tacx trainers. I have very limited IT skills but having managed to get a Tacx I-Magic working with the Fortius software, (care of Lunicus.Org who are also putting in lots of fantastic work), I am now riding on Zwift, cadence and speed sensor only so far, but am about to order two ant+ dongles and try your software.

Keep up the good work.

I will try and put myself on your fortiusant map.

Ian of Addlestone, Surrey, England

WouterJD, 2021-02-17:
Added to manual and closed

closed 2021-02-17 by WouterJD