WouterJD/FortiusANT #249: CTP would a cheaper ANT dongle work?

leovanr1, 2021-02-08:
Currently I have 1 hLine ANT dongle which communicates with FortiusANT perferctly. First I tried anothe advertised as specific for CTP like Zwift, which was much cheaper. However, it turned out to be a Cycplus dongle that didn't work with FortiusANT, which I therefore returned. Now I am considering buying an ANT dongle for a CTP (specifically RGT). Could it work to use such a cheaper ANTdongle  for the CTP  and exclusively using the hLine for FortiusANT?

switchabl, 2021-02-09:
Maybe. There have been plenty of reports of people having problems with Zwift and other CTP too. It is not simply that FortiusANT doesn't have support for these dongles, the problem is they are really buggy. I guess it depends on what features RGT uses and how much of an effort they have made to work around the issues (maybe also on what revision of the dongle you end up with, it seems that they partially fixed something at one point but not properly).

If you want to save money you could also look at some older (used) models. I have picked up older Garmin ones for less than the CYCPLUS (see also this post https://github.com/WouterJD/FortiusANT/issues/61#issuecomment-770216291).

leovanr1, 2021-02-09:
OK, thank you.

closed 2021-02-09 by leovanr1