WouterJD/FortiusANT #250: Does someone have a pinout of mini-DIP port on T1932 head module?

john-mad-biker, 2021-02-08:
Hi all. T1932 module has a 8-pin mini-DIP port. I know only one case of using this port: connecting of T1905 steering frame to head module. But AFAIK there's only 2 wires soldered to potentiometer in T1905, so remaining 6 pins of the port seems to be unused. Can someone tell:
1. What pins of the port are used by T1905? (I can disassemble T1905 and inspect this by myself, but at first want to go the more simple way and ask community for this info).
2. Are there any other scenarios of using this 8-pin port except connecting T1905?
3. What is pinout of mini-DIP port? (What every pin is responsible for?)

I have an idea to make external buttons (which will duplicate buttons of T1932) and place them somewhere near shifting levers to use FortiusANT electronic shifting in the same way as the traditional gear shifting (with palms placed on the hoods or drops). Of course I can drill the T1932 case for route cables from buttons into the T1932 module, but using of existing unused port is much more elegant idea.

Thanks for all the info you can share! And I promise to write a how-to manual if I can realize this idea.

switchabl, 2021-02-09:
There is some info on the steering port in the @totalreverse wiki: https://github.com/totalreverse/ttyT1941/wiki (in case you haven't seen already). The interface is a gameport-style ADC, a potentiometer is connected between the 5V pin and the input pin.

If Tacx ever planned to release other add-ons for the Mini-DIN port, it never happened. The USB protocol has fields for four axes (bytes 16-23), not just one. The hardware implementation appears to vary between head-units:
- T1942: all four values are actually sent
pin 2: +5V
pin 3: axis 2 input (steering)
pin 7: axis 4 input
pin 8: axis 3 input
I'm not sure if the other pins are connected to anything. Axis 1 doesn't seem to be available on the connector (?), maybe an internal reference resistor.
- T1932: only the 2 bytes for the second axis (steering) are non-zero, the other ones do not seem to be available

I don't have a T1905, but from what I understand it is a 100kOhm potentiometer connected between pins 2 and 3.
(see here for pin numbering: https://datasheet.octopart.com/1658998-1-TE-Connectivity-datasheet-14744456.pdf)

So your plan could work but it looks like you don't have any unused channels available on the T1932 so you could not use the steering unit at the same time. You could just switch different resistor values in for different buttons (that used to be a rather common technique).

btw I use an ANT+ remote for shifting from the hoods: https://www.o-synce-shop.de/media/image/63/e7/19/IMG_6089xU9UKYi3toJiR.jpg

john-mad-biker, 2021-02-10:
@switchabl Thanks for all the info!!! My idea was way more simple - I just wanna solder wires from buttons to the unused pins inside T1932, then solder external buttons to 8-pin mini-DIP male connector and connect it to the port (of course instead of steering unit connector).

But your idea of forming data by resistors with further software processing (it needs to modify FortiusANT code as I understand) is pretty cool! Maybe I will try to realize it in future.

Yep ANT+ remote is the easiest way, but I don't have it at the moment =)

WouterJD, 2021-02-22:
I suggest to reopen the issue when more information is exchanged here

closed 2021-02-22 by WouterJD