WouterJD/FortiusANT #251: Bluetooth problem on Windows 10

pawskocz, 2021-02-10:
Hello! I try to run bluetoth on windows 10 :) What i done? (without any erros)
a) Installed python 3.9.1
b) upgrade pip
c) pip install -r requirements.txt
d) instal git
e) install nodeJS with tools
f) replacing drivers with zdiag (I have CSR 4.0 0A12x001)
g) npm install 
If I run Fortious ant - without python -m http.server 9999 i see error 
.. requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000027DF0948F70>: Failed to establish a new connection: [WinError 10061]
when I open python server:
C:\fortiusant\node\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:136
    this._aclDataInEndpoint.on('data', this.onAclDataInEndpointData.bind(this));
                            ^

TypeError: Cannot read property 'on' of undefined
    at BluetoothHciSocket.start (C:\fortiusant\node\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:136:29)
    at Hci.init (C:\fortiusant\node\node_modules\bleno\lib\hci-socket\hci.js:111:18)
    at BlenoBindings.init (C:\fortiusant\node\node_modules\bleno\lib\hci-socket\bindings.js:97:13)
    at Bleno.<anonymous> (C:\fortiusant\node\node_modules\bleno\lib\bleno.js:49:22)
    at Bleno.emit (node:events:378:20)
    at _addListener (node:events:421:14)
    at Bleno.addListener (node:events:469:10)
    at new VirtualTrainer (C:\fortiusant\node\virtual-trainer.js:21:11)
    at Object.<anonymous> (C:\fortiusant\node\server.js:8:15)
    at Module._compile (node:internal/modules/cjs/loader:1108:14)

and in python window:
code 501, message Unsupported method ('POST')
::1 - - [10/Feb/2021 17:00:09] "POST /ant HTTP/1.1" 501 -

HELP :D :)

pawskocz, 2021-02-10:
I install new instance of Windows 10 (Vmware) and I have similar issue :(

switchabl, 2021-02-11:
Seems to be the same problem as #253 and https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-777077778

@marcoveeneman I still haven't really looked at the BLE part in detail, but looks like maybe some version incompatibility between bluetooth-hci-socket and the USB library?

pawskocz, 2021-02-12:
Hi! It is possible to move bluetooth funcionality to WindowsExecutable file?:)

WouterJD, 2021-02-22:
I have similar issues, probably not exactly equal.

Solution: start everything up in a carefull sequence and then it works.
But if stopped and restarted this kind of errors.

I will document if I find a solution

closed 2021-03-19 by WouterJD