WouterJD/FortiusANT #253: HTTPConnectionPool(host='localhost', port=9999

harrykl, 2021-02-11:
Hello,
if I start FortiusAnt on my MAC I become the Error
.. requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa09bdb5d00>: Failed to establish a new connection: [Errno 61] Connection refused'))

I did not found anything that resolves the problem

Greetings
Harry

harrykl, 2021-02-15:
With the Installation instructions Raspberry - Bluetooth now it works with Zwift. Tacx connect does not work but it ist godd enough for me.
I had to do: After installing node to the system, you have to set it up for FortiusANT

$ cd ~/FortiusANT/node/
$ npm install

closed 2021-02-15 by harrykl

WouterJD, 2021-02-17:
Hi @harrykl sorry for being quite late; I think the npm install command was overlooked - I think it was in the manual already (I am reviewing the manual and do not have older versions at hand)