WouterJD/FortiusANT #255: Connection Lost

meiki1, 2021-02-13:
I use a Tacx Fortius roller with FortiusANT software. As USB dongles I use the ones from Garmin (2 pieces).
I was able to start the roller successfully and also ride in Zwift. However, it now happens again and again that the connection is lost. Error message below. This happens sometimes after 10min times after an hour. To fix the error, I then have to remove the USB plug in the PC from the role and put back in and restart the programs (FortiusANT and Zwift).
I have also started the program with mode (-a -g -n).

The two ANT connectors have also already uninstalled both and reinstalled these.

20:21:29,000: T1942 head unit initialised (Fortius)
20:21:29,548: Tacx head unit returns insufficient data, len=24
20:21:29,548: To resolve, check all (signal AND power) cabling for loose contacts.
20:21:30,204: Tacx head unit returns insufficient data, len=24
20:21:30,204: To resolve, check all (signal AND power) cabling for loose contacts.
20:21:30,532: Motor Brake Unit Firmware=0x968 Serial= 2451 year=2005 type=T1941 Version2=3084 MotorBrake=True
20:21:30,641: FortiusAnt applies the MotorBrake power curve
20:21:32,063: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
20:21:32,251: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
20:21:32,439: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
20:21:32,626: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
20:21:32,829: FortiusANT receives commands from an ANT+ Generic Remote Control
20:21:33,079: ---------- ANT-devices are activated ----------
20:42:19,250: Read from USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: Ein an das System angeschlossenes Ger\xe4t funktioniert nicht.\r\n\n'
20:42:19,812: Tacx head unit returns insufficient data, len=0
20:42:19,812: To resolve, check all (signal AND power) cabling for loose contacts.
20:42:19,812: Write to USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: Ein an das System angeschlossenes Ger\xe4t funktioniert nicht.\r\n\n'
20:42:20,531: Tacx head unit returns insufficient data, len=0
20:42:20,531: To resolve, check all (signal AND power) cabling for loose contacts.
20:42:20,562: Write to USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
20:42:21,281: Tacx head unit returns insufficient data, len=0
20:42:21,281: To resolve, check all (signal AND power) cabling for loose contacts.
20:42:21,312: Write to USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
20:42:22,031: Tacx head unit returns insufficient data, len=0
20:42:22,031: To resolve, check all (signal AND power) cabling for loose contacts.
20:42:22,063: Write to USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
20:42:22,781: Tacx head unit returns insufficient data, len=0

switchabl, 2021-02-13:
@meiki1 The problem seems to be with the T1942 head unit, not with the ANT dongles. This is a Windows error:
> 20:42:19,812: Write to USB trainer error: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: Ein an das System angeschlossenes Ger\xe4t funktioniert nicht.\r\n\n'

(= A device attached to the system is not working.)

It points either towards a driver issue or a hardware problem. The fact that it is intermittent suggests it may be the latter, but it is hard to tell. Cold temperatures, moisture and sweat getting in the head unit have been known to mess with the electronics.

WouterJD, 2021-02-17:
In addition to @switchabl: check all connections including the power to the brake to be tight

closed 2021-02-17 by WouterJD