WouterJD/FortiusANT #256: Problem connecting Fortius to FortiusAnt

yancart, 2021-02-14:
Hi, similar problem as BarryO1. I have a Tacx Fortius (without steering) with T1932 display and a laptop with Windows10 64 bit on it. Have downloaded and installed FortiusAnt following the FortiusAnt manual and the libusb-win32_T19xx packaged USB driver.  I am planning to run TTS4 and FortiusAnt on the same laptop as it seems this is possible from comments in sec 5.4. To start up FortiusAnt, I use the Windows executable and I installed the Windows prepacked drivers.

I started FortiusAnt without additional command-line parameters and the FortiusAnt GUI appeared. However, although my Fortius was connected to the computer by the Tacx USB device, FortiusAnt did not recognize my Fortius. I got the message 'No Tacx trainer found'. 

Ive checked my drivers and this looks ok:-
[image]
[image]

Has anyone seen this problem?
Thanks

switchabl, 2021-02-14:
@yancart It seems the driver package doesn't automatically replace the Tacx driver in all cases. I will probably add that to the README.

1. Right-click on the "Fortius_Unint"
2. Update Driver Software...
3. Browse my computer for driver software
4. Let me pick from a list of device drivers on my computer
5. There should be an item "Tacx T1942 (no firmware)", install that.

You will likely have to repeat this once after you start FortiusANT for the first time (and the firmware is loaded). The device will then be called "Fortius" and the right driver is "Tacx T1942".

yancart, 2021-02-19:
Hi, cant seem to find the Tacx T1942 driver. 
Which bundle should that have come with the t19xx_usb-main.zip or the libusb-win32-bin-1.2.6.0.zip?
Thanks

switchabl, 2021-02-19:
@yancart That should have been installed with the libusb-win32_T19xx.exe package (https://github.com/switchabl/t19xx_usb/releases/download/v1/libusb-win32_T19xx.exe). Looks like that didn't work for some reason?

Anyway, if all else fails, use Zadig (https://zadig.akeo.ie/) to install manually. In my experience, that almost always works, it is just a few clicks more.
1. Run Zadig
2. Options -> List All Devices
3. Find the "Fortius_Unint" device in the drop-down list
4. Choose libusb-win32 (v1.2.6.0) as driver
5. Click "Replace Driver" (or possibly "Install Driver")
6. Run FortiusANT to load the firmware and repeat with the "Fortius" device if necessary

yancart, 2021-02-20:
Hi switchabl,
Many thanks for the info. Tried the driver update and all seemed ok but still having problems...

So ran Zadig and selected Fortius-Unint:-
[image]
[image]

Then selected the correct libusb-win32 driver:-
[image]

Then rebooted laptop and ran FortiusAnt and still no connection to Fortius trainer:-
[image]

Then rechecked the drivers and now I get this:-
Seems Jungo has changed from Fortius-Unint to Fortius => is this correct?
[image]

Then recheked with Zadig and now I get this:-
So no drivers listed now?
[image]

Any thoughts...?
Thanks

switchabl, 2021-02-20:
@yancart The change from "Fortius_Unint" to "Fortius" is as expected. After you have started FortiusANT and the firmware is loaded, the "Fortius" should show up in Zadig. Did you enable "Options -> List All Devices"?

yancart, 2021-02-23:
Hi @switchabl, Progress at last and FortiusAnt now seems to be connected to the Fortius unit and is showing pwr/speed and cadence.
But next step is to get this connected to Zwift and there now seems to be an issue with the USB ANT device in that its pairing with Zwift ok but not sending any data?
Its interesting to see that there is a regular error/dropout of the FortiusAnt info display (see below):-
[image]

The Zwift app seems to be all paired ok, just no data is being Rx:-
[image]

Ive swapped the ANT dongles around and same problem.
Im running FortiusAnt and one win10 laptop and Zwift on another win10 laptop and both are running at about 30/40% CPU capacity... Have you seen this before?
Thanks

WouterJD, 2021-02-23:
Unfortunately yes: yesterday evening.
This afternoon -same configuration- no issues.

Please try to powerdown and reboot to check.

BikeBeppe64, 2021-02-23:
Hi @yancart , how many ANT dongle do you get?
Normally you always need two.

WouterJD, 2021-02-23:
At my side I can confirm: had the same problem, did not happen before and after a good night's rest (I was done with it yesterday) I cannot reproduce. 
Systems has been in sleep-mode, so a sort-of restart.

MeanHat, 2021-02-23:
Hi @yancart and @WouterJD - I have had similar issues with "Connected" and "No Signal" messages. Once I confirmed that the Bluetooth devices were within range the problem was caused by other programs/apps hijacking the Bluetooth connection. Searching for and then removing these apps or their permissions resolved the problem.

yancart, 2021-02-23:
Hi tried rebooting both laptops still the and same issue.
For ref Ive got two ANT dongles, one in each laptop.
Does ExplorAnt help for diagnosing these issues, seems to show similar ANT dropout/restart issues....
Thanks

MeanHat, 2021-02-23:
Sorry @yancart - my comment related to BLE and you are using ANT.

WouterJD, 2021-02-23:
OK. I have done some more tests.
A write() appears to fail when there are (many) messages waiting to be processed. At my side I was improving the ANT-loop and it appears I have to flush the pending messages before sending.

I do not understand why this would occur in your case, because (I think) you did not change anything there.

Please provide logfile as created with the -d127 flag

WouterJD, 2021-02-23:
Well... issue not solved:

```
12:55:14,914: ---------- ANT-devices are activated ----------
12:55:14,914: Tacx2Dongle; start main loop
12:55:14,914: clsTacxTrainer.Refresh(True, 2)
12:55:14,914: clsTacxNewUsbTrainer._ReceiveFromTrainer()
12:55:14,914: Trainer recv hdr=0x21303 data="array('B', [32, 208, 32, 0, 0, 2, 0, 0, 10, 0, 0, 0, 0, 0, 0, 0, 0, 0, 223, 3, 0, 0, 0, 0, 3, 19, 2, 0, 50, 90, 20, 0, 188, 22, 128, 89, 0, 0, 65, 252, 0, 0, 0, 0, 0, 0, 3, 241, 252, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
12:55:14,914: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=20.1 TargetRes=0 CurrentRes=-959 CurrentPower=-43, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
12:55:14,914: clsTacxNewUsbTrainer.TargetPower2Resistance(100, 5820) = 2214
12:55:14,914: clsTacxUsbTrainer.SendToTrainer(T=2, M=1, P=100, G=0, R=2214, W=85, PE=0, S=20.1, C=0)
12:55:14,914: Trainer send data="01 08 01 00 a6 08 00 00 02 0a 10 04" (len=12)
12:55:14,914:                   tacx mode=2 target=2214 pe=0 weight=10 cal=1040
12:55:14,914: Flush ANT dongle buffer
12:55:14,914: devAntDongle.__ReadAndRetry(0x81,1000,1) ...
12:55:14,914: ... done
12:55:14,914: devAntDongle.read() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
12:55:14,914: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
12:55:14,914: devAntDongle.__ReadAndRetry(0x81,1000,1) ...
12:55:14,930: ... done
12:55:14,930: AntDongle.Read() returns: ["a4 03 40 00 01 03 e5"]
12:55:14,930: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check= 0xf, info="02 10 01 ff 00 00 00 00 00" (ch=2 p=16(Main data page))
12:55:14,930: devAntDongle.write(0x01,"a4 09 4e 02 10 01 ff 00 00 00 00 00 0f") ...
12:55:14,930: ... done
12:55:14,932: devAntDongle.__ReadAndRetry(0x81,1000,1) ...
12:55:14,932: ... done
12:55:14,932: devAntDongle.read() returns ["02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "00", "01", "09", "ef", "a4", "03", "40", "06", "01", "09", "e9", "a4", "03", "40", "02", "01", "09", "ed", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "00", "01", "09", "ef", "a4", "03", "40", "06", "01", "09", "e9", "a4", "03", "40", "02", "01", "09", "ed", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "00", "01", "09", "ef", "a4", "03", "40", "06", "01", "09", "e9", "a4", "03", "40", "02", "01", "09", "ed", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "00", "01", "09", "ef"] 
12:55:14,932: Dongle.Read: 4 characters skipped 
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe3, info="06 01 03" (ch=6, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" (ch=2, msg=0x1)
12:55:14,932: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe6, info="03 01 03" (ch=3, msg=0x1)
[... cut ...]
12:55:14,947: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe9, info="06 01 09" (ch=6, msg=0x1)
12:55:14,947: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xed, info="02 01 09" (ch=2, msg=0x1)
12:55:14,947: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xec, info="03 01 09" (ch=3, msg=0x1)
12:55:14,947: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xef, info="00 01 09" (ch=0, msg=0x1)

```

WouterJD, 2021-02-23:
Second example
```
12:52:51,952: Calibrate()
12:52:51,952: ResetDongle()
12:52:51,952: Dongle    send   : synch=0xa4, len= 1, id=0x4a Reset System         , check=0xef, info="00"
12:52:51,952: devAntDongle.write(0x01,"a4 01 4a 00 ef") ...
12:52:51,952: ... done
12:52:52,467: Dongle    send   : synch=0xa4, len= 2, id=0x4d Request Message      , check=0xbf, info="00 54" (ch=0, msg=0x54)
12:52:52,467: devAntDongle.write(0x01,"a4 02 4d 00 54 bf") ...
12:52:53,483: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
12:52:53,483: ... done
12:52:53,483: Dongle    send   : synch=0xa4, len= 2, id=0x4d Request Message      , check=0xd5, info="00 3e" (ch=0, msg=0x3e)
12:52:53,485: devAntDongle.write(0x01,"a4 02 4d 00 3e d5") ...
12:52:54,501: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
12:52:54,501: ... done
12:52:54,501: Dongle    send   : synch=0xa4, len= 9, id=0x46 Set NetworkKey       , check=0x64, info="00 b9 a5 21 fb bd 72 c3 45"
12:52:54,503: devAntDongle.write(0x01,"a4 09 46 00 b9 a5 21 fb bd 72 c3 45 64") ...
12:52:55,518: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
12:52:55,518: ... done
12:52:55,518: Dongle    send   : synch=0xa4, len= 9, id=0x46 Set NetworkKey       , check=0xea, info="01 00 00 00 00 00 00 00 00"
12:52:55,520: devAntDongle.write(0x01,"a4 09 46 01 00 00 00 00 00 00 00 00 ea") ...
12:52:56,536: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
12:52:56,536: ... done
12:52:56,536: devAntDongle.__ReadAndRetry(0x81,1000,1) ...
12:52:56,538: ... done
12:52:56,538: devAntDongle.read() returns ["a4", "03", "40", "00", "01", "03", "e5"] 
12:52:56,538: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe5, info="00 01 03" (ch=0, msg=0x1)
12:52:56,538: devAntDongle.__ReadAndRetry(0x81,1000,1) ...
12:52:56,538: ... done
12:52:56,538: devAntDongle.read() returns ["a4", "03", "40", "00", "01", "03", "e5", "a4", "09", "4f", "00", "31", "ff", "ff", "ff", "ff", "ff", "c8", "00", "e4", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "06", "01", "09", "e9", "a4", "03", "40", "02", "01", "03", "e7", "a4", "09", "4f", "02", "02", "fd", "ff", "ff", "fc", "fc", "fd", "fc", "1e", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "09", "4f", "02", "46", "ff", "ff", "fe", "ff", "04", "02", "01", "a0", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "03", "01", "09", "ec", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "09", "ef", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "09", "4f", "00", "31", "ff", "ff", "ff", "ff", "ff", "c8", "00", "e4", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", "40", "06", "01", "03", "e3", "a4", "03", "40", "02", "01", "03", "e7", "a4", "03", "40", "03", "01", "03", "e6", "a4", "03", "40", "00", "01", "03", "e5", "a4", "03", [... cut ...]
12:52:56,554: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xef, info="00 01 09" (ch=0, msg=0x1)
12:52:56,554: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe9, info="06 01 09" (ch=6, msg=0x1)
12:52:56,554: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xed, info="02 01 09" (ch=2, msg=0x1)
12:52:56,554: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xec, info="03 01 09" (ch=3, msg=0x1)
12:52:56,554: Dongle    drop   : synch=0xa4, len= 1, id=0x6f Start up             , check=0xea, info="20"

```

WouterJD, 2021-02-23:
In both cases above (incorrect received message and reap-error) after this error, a burst of messages come from the ANT-interface.

yancart, 2021-02-23:
where isthe -d127 log file stored and is that the actual name "-d127.log"...?
Thanks

WouterJD, 2021-02-23:
please check manual. The logfile is stored in the current directory.

yancart, 2021-02-23:
Have checked as per the manual? 
I have the FortiusAnt app as a shortcut on the desktop. 
Searched the whole hdd and cant see any log file?
Thanks

WouterJD, 2021-02-23:
If you check the StartUp directory
You can use "FortiusAnt (gui) d.bat" as an example
The logfile will be found in that directory

yancart, 2021-02-24:
Hi, sorry still no luck finding the .log file.
There are no FortiusAnt files in either of the StartUp dirs:-
C:\Users\Clive\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Startup
or
C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp

The FortiusAnt app is kept here and started by a shortcut on the desktop, but there are no log files here:
C:\Tacx\FortiusAnt sw

Any other places it might be?
Thanks

yancart, 2021-02-27:
Hi, still no luck finding the log file... 
Any other ideas where I might it?
Thanks

MeanHat, 2021-02-27:
Hi @yancart - have you tried searching you C: drive for a *.log file? If this still does not find any log files please let me know and I will check my Windows installation. Are you launching using the Windows executable or through python?

yancart, 2021-02-27:
Yes tried that and yes lots of .log files but no FortiusAnt files.
Is it called -d127.log or FortiusAnt.log as cant see these?
I'm using the windows exe...
Thanks

MeanHat, 2021-02-27:
> Yes tried that and yes lots of .log files but no FortiusAnt files.
> Is it called -d127.log or FortiusAnt.log as cant see these?
> I'm using the windows exe...
> Thanks

Hi @yancart - I started up Windows version of FortiusANT using argument -d1 and the log files were with the python code (in my case C:/Github/FortiusAnt/FortiusANT-master/pythoncode. From memory the files are named something like FortiusAnt-dd-mm-yy-hh-mm-ss.log and FortiusAntGUI-dd-mm-yy-hh-mm-ss.log. Let me know if you need the exact name.

RodeTheAlpe, 2021-02-28:
When I use the Windows Executable the log file winds up in the FourtiusANT-master/WindowsExecutable directory
ex. FortiusAnt.2021-01-23 15-36-50.log

Take a look there

WouterJD, 2021-03-07:
Is this issue still active? 
If there is no communication here, I assume can be closed.

yancart, 2021-03-08:
Hi, Yes still trying to get this to work correctly. 
The -d1 option seems to workok  and generates a txt file. I have been able to connect to zwift occasionally but it seems very intermittent. Also when it has connected ok and a session is running on zwift, the motor seems to change between harder and easier every minuite or so whan on the flat which isnt right as it work fine for TX4. I thought this might smthg to do with the motor calibration, but after calibtrating several times successfully its had no effect on this "hunting". Is this smthg youve seen before? I'll send through the txt file if this helps..
Thanks

BarryO1, 2021-03-08:
Hi,

The way that was succesful for me, is the following:

- I use TTS4 while the trainer is connected to USB port 1. So, check that the trainer is connected to the USB port you want to use with TTS.
- Switch the trainer to the USB port you want to use with FortiusANT (I use USB port 3), the device manager should automatically assign the Jungo driver by default
- Download the libusb driver as described in chapter 3.6.2. of the manual. Do not uninstall the TTS-driver. 
- Install the libusb driver (by executing the 64 bit version in case of Windows 10) and follow install option 1 in chapter 3.6.2 of the manual so that the driver is now assigned to the trainer.
-Then, I put my Suunto ANT+ dongle in USB port 1 and I start FortiusAnt with parameters -a -g -A.
- Calibrate as described in 4.1.4. of the manual.  
- After calibration, my heart rate is shown in the GUI and I put my second ANT+ dongle (brand: Anself) in USB port 2. 
- Now I am ready and I start Zwift. 

For me this works perfect as Windows remembers which driver to choose for which USB port. So I don't have to fiddle with the drivers anymore. I just marked the USB ports so it is easy to see where to put in the trainer for which use case ;)

WouterJD, 2021-03-08:
@BarryO1 thanks, I have added it to the manual

WouterJD, 2021-03-08:
It seems we are discussing different issues in one thread:
- Connect T1942, seems solved
- TTS or FortiusAnt
- Stuttering power

Regarding "motor seems to change between harder and easier every minuite or so"; I would suggest to have a clear installation avoiding that multiple packages are addressing the same trainer.

yancart, 2021-03-09:
Hi WouterJD, the connection to Zwift still seems to be intermittent and was thinking the stuttering power was the same/similar issue? The log file now seems to work fine with d1 added. Below is the log file for an attempt to connect to Zwift and a flat route. Seems there are regular wrire exception messages, whichcoulld be tying in with the load change on the motor? The calibration seems to complete ok without issue? I can start a new thread if that helps?
Thanks

09:46:38,377: FortiusANT started
09:46:38,379:     Restart=False debug=1
09:46:38,379: -c False
09:46:38,379: -C 0/0
09:46:38,379: -d 1 (0b1)
09:46:38,379: -D None
09:46:38,379: -H None
09:46:38,379: -p 1.0
09:46:38,387: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
09:46:38,387: -t False
09:46:38,387: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
09:46:38,387: ------------------
09:46:38,387: Windows executable started
09:46:38,387: Version info for the components
09:46:38,550: Version=Fortius Antifier v5.2, on github=Fortius Antifier v5.2.
09:46:38,550: Fortius Antifier v5.2
09:46:38,550:            FortiusAnt = 2021-01-10
09:46:38,550:             antDongle = 2020-12-30
09:46:38,550:                 antFE = 2020-12-28
09:46:38,550:                antHRM = 2020-12-27
09:46:38,550:                antPWR = 2020-12-28
09:46:38,550:                antSCS = 2020-12-27
09:46:38,550:             bleDongle = 2021-01-04
09:46:38,550:             constants = 2021-01-18
09:46:38,550:                 debug = 2020-12-18
09:46:38,550:        FortiusAntBody = 2021-02-03
09:46:38,550:     FortiusAntCommand = 2021-02-01
09:46:38,550:         FortiusAntGui = 2021-02-01
09:46:38,550:               logfile = 2020-11-25
09:46:38,550:            RadarGraph = 2020-05-07
09:46:38,550:              settings = 2021-01-19
09:46:38,550:       structConstants = 2020-01-25
09:46:38,550:             TCXexport = 2020-12-20
09:46:38,550:            usbTrainer = 2021-01-12
09:46:38,550:              argparse = 1.1
09:46:38,550:                 numpy = 1.19.4
09:46:38,550:                    os = nt
09:46:38,550:               windows = 10.0
09:46:38,550:                pickle = 4.0
09:46:38,550:              platform = 1.0.8
09:46:38,550:          sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
09:46:38,550:                   usb = 1.1.0
09:46:38,550:                    wx = 4.1.1
09:46:38,550: FortiusANT code flags
09:46:38,550:    UseMultiProcessing = True
09:46:38,550:                UseGui = True
09:46:38,550:          UseBluetooth = True
09:46:38,550: ------------------
09:46:38,550: Scan for hardware
09:46:38,550: Get Dongle
09:46:39,166: Dongle - Using CYCPLUS dongle
09:46:39,166: Get Tacx Trainer
09:46:39,197: Connected to Tacx Trainer T1942
09:46:39,598: Motor Brake Unit Firmware=0x1004 Serial= 5395 year=2007 type=T1941 Version2=3077 MotorBrake=True
09:46:39,744: FortiusAnt applies the MotorBrake power curve
09:46:39,980: Tacx   - Connected to Tacx Trainer T1942
09:46:39,996: AntHRM - Heartrate expected from Tacx Trainer
09:46:39,996: Scan for hardware - end
09:46:41,493: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
09:46:41,694: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
09:46:41,895: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
09:46:42,095: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
09:46:42,311: FortiusANT receives commands from an ANT+ Generic Remote Control
09:46:42,543: Ctrl-C to exit
09:46:42,543: Tacx   - * * * * G I V E   A   P E D A L   K I C K   T O   S T A R T   C A L I B R A T I O N * * * *
09:46:49,236: Tacx   - * * * * C A L I B R A T I N G   (Do not pedal) * * * *
09:46:49,299: Target=  0W Speed=19.5kmh hr=  0 Current=  0W Cad=120 r= 628 T= 0  0 0
[... 569 lines cut ...]
09:53:51,290: Target=1.2%(201W) Speed=26.7kmh hr=  0 Current= 10W Cad= 89 r=3350 T= 1  6 1
09:53:52,324: Target=1.1%(177W) Speed=25.3kmh hr=  0 Current= 11W Cad=  0 r=3116 T= 1  6 1
09:53:53,374: Target=1.1%(154W) Speed=23.8kmh hr=  0 Current= 12W Cad=  0 r=2874 T= 1  6 1
09:53:54,414: Target=1.0%(137W) Speed=22.7kmh hr=  0 Current= 13W Cad=  0 r=2679 T= 1  6 1
09:53:55,464: Target=0.9%(121W) Speed=21.6kmh hr=  0 Current= 13W Cad=  0 r=2494 T= 1  6 1
09:53:56,505: Target=0.8%(59W) Speed=14.9kmh hr=  0 Current=  9W Cad=  0 r=1760 T= 1  6 1
09:53:57,539: Target=0.8% Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 1  6 1
09:53:58,589: Target=0.5% Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 1  6 1
09:53:59,630: Target=0.4% Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 1  6 1
09:54:00,601: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: A device attached to the system is not functioning.\r\n\n'
09:54:00,601: devAntDongle.read exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: A device attached to the system is not functioning.\r\n\n'
09:54:00,617: ANT Dongle not available; try to reconnect after 1 second
09:54:03,457: ANT Dongle reconnected, application restarts
09:54:03,863: Tacx2Dongle; Processing time 4.233 is 3.983 longer than planned 0.250 (seconds)
09:54:04,411: Dongle - Using CYCPLUS dongle
09:54:05,638: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
[... cut ...]
09:54:06,961: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
09:54:07,162: FortiusANT receives commands from an ANT+ Generic Remote Control
09:54:07,400: Ctrl-C to exit
09:54:07,400: Tacx   - * * * * G I V E   A   P E D A L   K I C K   T O   S T A R T   C A L I B R A T I O N * * * *
09:54:07,400: Tacx   - Connected to Tacx Trainer T1942
09:54:07,416: Target=0.4% Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 1  6 1
09:54:08,450: Target=0.4% Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r=   0 T= 1  6 1

WouterJD, 2021-03-09:
`09:53:41,763: Dongle - Using CYCPLUS dongle`

Please refer to #61 CYCPLUS is always problematic

yancart, 2021-03-10:
Hi I thought my CYCPLUS might have been ok as it bought well over a yr ago but maybe not?
Ok will tray and find smthg else on UK ebay.
Thanks

yancart, 2021-03-10:
Trouble is most USB Ant+ on Ebay now are CYCPLUS/ANSELF....

MeanHat, 2021-03-10:
> Hi I thought my CYCPLUS might have been ok as it bought well over a yr ago but maybe not?
> Ok will tray and find smthg else on UK ebay.
> Thanks

@yancart - This one (Dynastream) from Wiggle UK works OK : https://www.wiggle.co.uk/lifeline-ant-usb

WouterJD, 2021-03-10:
BLE is cheaper. For this money you can consider to by a Raspberry. Raspberry is in last test-cycles and fully supported.

closed 2021-03-10 by WouterJD