WouterJD/FortiusANT #258: Tacx Fortius resistance?

Fonssch, 2021-02-15:
Hello everyone,

I have installed FortiusANT and connected my Tacx Fortius to it.
The calibration runs all the way through, but as soon as I start cycling, the trainer reacts irregularly. I keep getting varying resistance while I don't change anything.
Does anyone know how I can fix that?

switchabl, 2021-02-15:
@Fonssch Do you have the cadence sensor connected properly? The brake will stop after a few seconds if no pedal signal is detected (this is a safety limitation built into the Powerback unit).

see also #213 #243

MeanHat, 2021-02-16:
Hi @Fonssch - as well as the cadence sensor issue highlighted by @switchabl I also experienced the same problems with a CYCPLUS ANT dongle on with a Tacx Fortius. (see #61)

Fonssch, 2021-02-16:
Thank you, I did not connect the cadance sensor properly, I will try it

WouterJD, 2021-02-17:
HI @Fonssch 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

WouterJD, 2021-02-17:
> Thank you, I did not connect the cadance sensor properly, I will try it

I assume issue can be closed

closed 2021-02-17 by WouterJD

Fonssch, 2021-02-18:
Yes, it worked!
Nice piece of software that gave my tacx a new life.

The magnet of the cadence sensor did not work, replaced it with another one and it worked.

Thank you for your quick responses and sharing the software.