WouterJD/FortiusANT #259: Issue with T1942 fortius

JurgenRoeland, 2021-02-16:
Hi Wouter,

I'm having some troubles in connecting my fortius T1942 old blue head with fortiusant
I can see that the log box says Stop CPU and start CPU  but then the following message appears:
GetTrainer - Unable To load Firmware.
I can send you a screenshot by mail if it helps
I installed the libusb-win32-bin-1.2.6.0 drivers.

thanks in advance for some hints on making it work.

Jurgen Roeland

switchabl, 2021-02-16:
@JurgenRoeland The T1942 appears as a new device (with a different ID) after the firmware is loaded. You will need to install the libusb drivers again for that device. Alternatively you can install the pre-packaged driver from https://github.com/switchabl/t19xx_usb/ which should take care of that automatically.

JurgenRoeland, 2021-02-16:
Thanks for the quick answer, I'll give it a try this afternoon.

Mag ook in het Nederlands mocht je NL sprekend zijn :)

gr

Jurgen

JurgenRoeland, 2021-02-16:
With the prepackaged driver in the link, it installs but when running fortiusant says 'no tacx trainer found'
lights on my blue head are on and cable inserted properly.

switchabl, 2021-02-16:
@JurgenRoeland Can you find the trainer in the Windows device manager? If the driver was installed successfully, it should appear under "libusb-win32 devices". If not, it may be show up under "Jungo" or as an unknown device (see also #256)

JurgenRoeland, 2021-02-16:
That did the trick for the recognition of the tacx driver ! thanks.
Fortius ant is also broadcasting some devices on the ant. fortiusant did the calibration.
However, when I'm looking in the tacx desktop app, it can connect to devices (The devices seen by tacx desktop app are tacx neo2T 57591, speed cadance 57595 and power sensor 575596) I can connect to them but when I start workouts are anything using them, it keeps saying it lost connection.
Any idea's where I can look for ?

switchabl, 2021-02-17:
@JurgenRoeland Good to hear that the driver is working. I don't have any experience with the Tacx Desktop App (but I think others are using it successfully), maybe someone else can chime in on that too.

What ANT dongles are you using? There are some that are known to give problems. You could also try another Software to see if it the problem is specific to the Tacx App. RGT Cycling has a free version and Zwift and Rouvy both offer a free trial.

WouterJD, 2021-02-17:
@JurgenRoeland 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

I think issue can be closed

closed 2021-02-17 by WouterJD

JurgenRoeland, 2021-02-18:
Thanks for all the help so far. I have noticed that the two dongles I had received are or Cycplus brand. After reading about it I decided to send them bak to the shop. I have a friend who also bought some and I will first give it a try with his dongle to see if that works better. I'm located in Belgium/Geraardsbergen and using a very old Tacx Fortius old blue head T1492. If the new dongle is working with fortiusant I'll give it a try on trial swift first and will let you know how things went. cheers.

WouterJD, 2021-02-18:
Hi @JurgenRoeland "very old Fortius" is a Tacx qualification - when working it's as good as any other one!
Like to hear from you!

JurgenRoeland, 2021-02-18:
I could not agree more on your last statement... 
I'll update as soon as I get it working.

JurgenRoeland, 2021-02-24:
Just wanted to let you know that we have now tried with different ant dongles fom bikeinn.com (Zycle USB ANT+ Stick Black One ) and it now works fine with my old fortius blue. Thanks for all the help.