WouterJD/FortiusANT #260: Please make FortiusANT ID's dynamic

WouterJD, 2021-02-17:
I don't know if this device identifier is generated by FortiusANT itself or if FortiusANT is able to change the identifier. Or can I do this somewhere in the driver settings of one of the ANT+ dongles? Any help appreciated.

**EDIT:** already was able to fix it changing the device numbers (line 208 - 215 in pythoncode/antDongle.py) to a lower value (as you already took the highest possible values) on one of the two laptops. Now Zwift correctly recognizes two different devices (TACX FE-C 57491 and TACX FE-C 57591) if both laptops are running FortiusAnt. It would be really cool if the device IDs could be set in the json settings file too (restricting it to an appropriate value range).

Besides that we are really happy that you made it possible we can use our old trainers in a modern world :) thank you very much!

_Originally posted by @WolfRevo in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-773930707_

WouterJD, 2021-02-22:
In version 5.3 you can specify -B12345 defining the base of the DeviceNumbers, now 57590

WolfRevo, 2021-02-23:
Thank you very much! 😁

WouterJD, 2021-03-05:
You can test in version 6.0; branch Raspberry_Release6

WolfRevo, 2021-03-05:
Already done and works as expected 👍😎

closed 2021-03-05 by WouterJD