WouterJD/FortiusANT #261: Question regarding setup of tacx i-Vortex t1961

mmfire, 2021-02-20:
Hi guys!

I'm a newbie on github, so my apologies in advance if this is the wrong place to ask a question.

My hardware setup is the t1961 brake, t1962 head unit and a computer with an ant+ dongle.

I've managed to get everything connect and running in FortiusANT, but I don't understand how I can connect to my iphone or desktop with zwift/trainerroad. Do I need a bluetooth dongle connected to the computer for my iphone to connect? Or for desktop, do I need another ANT+ dongle inserted into the same computer? Please advice.

[image]

switchabl, 2021-02-20:
@mmfire Essentially yes, you need a second dongle. FortiusANT takes up the existing dongle to connect to the Vortex and create a virtual ANT "Smart" trainer. You have two option:
- add a second ANT dongle to the PC so Zwift can use that to connect to the virtual ANT trainer (and any other ANT sensors like heart rate strap or power meter that you might have)
- add a Bluetooth LE dongle to you PC; then FortiusANT can create a virtual Bluetooth trainer instead and your phone can connect to that directly (ANT dongle is still needed to communicate with the Vortex).

The Zwift desktop app actually works with both options because it can use you phone's Bluetooth (through the Zwift Companion app).

WouterJD, 2021-02-22:
hi @mmfire 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

mmfire, 2021-02-23:
@switchabl thank you for your reply!

I have added a Bluetooth LE dongle (already in use, working fine, connected to a speaker)
https://www.aliexpress.com/item/1005001273586697.html?spm=a2g0s.12269583.0.0.306f3c56n9BUcV

However, FortiusANT seem to have an issue with detecting/using it:

"Bluetooth interface cannot be opened; [WinError 2] The system cannot find the file specified"

[image]

Any tips?

switchabl, 2021-02-23:
@mmfire Setting up BLE is currently a bit more involved, carefully read the documentation first: https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md

Also note that FortiusANT needs to take direct control of the Bluetooth dongle, so not all models are supported (https://github.com/noble/node-bluetooth-hci-socket#windows) and you will not be able to use it for other things (like the speaker) at the same time.

I don't have any personal experience with the BLE option so if you can't get it to work someone else will likely need to help.

WouterJD, 2021-02-23:
Following the mentioned instructions -precisely- (also included in the manual) does the job.
But care must be taken!

I just went through the process on Windows and things work.

mmfire, 2021-02-23:
I got it working on my macbook air 🎉

Aha, I have not read this specific bluetooth documentation. Thanks! I will see if I can get it working for Windows as well.

Thanks a bunch! :)

closed 2021-02-23 by mmfire