WouterJD/FortiusANT #262: Unknow error in runtime "AttributeError: 'SpeedMeter' object has no attribute 'DisableFocusFromKeyboard'"

hasise, 2021-02-20:
Hi

When running FortiusAnt I get the following error "AttributeError: 'SpeedMeter' object has no attribute 'DisableFocusFromKeyboard'".
The UI actually starts but for example the speed is going faster and faster, very odd but I assume things has gone
wrong earlier in the process.

Does anyone have an idea what might have gone wrong?

Regards, Hans Simonsen

Running Ubuntu 20.04 on a Raspberry Pi 3

[kalle2.txt](https://github.com/WouterJD/FortiusANT/files/6016038/kalle2.txt)

switchabl, 2021-02-20:
@hasise I think I've seen that one before. If I remember correctly, the problem was that the wxPython version was too old.

hasise, 2021-02-20:
Fresh install so there isn't anything to update I'm afraid.

switchabl, 2021-02-20:
Can you give some more details on the Python environment (e.g. output of `python --version` and `python -m pip list`)? Did you install wxPython through pip or the system package manager?

hasise, 2021-02-21:
Version and list of all pip packages included.
wxPython does not exist in the official Ubuntu repositories, therefore used pip to install.

[list.txt](https://github.com/WouterJD/FortiusANT/files/6017814/list.txt)
[version.txt](https://github.com/WouterJD/FortiusANT/files/6017816/version.txt)

hasise, 2021-02-22:
A note, to be able to install I had to temporarily change to a newer Pi (4B, 2GB) since my Pi 3(1 GB) run out of memory and stopped responding after a very long time. Installing using the newer Pi worked, took some time but no errors. I then moved the SD card back to the Pi 3.
1 GB of memory is obviously not enough for installing FortiusANT running Ubuntu on a Pi 3.
If there isn't a solution I'm thinking of switching to Raspberry Pi OS. Unfortunately the 64 bit version is still in Beta
so the 32 bit will have to do.

hasise, 2021-02-28:
Second try, installed Raspberry Pi OS, changed the swap size and also killed the X server to save memory.
From that on the the guide in "3.5.5 Linux Ubuntu 20.04" was followed as much as possible.

1. In section 2 the ubuntu packages where skipped
2. The command "sudo python3 ./FortiusAnt/pythoncode/FortiusAnt.py" fails with an error like:
module not found lib_programname. lib_programname can't be installed from pip or directly from
git (https://github.com/bitranox/lib_programname.git)

Has anyone successfully installed FortiusANT as is on Raspberry Pi using Raspberry Pi OS?

Next try would be to install an a PC where TTS 4 is running but I'm not happy doing so since
I'm not convinced that it doesn't break TTS 4, correct me if I'm wrong :)
So my second question is if someone has done this?

MeanHat, 2021-03-01:
Hi @hasise - it is indeed possible to install FortiusANT on RPi with Raspberry Raspberry Pi OS installed using Pi Imager (https://www.raspberrypi.org/%20downloads/). The instructions for using Bluetooth were produced by @wtal710174 and are here: #192. If you want to use an ANT dongle the instructions were developed by @martin-vi and can be found here: https://gist.github.com/martin-vi/5206b6b26faf6f2d6121f6c8f300e44d. ANT works with RPi Zero/3/4 and BLE with RPi 3/4 (unless anyone else can confirm they are running BLE with FortiusANT on a Pi Zero)

hasise, 2021-03-03:
I have now installed FortiusANT according to the instructions, no issues during the installation.
The instructions doesn't mention any configuration for the Taxc usb port, I got an access denied so I used the instructions here except for the group that I changed to pi.
[](https://github.com/decodeais/FortiusANT/blob/Raspi-Status-LED/doc/HeadlessRaspi.md)
SUBSYSTEM=="usb", ATTRS{idVendor}=="3561", ATTRS{idProduct}=="1932",MODE="0666", OWNER="pi", GROUP="pi"

When testing with the flag -s -a it finds the Taxc hardware but I get an error for the ANT+ dongle.
From the log file:
      bInterval        :    0x1
20:28:27,496: GetDongle - Exception: [Errno 13] Access denied (insufficient permissions)
20:28:27,497: Dongle - GetDongle - ANT dongle in use
20:28:27,506: Connected to Tacx Trainer T1932
20:28:27,810: Motor Brake Unit Firmware=0x1902 Serial=0 MotorBrake=False
20:28:27,911: FortiusAnt applies the MagneticBrake power curve
20:28:28,012: Retry because short buffer (len=64) or incorrect header received (expected: 0x21303 received: 0xc03)
20:28:28,114: Tacx   - Connected to Tacx Trainer T1932
20:28:28,114: AntHRM - Heartrate expected from Tacx Trainer
20:28:28,114: FortiusAnt ended
[AntDebug.txt](https://github.com/WouterJD/FortiusANT/files/6079180/AntDebug.txt)

Second test this time without an parameters I get the same error as in my first installation. I guess
I can live without a UI since I will connect to Zwift but it would have been nice to be able to use the UI.

An example of a rules file in /etc/udev/rules.d/ that works with the type of installation @martin-vi provided would be apricated

switchabl, 2021-03-04:
@haise I'm sorry I still don't really have an idea about the GUI (seems to be different from the issue I had). But I do have some udev files that should work.

**65-antusb.rules**
```
# ANTUSB2
SUBSYSTEM=="usb", ATTR{idVendor}=="0fcf", ATTR{idProduct}=="1008", MODE="0664", GROUP="plugdev"

# ANTUSB-m
SUBSYSTEM=="usb", ATTR{idVendor}=="0fcf", ATTR{idProduct}=="1009", MODE="0664", GROUP="plugdev"
```
**66-tacx-trainer.rules**
```
# Tacx T1902 "solid green"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1902", MODE="0664", GROUP="plugdev"'

# Tacx T1904 "white/green"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1904", MODE="0664", GROUP="plugdev"'

# Tacx T1932 "white/blue"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1932", MODE="0664", GROUP="plugdev"

# Tacx T1942 "solid blue" (no firmware)
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="e6be", MODE="0664", GROUP="plugdev"

# Tacx T1942 "solid blue"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1942", MODE="0664", GROUP="plugdev"
```

hasise, 2021-03-04:
Thanks will test them later on today or tomorrow.
MODE is different so that might be something.
I noticed a difference in the syntax, first your example:
SUBSYSTEM=="usb", **ATTR**{idVendor}=="3561", **ATTR**{idProduct}=="1932", MODE="0664", GROUP="plugdev"
My example using the syntax from @martin-vi 
SUBSYSTEM=="usb", **ATTRS**{idVendor}=="3561", **ATTRS**{idProduct}=="1932",MODE="0666", GROUP="pi"
Should it be ATTR or ATTRS or doesn't it matter?

Anything else I can do to find out why the UI doesn't start? Any logs output or anything else I can provide?

MeanHat, 2021-03-04:
@hasise - I am successfully using ATTRS{..} from @martin-vi's gist with some modifications I was sent. This results in:

SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664", GROUP="plugdev"
SUBSYSTEM=="tty", ACTION=="add", ATTRS{idProduct}=="1009", ATTRS{idVendor}=="0fcf", MODE="0666", GROUP="plugdev"

I was reminded to make sure to add "plugdev" (e.g. $ sudo addgroup plugdev and $ sudo adduser pi plugdev) and after reboot $ lsusb -v should show your ant device (in my case as iManufacturer Dynastream Innovations, Inc., iProduct ANTusb-m Stick and iSerial 3 179).

Although the RPI 0 is run headless with ANT for testing/setup I was able to use the GUI and simulation mode with the -g and -s parameters when calling up FortiusAnt.py. When this was done using a startup script (like the one from @wtal710174 in #192) the GUI worked if I called up the script in "terminal mode".

hasise, 2021-03-04:
Thanks for the reply, will also test with my second ANT+ dongle (Garmin USB 2 instead of the Tacx dongle)
Just a quick question, is there something in your installation instructions that actually prevents from running the UI?
(except from the start parameters of course)

WouterJD, 2021-03-05:
@hasise 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

hasise, 2021-03-06:
Thanks, will do a proper introduction as soon as I get things working.
Today I made progress and I have 3 new question. First of all a progress report.
The USB problems are now sorted by applying these rules (running as "pi"): 
# ANTUSB2
SUBSYSTEM=="usb", ATTR{idVendor}=="0fcf", ATTR{idProduct}=="1008", MODE="0664", GROUP="pi"
# Tacx T1932 "white/blue"
SUBSYSTEM=="usb", ATTR{idVendor}=="3561", ATTR{idProduct}=="1932", MODE="0664", GROUP="pi"

When starting without a UI it starts up without errors and then starts to display rows with current speed, cadence, power etc.
If I then start pedaling it follows what I do, really promising.

1. I still can't run a the UI and started to think about if it is possible to run this if installed as @martin-vi described. Maybe something isn't ok when running a UI or even not possible at all using venv?
2. How is the command "-c" used? I assume I need to calibrate or maybe I'm misunderstood the meaning of this command.
When for example using TTS I can and should calibrate regularly (device configuration/brake calibration)
3. Is there a way to test the ANT+ connection, using something that works in a similar way as Zwift on a PC?

MeanHat, 2021-03-06:
Hi @hasise:
1. @martin-vi's installation works fine for me either headless or with GUI using the -g parameter when calling FortiusAnt.py
2. when using a T1932 FortiusAnt automatically enters calibrate more for me unless I specify -n. When using a Tacx iMagic which uses a T1902 head unit FortiusAnt does not enter calibration mode whether -n is specified or not. I calibrate this manually using a manual runoff test.
3. Could you test your ANT connection using a Garmin device? I've not tried this so just a suggestion...

WouterJD, 2021-03-06:
> 3\. Is there a way to test the ANT+ connection,

Why would you want to do that?

hasise, 2021-03-07:
About bullet 3, to check if it is worth trying to connect from for example Zwift. In fact I did test it last night using my Garmin watch and it worked eg found 3 new sensors.

hasise, 2021-03-07:
Still can't get the UI to work and thought that the python3-wxgtk4.0 package might be the problem.
Experimented a bit by uninstalling the python3-wxgtk4.0 package since it should be possible to install without it.
So I used the command "sudo apt remove python3-wxgtk4.0"
and then "(venv) pi@raspberrypi:~ $ pip3 install -r ~/FortiusANT/pythoncode/requirements.txt
[install.txt](https://github.com/WouterJD/FortiusANT/files/6097115/install.txt)

This generates a lot of errors, see attached file.
@WouterJD When installing did you install python3-wxgtk4.0 manually or did you just install FortiusANT/pythoncode/requirements.txt ?

wtal710174, 2021-03-07:
@hasise I think the best way to install the wxPython package is to follow the instruction in #192. Search for "Install wxPython" there is a package to download and the installation instruction is right below the download link. You can also read the new [FortiusAnt manual] (https://github.com/WouterJD/FortiusANT/blob/Raspberry_Release6/doc/FortiusANTUserManual.pdf) that @WouterJD created inside the Raspberry_Release6 branch and follow the instruction for the Raspberry install to do a complete new install, which is simpler than following the #192 install. Since this is build for bluetooth you will still have to add the additional rights for the ANT Dongle.

WouterJD, 2021-03-07:
@hasise I installed following the manual, being tge instructions @wtal710174 provided.
I would suggest: take a clean SD-card and try to reproduce. Would be interesting to see what happens then

hasise, 2021-03-08:
Will do, great that ther is a Raspberry pi section in the manual.

hasise, 2021-03-09:
I installed FortiusANT according to the new manual, no major issues, some excepted warnings and I had to set
the scrips to executable that's all.
When booting FortiusANT starts and the GUI is also started, a bit poor picture quality but it is possible to see.
[image]
[image]

So the Taxc trainer is found but no ANT, I tested to ride my bike but nothing happened, maybe that is expected behavior. 
I modified the start script a bit and tested with the following parameters -a -g -D-1 (I don't have Bluetooth or LED's)
In my previous attempt I didn't use the -D parameter, the setting "-D-1" what does that do? ANT worked fine in my
previous installation, it's a Garmin USB2 ANT+ device.

WouterJD, 2021-03-09:
See manual:
`-D antDeviceID	Select one specific antDongle (perhaps with a non-standard deviceID). The following types are known: 4104 = Suunto, 4105 = Garmin, 4100 = Older.`

-D -1 means: no ANT-dongle

So remove -D-1 from FortiusAnt.sh

hasise, 2021-03-10:
Sorry didn't understand -D-1 meant no ANT dongle. I have now removed the -D parameter and will do a new test later on today.

WouterJD, 2021-03-11:
great news!

WouterJD, 2021-03-17:
Heaps of details here; please help me forward: what changes to the manual would you suggest?

hasise, 2021-03-17:
Seems like the last part of the manual has been cut off. The last page is 25 of 58 can you please check.

WouterJD, 2021-03-17:
Will check, printing has gone wrong. It's at my side :-(

WouterJD, 2021-03-19:
Manual is repaired, was a driver issue.
Is there anything on this issue I can help with, otherwise we can close

hasise, 2021-03-19:
The manual can now be downloaded so it's ok to close. I'm in the process of reading through it but haven't completed that task yet. Will let you know if there is something I think should be be changed.

closed 2021-04-29 by WouterJD