WouterJD/FortiusANT #263: Single sponsorship payment

Mickyduck55, 2021-02-20:
I must be stupid but can no way to ask a question without opening an issue.  I would like to send a single sponsorship payment .. not happy with an on going model.  Could this option be added to the sponsor page?
Wouter please delete after you have read this ;)

WouterJD, 2021-02-22:
Hi @Mickyduck55 
YOu could ping me on facebook messenger with your question.

WouterJD, 2021-02-22:
Currently github sponsorship is monthly and I did not find an option to enable a Single sponsorship.

WouterJD, 2021-02-22:
Hi @Mickyduck55 just let me know how you like to contact

WouterJD, 2021-02-22:
I have added following remark to the [sponsoring page](https://github.com/sponsors/WouterJD):
- - -
PS. Github does not provide one-time sponsoring.
If you would prefer that option, you can find me on paypal; "Wouter Dubbeldam"

closed 2021-02-22 by WouterJD