WouterJD/FortiusANT #264: Elite turbo trainers

Paulospg99, 2021-02-21:
Hello, there is a possibility to put my old Elite Realpower CT Internet (wireless) working with FortiusAnt?

switchabl, 2021-02-22:
@Paulospg99 No, FortiusANT has support for legacy Tacx trainers (Fortius, i-Magic, Flow, Vortex, Bushido, Genius) only.

From a technical perspective, the Elite Realpower Wireless seems to use a proprietary ANT-based protocol similar to the Vortex/Bushido/Genius. So something like FortiusANT could likely be built for the Elite but someone would have to reverse-engineer the protocol first which is not a trivial task.

WouterJD, 2021-02-22:
HI @Paulospg99 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

closed 2023-12-15 by WouterJD