WouterJD/FortiusANT #266: Snelheid FortiusAnt niet gelijk aan Zwift

BikerVibor, 2021-02-24:
Hoi,
Sorry voor NL, Engels niet echt in thuis.
Ik heb alles mooi aan de praat, maar als ik volgens de FortiusAnt 25 km/u rij, maakt Zwift daar 15 km/u van, ben er nog niet achter hoe ik dat kan synchroniseren. Hoe kan ik dat aanpassen?
Het fietsen gaat ook nogal schokkerig.
Setup:
Beone op Fortius, via USB op 1 PC met 2 BT/ANT modules. 

[image]

WouterJD, 2021-02-24:
Hoi @BikerVibor 
Welkom in de FortiusANT community
- - -
Ik ben altijd nieuwsgierig met wie ik communiceer, waar FortiusANT wordt gebruikt en in welke configuratie.
Welke bundle heb je gekocht, en welke rem en head-unit gebruik je?
Vertel meer over jezelf, eventueel onder issue #14.
- - -

WouterJD, 2021-02-24:
Zwift en FortiusANT zullen hetzelfde vermogen tonen.
FortiusANT toont de snelheid van het wiel
Zwift rekent de snelheid uit op basis van vermogen en gewicht / helling van de route.

Deze twee snelheden zijn niet gelijk

BikerVibor, 2021-02-25:
Hoi Wouter,
Sorry, ik dacht dat ik mijzelf al op de kaart gezet had. Maar vast een missertje gehad met opslaan of zoiets. Alsnog gedaan.

Leuk programma FortiusAnt, kost wel wat moeite om je er in in te leven, maar dat gaat steeds beter. 
De afwijking in snelheid is wel erg groot met Zwift. Ik ga eens wat spelen met de instellingen in Zwift en andere trainigsprogramma's proberen.

Grt. Vincent

WouterJD, 2021-03-10:
Speed in Zwift and FortiusAnt will never be equal, due to different meaning.

closed 2021-03-10 by WouterJD