WouterJD/FortiusANT #267: BKool support

WilburBear, 2021-02-24:
I am running a set up with Bluetooth initially on zwift.  Wanted to try BKool but it cant pair to me windows PC.  I only have free subscription but doubt that's the issue.

I know BKOOL isn't on the list of target platforms - is my issue because BKOOL isn't compatible?

WilburBear, 2021-02-25:
As an additional note - upgraded to v5.2 and also tried BKOOL on a different (more powerful) win10 PC, same result, although as an additional point, I am able to connect and read heart rate from FortiusAnt, I start it with gui, autostart, BLE, d, s

WilburBear, 2021-03-03:
Logged a ticket with BKool - but no real help (suggested I go through pairing process etc....)  Does anyone have experience with BKool or is it just the case its not compatible?

switchabl, 2021-03-03:
@WilburBear The BLE protocol (FTMS) is standardized so it _should_ work even if it not explicitly listed. It is however possible that no one has tried that combination (BKool + BLE) and there is some bug there. Unfortunately I don't have a working Bluetooth setup right now so I can't try myself.

WilburBear, 2021-03-04:
I tried a few other things:

BKool running on laptop with built-in BLE 
Upgrade to premium subscription

Neither of these made any difference.  In all cases FortiusANT heartrate is able to pair but nothing else 

I have paired BKOOL with my sperate heartrate and speed cadence sensors also...

WouterJD, 2021-03-05:
@WilburBear 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

Easiest way to debug: install Trainer Road or Zwift on your phone with trial account and try to pair. TR and Zwift have quite advanced pairing pages that are known working well.

WouterJD, 2021-03-17:
No additional info received

closed 2021-03-17 by WouterJD

taylor42788, 2021-04-10:
I have a running setup of FortiusAnt 5.2 that works with Zwift with ANT+.
I've tried today to install BKool and to connect BKool with FortiusAnt as "SmartTrainer" with ANT+.
After some retries, it finally seem to connect but seem to disconnect right after & the setup is not working.
I've scrolled the logs but did not see any error/warning reported.
Any clue?
Thanks