WouterJD/FortiusANT #268: Tacx Genius

mrKevin07, 2021-03-01:
Hi
I have read the manual and all the wiki etc. 

I have FortiusANT 5.2 working but swift won't find my genius as controllable device. It finds it as power meter and speed sensor. 

Any suggestions? 
Thanks

switchabl, 2021-03-01:
@mrKevin07 It would be helpful if you could provide some more detailed info on your setup:
- What platform are you using for FortiusANT (Windows/Linux/Mac)?
- What platform are you using for Zwift: PC? Android? iOS?
- Are you using ANT or Bluetooth for Zwift and what ANT/BLE sticks are you using?

I seem to remember that Zwift on Android did not support ANT controllables. So if that is what you are using you will either need to switch to Bluetooth or install Zwift on the PC.

mrKevin07, 2021-03-01:
Thanks for your help.

I'm using windows for fortiusant, with Garmin ant sticks. Using Android for swift. I'll try another platform and report back if it works. Cheers

mrKevin07, 2021-03-03:
Ok tested on PC and follow it connected easily so definitely a problem with genius and zwift on Android using ant. Cheers

closed 2021-03-03 by mrKevin07