WouterJD/FortiusANT #269: Calibrating T1901/T1932 (i-Flow T2220/T2230 bundle)

kjoglum, 2021-03-01:
First of all, many thanks for the FortiusANT contribution. Being a rider from Norway, motivated by FortiusANT/Zwift to start using my Flow Multiplayer trainer again.

So, with FortiusANT v5.2, I successfully connected my trainer, and I also successfully connected to Zwift. After doing some test rides, I did a FTP test with ERG enabled (all from Zwift), and the measured/"felt" power seemed realistic (although not having any power meter to actually confirm values).

Next day, after checking tyre pressure/performing runoff test, I started a zone benchmarking workout (ERG enabled), based on the FTP from the previous day. However, the felt resistance made it almost impossible to achieve target power (Zwift set at 50%), even at first steps (trying to maintain high cadence). Reading from wiki/manual, I see there is a narrow dynamic window wrt resistance, although the chosen Zwift track was not very steep, and Zwift was set at 50%.

Wondering if there are any tests/calibration to perform, without power meter, to validate my setup?

kjoglum, 2021-03-02:
Just realized there might be an issue with my cadence measurements.

As part of my initial setup, cadence was registerred/measured in FortiusANT (then using the standard cadence equipment following the Tacx trainer). For my Zwift runs, I did not check FortiusANT wrt measurements, as signals were transferred and all measurements (including cadence) were active in Zwift. 

However, now troubleshooting after the exhausting benchmarking ride, I realized that cadence is not measured (remains 0) in FortiusANT, although cadence still has signal in Zwift and is registerred/measured.

switchabl, 2021-03-03:
@kjoglum There is no proper calibration (without power meter) for the T1901 yet. The best practice currently looks something like this:
- make sure your tire is inflated to 7-8 bars (assuming normal 23mm road/trainer tire)
- leave default calibration value as it is (do not set -c or -p)
- leave "Trainer difficulty" at default (50%) setting in Zwift (for Rouvy only: set -c 50/50 in FortiusANT)
- ride for ~5 minutes at low intensity to warm up the tire
- then use the "runoff" function and adjust the knob on the trainer until you get a time of about 6.5s
That way power/resistance should be roughly right at least.

The limited resistance range means that steeper climbs (and downhills) will not be realistic. Also you may well have to adjust your gears to stay within the brake range in ERG mode.

The cadence issue sounds strange. Is Zwift receiving the cadence signal from FortiusANT or do you have an additional cadence/power sensor connected? If FortiuANT does indeed transmit cadence but does not show it that sounds like a bug. In any case it should not have an impact on power/resistance. With the T1901 the cadence data is really just displayed/transmitted, it is not used for brake control.

kjoglum, 2021-03-04:
@switchabl 
Appreciate your help.

Calibration procedure was followed, and I did/do not have any calibration parameters set. Zwift difficulty also set at 50%. My gear setting in FortiusANT set and kept as per default, 50/19.

Uncertain what happened wrt cadence. During testing afterwards, FortiusANT reported 0, whereas values >0 was measured in Zwift (only one cadence sensor, connected directly to trainer, thus FortiusANT should be the only option for sending signal). Went over the cadence setup, so now I have measured values both in FortiusANT and Zwift. 

Also, haven’t been able to reproduce the issue of high resistance/unable to achieve target power.

WolfRevo, 2021-03-05:
We have two 1901/1932 systems here and both struggle with the cable connection of the cadence sensor every few month. Then you get weird cadence readings like only 60% of the actual cadence or it just doesn't read the cadence at all. Stupid as it sounds it just helps always to check the cable and the connector each time to get the cadence sensor running again.

WouterJD, 2021-03-05:
Hi @kjoglum 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
As @WolfRevo said: check cabling, Tacx trainers are sensitive to bad contacts. There is no (effective) error-checking on these interfaces. Bad contact = incorrect data.

kjoglum, 2021-03-05:
@WolfRevo @WouterJD 
Appreciate your response.

Seems about right regarding faulty cable/connection. As part of troubleshooting, I cut the plastic strip used to secure the cable to the bike, disconnected and reconnected cable, and the cadence value jumped to an insane value in FortiusANT before dropping to 0 (could off course also be related to signal transfer while connecting). During riding, normal cadence was achieved, both in FortiusANT and Zwift.

Although cadence apparently does not relate to power/resistance, and I have no clue why I had this experience of a exhausting ride during a Zwift workout at even at warmup (ERG mode), I will close this raised issue as I haven't had this issue since, and I am really satisfied with the system/setup.

Thanks again for enabling smarter "old" trainers!

closed 2021-03-05 by kjoglum