WouterJD/FortiusANT #270: Rouvy does not detect FortiusANT

slilja, 2021-03-05:
I am using Bushido T1981 brake with FortiusANT experimental `BushidoDirect` option. FortiusANT connects to brake (displays correct speed, cadence and power) but I can not connect Rouvy to FortiusANT. 
I am running FortiusANT on MacBook, using CHILEAF ANT+ dongle. This Mac also has USB BT dongle. I am running Rouvy Mobile on iPhone. 
When I start Rouvy it asks for trainer type. Which trainer should I select? Does FortiusANT work with Rouvy?
[image]

BikeBeppe64, 2021-03-05:
@slilja 

Try to select:
* Other
* General ANT+ FEC trainer

slilja, 2021-03-06:
I made a mistake with command and forget to use option `-b` when starting FortiusANT. 

But now I have a new issue: when I start FortiusANT with `python FortiusAnt.py -b -a -g -t  BushidoDirect` I get lot of error messages
requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x1159696a0>: Failed to establish a new connection: [Errno 61] Connection refused')) 

I noticed an error message regarding bleno module, but did not find anything that helps me to install that. I also checked issues #253 and #192 but with no help. One tip was to run `npm install` in subdir node/. I did that, but also that command fails with something related to bleno.
[errorlog.txt](https://github.com/WouterJD/FortiusANT/files/6094988/errorlog.txt)

slilja, 2021-03-06:
Actually, I can close this. Seems that problems are related to BushidoDirect-branch. When I run main branch FortiusAnt with T1982 head unit, is now connects nicely to Rouvy on iPhone and Zwift on AppleTV.
- - -
If someone wants to look at problems in BushidoDirect branch, here is some more debugging. 
I am using Bushido T1981 brake and T1982 headunit. ANT+ connects to headunit. FortiusANT runs on MacBook.

Case 1 `python FortiusAnt.py -a -g -b`: GUI starts, no errors on console. Does not open bluetooth interface  and also does not find trainer.
Case 2 `python FortiusAnt.py -a -g -b -t Bushido`: GUI starts and finds trainer ("Tacx Bushido paired: 5404 / Using Dynastream Innovations dongle, Bluetooth interface opened to CTP"). Writes a lot of (bleno-related?) errors on console.
Whatever I do, I can not connect Rouvy or Rouvy AR or Zwift (all running on iPhone). CTP does not find device.

If I remember correctly, the same issues happen when I use `-t BushidoDirect` instead of `-t Bushido`.

closed 2021-03-06 by slilja