WouterJD/FortiusANT #272: Bushido/Tacx T1991 AntUSB2 driver

AdyGB, 2021-03-06:
Hi, I'm really struggling to find a driver that's win10 compatible for my T1991 PC Upgrade ANT stick. The result is FortiusANT will not recognize my T1982 head unit or T1981 brake.

Any help for a programming complete amateur would be greatly received.

Are others who are using a Bushido setup able to connect via another ANT stick?

WouterJD, 2021-03-07:
Please run FortiusANT with -d127 flag and post logfile.
If your dongle has a deviating deviceID, use -D flag to specify

AdyGB, 2021-03-08:
10:59:33,603: FortiusAnt GUI started in child-process
10:59:33,991: Check FortiusAntTitle.py on github
10:59:33,992: Version=Fortius Antifier v5.2, on github=Fortius Antifier v5.2.
10:59:34,198: mp-GuiMessageToMain(conn, 19592, True, None, None)
10:59:34,206: mp-GuiAnswerFromMain(conn) returns (19596, (None, 'No (free) ANT-dongle found', None))
10:59:34,206: mp-GuiAnswerFromMain(conn) returns (19596, ('Pair with Tacx Bushido controller', None, None))
10:59:34,206: mp-GuiAnswerFromMain(conn) returns (19596, (None, None, 'Heartrate expected from Tacx Trainer'))
10:59:34,261: mp-GuiAnswerFromMain(conn) returns (19592, False)
10:59:34,404: SetValues() done in 143 ms

AdyGB, 2021-03-08:
I'm using the Fortius driver for the Tacx T1991 ant stick from the Jungo driver package - is this correct?

WouterJD, 2021-03-08:
Could you please attach the whole logfile, FortiusAnt.2021-*.log
I'm curious what ANT-id's are returned.

AdyGB, 2021-03-08:
11:33:36,870: FortiusANT started
11:33:36,871:     Restart=False debug=127
11:33:36,871: -a
11:33:36,872: -A
11:33:36,872: -c False
11:33:36,872: -C 0/0
11:33:36,873: -d 127 (0b1111111)
11:33:36,873: -D None
11:33:36,873: -g
11:33:36,874: -H 0
11:33:36,875: -p 1.0
11:33:36,885: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
11:33:36,890: -t Bushido
11:33:36,891: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
11:33:36,892: ------------------
11:33:36,894: Windows executable started
11:33:36,895: Version info for the components
11:33:37,247: Check FortiusAntTitle.py on github
11:33:37,247: Version=Fortius Antifier v5.2, on github=Fortius Antifier v5.2.
11:33:37,252: Fortius Antifier v5.2
11:33:37,253:            FortiusAnt = 2021-01-10
11:33:37,253:             antDongle = 2020-12-30
11:33:37,253:                 antFE = 2020-12-28
11:33:37,253:                antHRM = 2020-12-27
11:33:37,253:                antPWR = 2020-12-28
11:33:37,253:                antSCS = 2020-12-27
11:33:37,253:             bleDongle = 2021-01-04
11:33:37,253:             constants = 2021-01-18
11:33:37,253:                 debug = 2020-12-18
11:33:37,253:        FortiusAntBody = 2021-02-03
11:33:37,253:     FortiusAntCommand = 2021-02-01
11:33:37,253:         FortiusAntGui = 2021-02-01
11:33:37,253:               logfile = 2020-11-25
11:33:37,253:            RadarGraph = 2020-05-07
11:33:37,253:              settings = 2021-01-19
11:33:37,253:       structConstants = 2020-01-25
11:33:37,253:             TCXexport = 2020-12-20
11:33:37,253:            usbTrainer = 2021-01-12
11:33:37,253:              argparse = 1.1
11:33:37,253:                 numpy = 1.19.4
11:33:37,253:                    os = nt
11:33:37,253:               windows = 10.0
11:33:37,253:                pickle = 4.0
11:33:37,253:              platform = 1.0.8
11:33:37,253:          sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
11:33:37,253:                   usb = 1.1.0
11:33:37,253:                    wx = 4.1.1
11:33:37,253: FortiusANT code flags
11:33:37,253:    UseMultiProcessing = True
11:33:37,253:                UseGui = True
11:33:37,253:          UseBluetooth = True
11:33:37,253: ------------------
11:33:38,574: mp-MainCommandFromGui() returns (19592, None, None)
11:33:38,574: Scan for hardware
11:33:38,574: Get Dongle
11:33:38,574: GetDongle - Check for dongle 4104 Suunto
11:33:38,589: GetDongle - Check for dongle 4105 Garmin
11:33:38,590: GetDongle - Check for dongle 4100 Older
11:33:38,590: GetDongle() returns: No (free) ANT-dongle found
11:33:38,590: mp-MainDataToGUI(19596, (None, No (free) ANT-dongle found, None))
11:33:38,590: Get Tacx Trainer
11:33:38,590: clsTacxTrainer.GetTrainer()
11:33:38,590: clsTacxTrainer.__init__()
11:33:38,590: clsTacxAntTrainer.__init__()
11:33:38,590: mp-MainDataToGUI(19596, (Pair with Tacx Bushido controller, None, None))
11:33:38,590: mp-MainDataToGUI(19596, (None, None, Heartrate expected from ANT+ HRM))
11:33:38,590: Scan for hardware - end
11:33:38,590: mp-MainRespondToGUI(19592, False)
11:33:38,798: clsTacxTrainer.Refresh(True, 0)
11:33:38,819: clsTacxTrainer.Refresh(True, 0)
[... 385 lines cut ...]
11:35:09,423: clsTacxTrainer.Refresh(True, 0)
11:35:09,675: clsTacxTrainer.Refresh(True, 0)
11:35:09,934: clsTacxTrainer.Refresh(True, 0)
11:35:10,179: clsTacxTrainer.Refresh(True, 0)
11:35:10,190: Bushido page 0 (OUT) Keep-alive
11:35:10,439: clsTacxTrainer.Refresh(True, 0)
11:35:10,698: clsTacxTrainer.Refresh(True, 0)
11:35:10,948: clsTacxTrainer.Refresh(True, 0)
11:35:11,198: clsTacxTrainer.Refresh(True, 0)
11:35:11,442: clsTacxTrainer.Refresh(True, 0)
11:35:11,705: clsTacxTrainer.Refresh(True, 0)
11:35:11,957: clsTacxTrainer.Refresh(True, 0)
11:35:12,213: clsTacxTrainer.Refresh(True, 0)
11:35:12,455: clsTacxTrainer.Refresh(True, 0)
11:35:12,714: clsTacxTrainer.Refresh(True, 0)
11:35:12,973: clsTacxTrainer.Refresh(True, 0)
11:35:13,221: clsTacxTrainer.Refresh(True, 0)
11:35:13,482: clsTacxTrainer.Refresh(True, 0)
11:35:13,749: clsTacxTrainer.Refresh(True, 0)
11:35:14,007: clsTacxTrainer.Refresh(True, 0)
11:35:14,259: clsTacxTrainer.Refresh(True, 0)
11:35:14,515: clsTacxTrainer.Refresh(True, 0)
11:35:14,774: clsTacxTrainer.Refresh(True, 0)
11:35:15,033: clsTacxTrainer.Refresh(True, 0)
11:35:15,298: clsTacxTrainer.Refresh(True, 0)

AdyGB, 2021-03-08:
Update..... I've successfully updated the driver for my T1991 usb ant stick and now and older version of TTS can connect to my T1982 headunit - partial success!
Still unable to connect to FortiusAnt though.
Most recent log file:

17:12:42,167: FortiusANT started
17:12:42,167:     Restart=False debug=127
17:12:42,168: -a
17:12:42,168: -A
17:12:42,168: -c False
17:12:42,168: -C 0/0
17:12:42,169: -d 127 (0b1111111)
17:12:42,169: -D None
17:12:42,169: -g
17:12:42,170: -H 0
17:12:42,172: -p 1.0
17:12:42,173: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
17:12:42,173: -t Bushido
17:12:42,174: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
17:12:42,175: ------------------
17:12:42,176: Windows executable started
17:12:42,177: Version info for the components
17:12:42,573: Check FortiusAntTitle.py on github
17:12:42,573: Version=Fortius Antifier v5.2, on github=Fortius Antifier v5.2.
17:12:42,575: Fortius Antifier v5.2
17:12:42,575:            FortiusAnt = 2021-01-10
17:12:42,575:             antDongle = 2020-12-30
17:12:42,575:                 antFE = 2020-12-28
17:12:42,575:                antHRM = 2020-12-27
17:12:42,575:                antPWR = 2020-12-28
17:12:42,575:                antSCS = 2020-12-27
17:12:42,575:             bleDongle = 2021-01-04
17:12:42,575:             constants = 2021-01-18
17:12:42,575:                 debug = 2020-12-18
17:12:42,575:        FortiusAntBody = 2021-02-03
17:12:42,575:     FortiusAntCommand = 2021-02-01
17:12:42,575:         FortiusAntGui = 2021-02-01
17:12:42,575:               logfile = 2020-11-25
17:12:42,575:            RadarGraph = 2020-05-07
17:12:42,575:              settings = 2021-01-19
17:12:42,575:       structConstants = 2020-01-25
17:12:42,575:             TCXexport = 2020-12-20
17:12:42,575:            usbTrainer = 2021-01-12
17:12:42,575:              argparse = 1.1
17:12:42,575:                 numpy = 1.19.4
17:12:42,575:                    os = nt
17:12:42,575:               windows = 10.0
17:12:42,575:                pickle = 4.0
17:12:42,575:              platform = 1.0.8
17:12:42,575:          sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
17:12:42,575:                   usb = 1.1.0
17:12:42,575:                    wx = 4.1.1
17:12:42,575: FortiusANT code flags
17:12:42,575:    UseMultiProcessing = True
17:12:42,575:                UseGui = True
17:12:42,575:          UseBluetooth = True
17:12:42,575: ------------------
17:12:44,173: mp-MainCommandFromGui() returns (19592, None, None)
17:12:44,173: Scan for hardware
17:12:44,173: Get Dongle
17:12:44,173: GetDongle - Check for dongle 4100 Older
17:12:44,238: GetDongle - Check for dongle 4105 Garmin
17:12:44,246: GetDongle - Check for dongle 4104 Suunto
17:12:44,246: GetDongle() returns: No (free) ANT-dongle found
17:12:44,246: mp-MainDataToGUI(19596, (None, No (free) ANT-dongle found, None))
17:12:44,246: Get Tacx Trainer
17:12:44,246: clsTacxTrainer.GetTrainer()
17:12:44,246: clsTacxTrainer.__init__()
17:12:44,246: clsTacxAntTrainer.__init__()
17:12:44,246: mp-MainDataToGUI(19596, (Pair with Tacx Bushido controller, None, None))
17:12:44,255: mp-MainDataToGUI(19596, (None, None, Heartrate expected from ANT+ HRM))
[... 2720 lines cut ...]
17:24:00,833: clsTacxTrainer.Refresh(True, 0)
17:24:01,082: clsTacxTrainer.Refresh(True, 0)
17:24:01,340: clsTacxTrainer.Refresh(True, 0)
17:24:01,588: clsTacxTrainer.Refresh(True, 0)
17:24:01,845: clsTacxTrainer.Refresh(True, 0)
17:24:02,104: clsTacxTrainer.Refresh(True, 0)
17:24:02,341: clsTacxTrainer.Refresh(True, 0)
17:24:02,342: Bushido page 0 (OUT) Keep-alive
17:24:02,610: clsTacxTrainer.Refresh(True, 0)
17:24:02,869: clsTacxTrainer.Refresh(True, 0)
17:24:03,129: clsTacxTrainer.Refresh(True, 0)
17:24:03,385: clsTacxTrainer.Refresh(True, 0)
17:24:03,635: clsTacxTrainer.Refresh(True, 0)
17:24:03,878: clsTacxTrainer.Refresh(True, 0)
17:24:04,127: clsTacxTrainer.Refresh(True, 0)
17:24:04,388: clsTacxTrainer.Refresh(True, 0)
17:24:04,656: clsTacxTrainer.Refresh(True, 0)
17:24:04,906: clsTacxTrainer.Refresh(True, 0)
17:24:05,162: clsTacxTrainer.Refresh(True, 0)
17:24:05,421: clsTacxTrainer.Refresh(True, 0)
17:24:05,669: clsTacxTrainer.Refresh(True, 0)
17:24:05,928: clsTacxTrainer.Refresh(True, 0)
17:24:06,185: clsTacxTrainer.Refresh(True, 0)
17:24:06,447: clsTacxTrainer.Refresh(True, 0)
17:24:06,700: clsTacxTrainer.Refresh(True, 0)

WouterJD, 2021-03-08:
Isn't there a logfile FortiusAnt.2021-*.log, I think this is FortiusAnt**GUI**.2021-*.log

AdyGB, 2021-03-09:
@WouterJD Apologies, still learning!

ran again this morning to refresh it...
07:36:07,411: FortiusANT started
07:36:07,411:     Restart=False debug=127
07:36:07,411: -a
07:36:07,411: -A
07:36:07,411: -c False
07:36:07,411: -C 0/0
07:36:07,411: -d 127 (0b1111111)
07:36:07,411: -D None
07:36:07,411: -g
07:36:07,411: -H 0
07:36:07,411: -p 1.0
07:36:07,411: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
07:36:07,416: -t Bushido
07:36:07,416: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
07:36:07,417: ------------------
07:36:07,417: Windows executable started
07:36:07,417: Version info for the components
07:36:07,921: Check FortiusAntTitle.py on github
07:36:07,921: Version=Fortius Antifier v5.2, on github=Fortius Antifier v5.2.
07:36:07,921: Fortius Antifier v5.2
07:36:07,921:            FortiusAnt = 2021-01-10
07:36:07,921:             antDongle = 2020-12-30
07:36:07,921:                 antFE = 2020-12-28
07:36:07,921:                antHRM = 2020-12-27
07:36:07,921:                antPWR = 2020-12-28
07:36:07,921:                antSCS = 2020-12-27
07:36:07,921:             bleDongle = 2021-01-04
07:36:07,921:             constants = 2021-01-18
07:36:07,921:                 debug = 2020-12-18
07:36:07,921:        FortiusAntBody = 2021-02-03
07:36:07,921:     FortiusAntCommand = 2021-02-01
07:36:07,921:         FortiusAntGui = 2021-02-01
07:36:07,921:               logfile = 2020-11-25
07:36:07,921:            RadarGraph = 2020-05-07
07:36:07,921:              settings = 2021-01-19
07:36:07,921:       structConstants = 2020-01-25
07:36:07,921:             TCXexport = 2020-12-20
07:36:07,921:            usbTrainer = 2021-01-12
07:36:07,921:              argparse = 1.1
07:36:07,921:                 numpy = 1.19.4
07:36:07,921:                    os = nt
07:36:07,921:               windows = 10.0
07:36:07,921:                pickle = 4.0
07:36:07,921:              platform = 1.0.8
07:36:07,921:          sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
07:36:07,921:                   usb = 1.1.0
07:36:07,921:                    wx = 4.1.1
07:36:07,921: FortiusANT code flags
07:36:07,921:    UseMultiProcessing = True
07:36:07,921:                UseGui = True
07:36:07,921:          UseBluetooth = True
07:36:07,921: ------------------
07:36:09,217: mp-MainCommandFromGui() returns (19592, None, None)
07:36:09,217: Scan for hardware
07:36:09,217: Get Dongle
07:36:09,217: GetDongle - Check for dongle 4100 Older
07:36:09,231: GetDongle - Check for dongle 4105 Garmin
07:36:09,234: GetDongle - Check for dongle 4104 Suunto
07:36:09,234: GetDongle() returns: No (free) ANT-dongle found
07:36:09,234: mp-MainDataToGUI(19596, (None, No (free) ANT-dongle found, None))
07:36:09,234: Get Tacx Trainer
07:36:09,234: clsTacxTrainer.GetTrainer()
07:36:09,234: clsTacxTrainer.__init__()
07:36:09,234: clsTacxAntTrainer.__init__()
07:36:09,234: mp-MainDataToGUI(19596, (Pair with Tacx Bushido controller, None, None))
07:36:09,234: mp-MainDataToGUI(19596, (None, None, Heartrate expected from ANT+ HRM))
07:36:09,234: Scan for hardware - end
[... 435 lines cut ...]
07:37:54,858: clsTacxTrainer.Refresh(True, 0)
07:37:55,119: clsTacxTrainer.Refresh(True, 0)
07:37:55,371: clsTacxTrainer.Refresh(True, 0)
07:37:55,618: clsTacxTrainer.Refresh(True, 0)
07:37:55,862: clsTacxTrainer.Refresh(True, 0)
07:37:56,116: clsTacxTrainer.Refresh(True, 0)
07:37:56,367: clsTacxTrainer.Refresh(True, 0)
07:37:56,606: clsTacxTrainer.Refresh(True, 0)
07:37:56,864: clsTacxTrainer.Refresh(True, 0)
07:37:57,111: clsTacxTrainer.Refresh(True, 0)
07:37:57,366: clsTacxTrainer.Refresh(True, 0)
07:37:57,618: clsTacxTrainer.Refresh(True, 0)
07:37:57,874: clsTacxTrainer.Refresh(True, 0)
07:37:58,137: clsTacxTrainer.Refresh(True, 0)
07:37:58,391: clsTacxTrainer.Refresh(True, 0)
07:37:58,681: clsTacxTrainer.Refresh(True, 0)
07:37:58,937: clsTacxTrainer.Refresh(True, 0)
07:37:59,197: clsTacxTrainer.Refresh(True, 0)
07:37:59,455: clsTacxTrainer.Refresh(True, 0)
07:37:59,724: clsTacxTrainer.Refresh(True, 0)
07:37:59,978: clsTacxTrainer.Refresh(True, 0)
07:38:00,234: clsTacxTrainer.Refresh(True, 0)
07:38:00,494: clsTacxTrainer.Refresh(True, 0)
07:38:00,732: clsTacxTrainer.Refresh(True, 0)
07:38:00,732: Bushido page 0 (OUT) Keep-alive

AdyGB, 2021-03-09:
From my limited (very) understanding of what's going on.... it appears FortiusAnt isn't picking up my Ant stick. I've double checked the driver - working ok, I've also uninstalled Jungo drivers and also TTS completely.
TTS did pair with my headunit via the same T1991 Ant stick previously.
I'm at a loss!

AdyGB, 2021-03-11:
@WouterJD I've ordered a reasonably priced garmin ant+ stick to try instead of my old Tacx T1991. Fingers crossed it works. I'll close this issue. Thanks.

closed 2021-03-11 by AdyGB

WouterJD, 2021-03-11:
@AdyGB hope to receive positive news!