WouterJD/FortiusANT #273: ANT+ dongle found on Windows but not on Mac/OSX

pietercjanssen, 2021-03-07:
Hello,

I have just started to try and use FortiusANT, and it seems a very nice solution to keep using my old Tacx I-Flow. Thanks a lot!

However, so far I have not succeeded to run it satisfactory on my MacBook Air. The programme is running in the manual ride mode and my Tacx trainer connects correctly. But my ANT+ dongle is not recognized. The error message I get is "No backend, check libusb: No backend available", although I did install libusb using 'brew install libusb'. 

In addition, on a Windows PC, FortiusANT correctly recognizes the dongle (Dynastream Innovations), so the dongle appears to be correct.

Would any Mac/OSX user have a suggestion how to solve this problem?

Kind regards,
Pieter Janssen, Den Haag NL

pietercjanssen, 2021-03-07:
In addition to my earlier message: 
Looking closer, also my trainer appears not to be recognized in FortiusANT on my new Macbook, with the same error message: "Get Trainer - No backend, check libusb: No backend available".
While this works fine on my old Macbook (just got a new one since last week). 

So it should be something in my OSX settings, libraries, .... 

Hope that anyone can help.

regards
Pieter

WouterJD, 2021-03-08:
Hi @pietercjanssen welcome!
Sorry that I cannot help you, not having a Mac.

> I did install libusb using 'brew install libusb'.
and
> it should be something in my OSX settings, libraries, ....

I would think this points in the right direction; a full clean install is not fun but might provide a cleat situation

pietercjanssen, 2021-03-08:
Hi @WouterJD 
Thanks for your reply.
Unfortunately a full clean install did not do the trick.

Hopefully there is someone else on the platform to help me out. 

In the meantime I will try once more with another version of Python, since I discovered some issues with the latest versions in combination with my OSX version.
And I'll try to find an OSX/programming expert (which I am certainly not myself...) among my friends.

marcoveeneman, 2021-03-08:
Hi Pieter,

I'm using a mac to run FortiusAnt and recently i also configured one for my dad, works like a charm. I can try to help you with this issue, however i am running with BLE only and didn't encounter the error you are seeing here so i'm not 100% sure where to start.

I see you got an error like 'No backend available', this is an error from the pyusb library that is used by FortiusAnt to communicate with the Tacx and also the ANT+ dongles. Somehow the library failed to find the required libusb driver. I found a page on GitHub that may help to identify what is wrong: https://github.com/walac/pyusb/blob/master/docs/faq.rst#how-do-i-fix-no-backend-available-errors Can you give this a try and see what the result is?

pietercjanssen, 2021-03-09:
Hi @marcoveeneman,

Thanks for your help.
I tried the script you suggested and got the following:

ERROR:usb.libloader:'Libusb 1' could not be found
ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
ERROR:usb.libloader:'OpenUSB library' could not be found
ERROR:usb.backend.openusb:Error loading OpenUSB backend
ERROR:usb.libloader:'Libusb 0' could not be found
ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

Indeed, it appears that the required libusb driver is not found.

Hope you can help me further!

WouterJD, 2021-03-09:
@pietercjanssen I assume you have same result?

pietercjanssen, 2021-03-09:
@WouterJD 
yes, unfortunately.
forgot to mention; sorry

marcoveeneman, 2021-03-09:
Hi @pietercjanssen,

Thanks for trying to reinstall libusb. As far as i can see the log doesn't show an error as you already mentioned. However, one thing i notice, it shows arm64 platform a couple of times. Are you running on a mac with the new M1 chip? Maybe this can be a clue for further investigation.

pietercjanssen, 2021-03-09:
Hi @marcoveeneman 

Ah, indeed I am running on a mac with the new m1 chip. Do you have any suggestions for a next step?
I will start a search for solutions on the web as well.

WouterJD, 2021-03-09:
Reading Marco's link https://github.com/walac/pyusb/blob/master/docs/faq.rst#how-do-i-fix-no-backend-available-errors

Generally, there are four possible causes for this problem:
1. You didn't install libusb library.
2. Your libusb library isn't in the standard shared library paths.
3. Your libusb version is too old.
4. Your PyUSB version is too old.

all we did now is address #1 and I assume also #3.
How about options #2 and #4 and (#2) did you reboot after install (to get the path updated)?

"stupid" questions, but it could be something easy

marcoveeneman, 2021-03-09:
I found an issue on GitHub which looks similar to this one. I think it is related to the M1 chip so i think you can forget about the other link. See https://github.com/pyusb/pyusb/issues/355. Maybe it will already help you with searching for a solution already, I'll try to dive a bit further into this tomorrow.

pietercjanssen, 2021-03-09:
> Reading Marco's link https://github.com/walac/pyusb/blob/master/docs/faq.rst#how-do-i-fix-no-backend-available-errors
>
> Generally, there are four possible causes for this problem:
>
> 1. You didn't [...]

hi @WouterJD 
Just tried a full restart. Didn't help.
And what I did before also is add the paths above (/opt/homebrew/cellar/libusb/...... etc.) to the path via the .zprofile (in fact trying to solve #2), but that didn't help as well.

pietercjanssen, 2021-03-09:
> I found an issue on GitHub which looks similar to this one. I think it is related to the M1 chip so i think you can forget about the other link. See [pyusb/pyusb#355](https://github.com/pyusb/pyusb/issues/355). Maybe it will already help you with searching for a solution, I'll try to dive a bit further into this tomorrow.

Thanks! @marcoveeneman. I will try to look into it as well (but I'm afraid it's beyond my current knowledge...)

pietercjanssen, 2021-03-10:
Just to let you know: I tried several different re-installs of brew and python versions, but so far no luck.

pietercjanssen, 2021-03-12:
Hi @WouterJD and @marcoveeneman, 

It looks like I have come a serious step further. The 'No backend available' error is solved, and my Tacx i-Flow trainer is now recognized. Basically, installing python3.9 under Homebrew and doing a number of the subsequent installation steps manually solved the issue. 

Unfortunately, I can't check if the ANT+ dongles would be recognized as well, because yesterday I decided to return them to the webshop (since I was not sure if I would be able to use them). But I am rather confident they would be recognized.

I will now first try to get fully operational with BLE. After that, when I am sure everything is working, I'll be glad to write down the steps I have taken for the interest of other users.

WouterJD, 2021-03-13:
> I'll be glad to write down the steps I have taken for the interest of other users.

Good to hear, and I'm glad to hear to installation steps to be taken.

pietercjanssen, 2021-03-13:
Thought I solved it, but now I get unexpected errors all of a sudden, when starting FortiusANT....
Looking further...

pietercjanssen, 2021-03-13:
Below are the steps I took for installation on a MacBook Air with M1 chip (running OSX Big Sur, v11.1).
This worked to recognize the Tacx trainer in FortiusANT (but unfortunately I still have an issue regarding BLE connection, see next comment).

1. Install Homebrew:
`/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"`

2. Install Python via Homebrew:
`brew install python@3.9`

3. Add the correct paths to .zprofile:
```
PATH="/opt/homebrew/bin:/opt/homebrew/sbin:/opt/homebrew/opt/python@3.9/bin:/opt/homebrew/opt/python@3.9/libexec/bin:${PATH}"
export PATH
```
4. Add alias for python and pip in ‘zprofile:
```
alias python=python3
alias pip=pip3
```

5. Install pyusb:
`pip install pyusb`

6. Install libusb with brew (if I remember correctly):
`brew install libusb`

7. Step-by-step install the modules _pyinstaller_, _requests_, _wxPython_ and _wheel_, as also indicated in the requirements.txt file (since pip install -r requirements.txt did not work):
`brew install numpy `
(as pip install did not work)

```
pip install pyinstaller
pip install requests
pip install wxPython
pip install wheel
```

8. Manually install the module lib_programname (since pip install lib_programname did not work) as follows:
`python -m pip install lib_programname `

WouterJD, 2021-03-13:
> Are you running on a mac with the new M1 chip?

> Thought I solved it, but now I get unexpected errors all of a sudden, when starting FortiusANT....
Looking further...

Hmm... I hope for you, you are not getting into a portability issue, which will be difficult to solve without appropriate access to the affected code.

pietercjanssen, 2021-03-13:
> Hmm... I hope for you, you are not getting into a portability issue, which will be difficult to solve without appropriate access to the affected code.

This one I solved, by re-installing node (no idea why it crashed). See above

pietercjanssen, 2021-03-13:
In addition to the above, I applied the installation steps for use with BLE (manual, section 3.8.2)
In FortiusANT, this results in the message _"Bluetooth interface opened to CTP"_.
But unfortunately, the device is not recognized in the CTP (Tacx in my case), and I get error messages from FortiusANT:

```
23:28:26,679: ---------- BLE-devices are activated ----------
23:28:26,690: ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x12268f580>: Failed to establish a new connection: [Errno 61] Connection refused'))
node:internal/modules/cjs/loader:927
  throw err;
  ^

Error: Cannot find module 'bleno'
Require stack:
- /Applications/Github/FortiusANT/node/virtual-trainer.js
- /Applications/Github/FortiusANT/node/server.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:924:15)
    at Function.Module._load (node:internal/modules/cjs/loader:769:27)
    at Module.require (node:internal/modules/cjs/loader:996:19)
    at require (node:internal/modules/cjs/helpers:92:18)
    at Object.<anonymous> (/Applications/Github/FortiusANT/node/virtual-trainer.js:1:15)
    at Module._compile (node:internal/modules/cjs/loader:1092:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1121:10)
    at Module.load (node:internal/modules/cjs/loader:972:32)
    at Function.Module._load (node:internal/modules/cjs/loader:813:14)
    at Module.require (node:internal/modules/cjs/loader:996:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/Applications/Github/FortiusANT/node/virtual-trainer.js',
    '/Applications/Github/FortiusANT/node/server.js'
  ]
}
23:28:26,954: ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x12268feb0>: Failed to establish a new connection: [Errno 61] Connection refused'))
```

I will do my best to find a clue and let you know if I solved the issue (but being far from a programming expert, any suggestions are appreciated as well).

pietercjanssen, 2021-03-14:
Hi @WouterJD 

Could it be that there is an omission in the manual, section 3.8?

In the last line I read: 
• Install dependencies:   `cd node` and `npm install`

I understood this as: from the folder in which node is situated run `npm install bleno`
However, that didn't resolve the issue. And several other solutions (found on the web) were not successfull either.

WouterJD, 2021-03-15:
The BLE installation instructions for Mac OS are quite compact and perhaps candidate for improvement.

In the node directory you have to execute the npm install command to "compile the FortiusAnt part of Node"

pietercjanssen, 2021-03-29:
Hi @WouterJD and @marcoveeneman,

I fully forgot to let you know that - using the 8 installation steps I indicated above, I succeeded to connect my Tacx i-flow to a CTP via ANT+ (2 dongles).

Unfortunately, i did not manage to get the BLE-connection operatrional. I searched a lot about this issue on the web, which revealed that possibly this is due to the fact that there's not yet 100% compatibility between several (Python) modules and the new apple M1 chip.
I might try again later this year.

In the meantime I am really happy with my now functioning trainer! Very good job, this FortiusANT!

I will put my location on the map, and I have joined the sponsoring program.
Kind regards

WouterJD, 2021-03-30:
> I have joined the sponsoring program

I received the confirmation; Thanks!

closed 2021-06-07 by WouterJD