WouterJD/FortiusANT #274: Testing Raspberry_Release6

decodeais, 2021-03-07:
I started the test.
First problem I saw was that the LED's for Tacx and Shutdown show a dim light. I changed them to GPIO16 and DPIO20.
On this ports every thing is OK. The reason are the default internal pullups. Another solution would be to make them activ low and connect them to a power pin, 
Second Problem switching on does not work on RPi4 by GPIO5. I made a bridge to GPIO3, then it works, but there may be trouble when I2C should be used.

WouterJD, 2021-03-07:
1. I did not realize pull-up/pull-down difference during Raspberry Pi start-up untill FortiusANT is running, because then LEDs are initialized. The selection was 5 leds on a row next to GND. To be added in manual "dim light when booting"
I don't know whether it's a problem.

2. I don't understand what you mean re GPIO5. Please explain.

Thanks for feedback!

decodeais, 2021-03-07:
When the program runs everything is OK. Such dimmed LEDs make me nervous. In such situations i am always thinking is there something wrong?
How do you switch on again. On some RPi you can use GPIO5, on the RPi4 you can use GPIO3 to restart. I made bridge from the button input to this input. Now I can restart with te same button.

WouterJD, 2021-03-07:
> How do you switch on again.

Good point, please provide some more info. 
As complete novice on RPI I pull plug and reconnect, but you are going to give a better solution, I understand😉

decodeais, 2021-03-07:
Make a wire bridge from the button (GPIO12) to GPI03 and it works to switch on. This was the reason for connecting the button in my version to GPIO3. It saves one IO.

WouterJD, 2021-03-07:
Ref

> There's nothing to build here, but we need to understand how to wake up the Pi from a halt state before we build the shutdown functionality. Simply put, shorting pins 5 and 6 (GPIO3 and GND) together will wake the Pi up from a halt state.

You mean: FortiusANT should use GPIO3 instead of GPIO5?

decodeais, 2021-03-08:
For the RPi4  GPIO3 is a better solution, but I don't know all variants.
I think GPIO3 was used by I2C too. But pull down this pin should by also possible.
In my manual I additional configured the start of the `shutdown`   in the config.txt file. It is nice to be able to shutdown without additional software.

BikeBeppe64, 2021-03-08:
[image]

Get it now

BikeBeppe64, 2021-03-08:
[image]

Get it now

WouterJD, 2021-03-08:
> my manual I additional configured the start of the `shutdown` in the config.txt file

DId I miss your manual; please share link

decodeais, 2021-03-08:
Its only in the discription how to install  jow to install without any display or keyboard. https://github.com/decodeais/FortiusANT/blob/Raspi-Status-LED/doc/HeadlessRaspi.md

BikeBeppe64, 2021-03-08:
> Its only in the discription how to install  jow to install without any display or keyboard. https://github.com/decodeais/FortiusANT/blob/Raspi-Status-LED/doc/HeadlessRaspi.md

My 2 cent: 
* we can install Raspberry OS on a SD card * USB 3 , It's 2/3 times faster and more than 32GB space, just in case.

[image]

WouterJD, 2021-03-08:
Good work @BikeBeppe64 !
In next version, button is on GPIO3 for described reasons.
Just modify https://github.com/WouterJD/FortiusANT/blob/Raspberry_Release6/pythoncode/raspberry.py#L129

> * we can install Raspberry OS on a SD card * USB 3 ...
  
What is the reason that the SD_card on USB3 is faster than in the onboard SD-card slot?

WouterJD, 2021-03-08:
@decodeais @wtal710174 @meanhat @martin-vi @BikeBeppe64 
what do you think, should GPIO-pins be customizable? -l [on|off]/3/5/6/13/19/26 (defaults if not specified)

JSON:

[Raspberry]
Leds = True | False
LedTacx = 5
LedShutdown = 6
LedCadence = 13
LedBLE = 19
LedANT = 26
BtnShutdown= 3

WouterJD, 2021-03-08:
> Its only in the discription

@decodeais I saw that link and assumed that issue #192 would contain all info; no problem - has been adopted with pleasure

wtal710174, 2021-03-08:
> @decodeais @wtal710174 @MeanHat @martin-vi @BikeBeppe64
> what do you think, should GPIO-pins be customizable? -l [on|off]/3/5/6/13/19/26 (defaults if not specified)

@WouterJD: Yes i think this is a great idea, it allows to quickly configure the pins used and does not really affect the code. Maybe another option for the button to specify, if the button is connected as pull up or pull down button.

MeanHat, 2021-03-08:
@WouterJD - release 6 works well on RPi 0 in ANT+ mode (I wanted to obtain a log file for ANT in order to allow the miniTFT to work with ANT as well as BLE like you suggested). I have yet to get RPi0 and Bluetooth to work despite Node adaptations for Armv6l architecture.
Like @wtal710174 I like the idea of making the GPIO pins customisable.

BikeBeppe64, 2021-03-08:
> Good work @BikeBeppe64 !
> In next version, button is on GPIO3 for described reasons.
> Just modify https://github.com/WouterJD/FortiusANT/blob/Raspberry_Release6/pythoncode/raspberry.py#L129
>
>
> What [...]

https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md

bandwidth and I can use more than one OS on IT.

- Raspberry PI 4 B has the new eeprom and can boot from usb 2/3
-reconized ANT+ usb Antifer dongle

WouterJD, 2021-03-08:
@BikeBeppe64 I have added it to the manual in section "Installation decisions"; the manual will remain a straight-forward FortiusAnt-dedicated installation which is good for the novice and a reference for others.

WouterJD, 2021-03-08:
> release 6 works well on RPi 0 in ANT+ mode

@MeanHat I'm really looking forward to the docu; I will integrate with pleasure. Rpi0 would be THE device for the FortiusAnt bridge. Could be a EUR 25,= solution (excluding the ANTdongle, but BLE available)

MeanHat, 2021-03-09:
@WouterJD - As requested, here are some suggested additions/changes for RPi0 which I hope are useful. This explains some of the differences and how to get it running with ANT (it will need a further script and one modified script based on @martin-vi's instructions to do this. I have highlighted the changes that would be needed.
[FortiusAnt User Manual v6 - suggested RPi0 modifications.docx](https://github.com/WouterJD/FortiusANT/files/6111430/FortiusAnt.User.Manual.v6.-.suggested.RPi0.modifications.docx)
I think the next step for me is to continue to figure out the install of BLE on RPi0 with armv6l architecture or ask for help.
Hope this helps and is what you were looking for...

WouterJD, 2021-03-10:
@MeanHat thanks for the feedback. Unfortunately, github does not provide a neat WYSIWYG and version-managed wiki and therefore it's word/pdf.
 
- SSH remark added
- password remark already there
- Rpi0 mentioned; manual focusses on Rpi3/Rpi4
- 2_InstallPackagesANT.sh --> please provide
- 2_InstallPackagesBLE.sh renamed
- 4_InstallWxPython.sh renamed and modified
- 9_GrantAccessToUSB_withReboot --> please provide additional rules. I do not particularly like the cat>> construction and would prefer to have a 10-usbaccess.rules file in the directory. What do you think? Also [SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664", GROUP="usbtacx"] might be generic enough. I would keep it simple; if somebody wants to finetune USB-access, I would leave that for "personal refinement". See also 3.12.2 Installation decisions.
- Command line reference added
- Button / TFT; planned for release 6.1. I guess that there are many possibilities which we cannot support all in FortiusAnt. As soon as my TFT arrives we can do the integration and then document it as well.

WouterJD, 2021-03-10:
> I think the next step for me is to continue to figure out the install of BLE on RPi0 with armv6l

Is the message that FortiusAnt fully works on Rpi0, with exception of -g?

WouterJD, 2021-03-10:
Manual and scripts promoted to Raspberry_Release6; I will do a test myself as well.

MeanHat, 2021-03-10:
> @MeanHat thanks for the feedback. Unfortunately, github does not provide a neat WYSIWYG and version-managed wiki and therefore it's word/pdf.
>
> * SSH remark added
> * password remark already there
> * [...]

Great work @WouterJD - 

I've just realised that parameter -D-1 means no ANT dongle so:
1. that would need to be reflected in last text before 3.12.6 '(If you want to start-up with other parameters, you can modify FortiusAnt.sh. For example, the run headless remove “-g” and to use ANT remove “-b” **and -D-1**. (See section 4.2  for more information). Right-click and start editor:' and
2. add "-D-1" to section 4.2.

Also:
- I noticed a minor typo in the welcome text "For more info, please refer to the wiki on github. Succes!" - Succes! should read "Success!"
- As requested, I'll work on "2_InstallPackagesANT.sh" and "9_GrantAccessToUSB_withReboot" noting your comments

MeanHat, 2021-03-10:
> > I think the next step for me is to continue to figure out the install of BLE on RPi0 with armv6l
> 
> Is the message that FortiusAnt fully works on Rpi0, with exception of -g?
In summary for RPi0:
- **GUI**. Yes, that's correct. Someone smarter and more patient than me might be able to get the GUI working (when I have tried to install wxPython is takes almost a day (as pointed out by others) and ultimately failed). I will try again though.
- **ANT**. Works fine (one install running FortiusANT headless for several months)
- **BLE**. Not working yet as result of armv7l to armv6l conversions but still testing. I will let you know. I think this is more of a priority than the GUI as I suspect the main attraction of the RPi0 is the low cost and a display/monitor would add to the cost

WouterJD, 2021-03-10:
-D-1 is already in manual, I think.
succes modified in code and (next) manual

- GUI; evidently there is an issue with Raspberry/wxPython; I'm not particularly happy with the OndeDrive download of a package; wxPython should support Raspberry in a better fashion.
- ANT is nice to have document, since there is some reason to use it
- BLE/rpi0 is nice indeed since BLE is natively available

MeanHat, 2021-03-10:
@WouterJD - here is an early draft for adding ANT dongle to install 9. 
[DRAFT_forANTdongle-9_GrantAccessToUSB_withReboot.sh.docx](https://github.com/WouterJD/FortiusANT/files/6115843/DRAFT_forANTdongle-9_GrantAccessToUSB_withReboot.sh.docx) along with comments

WouterJD, 2021-03-10:
@MeanHat this evening I will do a test with ANT; enough dongles available :-)

When enhancing a Tacx Fortius with Raspberry Pi0 or Pi4, I would think the price-difference is not very important compared to the whole (Tacx + RPI + work), differently from building a commercial product. If you would build a "Tacx Fortius 2021 Zwift Ready", why would you chose for a Raspberry Rp0?

MeanHat, 2021-03-10:
@WouterJD - good news. The RPi0 is now working in Bluetooth mode! Here's a photo of the output from simulation mode:

[image]

I'll test it with a trainer and let you know. Would you like me to send you the list of instructions I used or would you prefer me to try to build them into your set up scripts?

MeanHat, 2021-03-10:
Probably main reasons for choosing RPi0 are lower cost, smaller form factor
and lower power consumption but not necessarily major issues

WouterJD, 2021-03-10:
@MeanHat I have reviewed the install scripts and made some minor modifications; [PiShare] is now also automated and scripts can be re-run.

WouterJD, 2021-03-11:
> Here is an early draft for adding ANT dongle to install 9

@MeanHat please refer to `9_GrantAccessToUSB_withReboot.sh` and `Show USB devices.sh` in our branch.

MeanHat, 2021-03-11:
@WouterJD - thanks for links to additional scripts.
- I agree: good plan to rename with FortiusAnt prefix. (ant-usb2.conf in modprobe.d did not exist but I prefer your FortiusAnt prefix)

I'm about to test RPi0 setup again using your latest scripts.
Suggestions/questions I had at this stage are:
1. Should we accept that it is recommended to run RPi0 headless without GUI because installing wxPython is time-consuming and complex (though the ideal solution is that someone creates a pre-built armv6l wxPython package like the one from @wtal710174 that you use in your script)? If so, install script 4 can be skipped for RPi0.
2. Also, to avoid wxPython I removed (commented out) wxPython from "requirements.txt" to avoid receiving a large number of error messages.
3. In script 3 I changed the source for RPi0 (with armv6l rather than armv7l) to https://unofficial-builds.nodejs.org/download/release/v14.15.3/node-v14.15.3-linux-armv6l.tar.xz. This worked OK.Would it be possible to detect the RPi architecture and then select appropriate download (perhaps using "uname -m")?
4.

WouterJD, 2021-03-11:
1. Manual says "If you want to use the GUI, a modified wxPython must be installed. Otherwise, you may skip this step. Skip for Rpi0.
2. Perhaps deploy a requirementsRpi0.txt
3. Tell me what uname -m gives on Rpi0

MeanHat, 2021-03-11:
1. Sounds good
2. That would work well but requires an additional requirements.txt file (if you're happy having an extra file)
3. uname -m gives armv6l on RPi0W and armv7l on RPi3B+. I don't have an RPi4 to check

MeanHat, 2021-03-11:
@WouterJD - I have tested the scripts on **RPi0**. The script numbers refer to the numbered list below. General comments are (a) I miss the relative speed of an RPi3 or 4! (b) now seeing "Executive in Terminal" option when double clicking script from file manager (can't explain this but positive news) and (c) love your scripts: makes install significantly easier.
Scripts:
1. OK
2. OK
3. I modified script as follows: (a) line 5 changed NodeJS=...armv7l... to ...armv6l... (b) line 14 changed wget target to: https://unofficial-builds.nodejs.org/download/release/v14.15.3/
Installed OK and by using the same 14.15.3 version it resulted in same node -v and npm -v results as expected
4. X skipped
5. Modified requirements.txt file to removed (commented out) wxPython. Installed OK and avoided the multiple errors
(reported PATH errors related to cli_exit-tools, lib-programname etc. where recommended to add home/pi/.local/bin to PATH
6. OK
7. OK
8. X skipped
9. 90 OK; 9 OK
10. (a) start up on reboot works OK and launches FortiusAnt (b) modified /raspberry/FortiusAnt.sh parameters to -a -b -s -d127. **_Mission success!_** I'll try with actual trainer next

WouterJD, 2021-03-11:
Scripts 3 and 5 updated for Rpi0

WouterJD, 2021-03-11:
I did a complete reinstall and ANT works immediatly!
Great collaboration!

[image]

WouterJD, 2021-03-11:
Bad experience: I (mistakenly) put a BLEdongle in the raspberry. Some special things occurred (flickering screen etc).
The BLE-dongle was reported by the lsusb command.

BUT: After removal, BLE was dead and I decided to reinstall the system; which provided another end:end test, which takes 15 minutes from first boot.

MeanHat, 2021-03-11:
Great work! 15 minutes to completely reinstall is super fast and a further confidence boost. All looking great!
Is there anything else you would like a hand with?

MeanHat, 2021-03-11:
> 
> 
> Scripts 3 and 5 updated for Rpi0

The 2 new scripts look like a great solution. I'll happily test them tomorrow on RPi0 :)

WouterJD, 2021-03-11:
I have updated all scripts
They are all rerunnable and fit for Rpi0; of course warning that wxPython is not available

I have run the scripts and from a clean disk it ends with FortiusAnt running :-)

This is good for USB-connected Tacx trainers and BLE-communication.
When ANT-communication is desired, some modifications are required.
Will check manual for that

WouterJD, 2021-03-11:
Made some (final) updates to the manual. @MeanHat I would appreciate your corrections, being native English speaker. Would you mind cheking?
[FortiusANT User Manual 6.0.c.zip](https://github.com/WouterJD/FortiusANT/files/6125619/FortiusANT.User.Manual.6.0.c.zip)

MeanHat, 2021-03-11:
Of course. I'll take a look and get back to you with any comments. Great work.

MeanHat, 2021-03-12:
@WouterJD - your Manual is a really impressive and fabulous achievement. And your English is really excellent. I have gone through the entire manual and made some suggestions for your consideration. In some cases I have checked for errors and in others I have suggested text which may make it easier to understand some of the complexities. I have tried to follow Plain English guidelines (https://plainenglish.co.uk/) as these have been found to be a good basis for English language manuals used internationally. I hope the suggestions are useful but please feel free to disregard any or all of them.

[FortiusANT User Manual 6.0c - suggested edits.zip](https://github.com/WouterJD/FortiusANT/files/6130564/FortiusANT.User.Manual.6.0c.-.suggested.edits.zip)

MeanHat, 2021-03-12:
@WouterJD - in FortiusAntMain.sh all old the logfiles are deleted. I wondered if you might also want to empty the WasteBasket to prevent the build-up of a large number of files over time. In the python program for the minTFT display I set a number of files I wanted to retain (in case I needed to to any retrospective troubleshooting) and then deleted all the older ones in '/home/pi/.local/share/Trash/info/*.*' and '/home/pi/.local/share/Trash/files/*.*'

WouterJD, 2021-03-12:
> @WouterJD - your Manual is a really impressive and fabulous achievement. And your English is really excellent. I have gone through the entire manual and made some suggestions for your consideration. [...]

Wauw Keith, what a great job you did; thanks for compliments. I revised the manual and adopted 99% of the remarks. Attached is 6.0.d. Section 5.1 is rewritten, remark on GUIlogfile added. 
[FortiusANT User Manual 6.0.d.zip](https://github.com/WouterJD/FortiusANT/files/6132644/FortiusANT.User.Manual.6.0.d.zip)

WouterJD, 2021-03-12:
> In FortiusAntMain.sh all old the logfiles are deleted

Instead of 
`rm *.log *.json *.tcx 2>/dev/null`
use
`find ./ -name '*.log'  -type f -mtime -30 -delete`
`find ./ -name '*.json' -type f -mtime -30 -delete`
`find ./ -name '*.tcx'  -type f -mtime -30 -delete`
?

Something similar for the trashcan?

MeanHat, 2021-03-13:
> > In FortiusAntMain.sh all old the logfiles are deleted
> 
> Instead of
> `rm *.log *.json *.tcx 2>/dev/null`
> use
> `find ./ -name '*.log' -type f -mtime +30 -delete`
> `find ./ -name '*.json' -type f -mtime +30 -delete`
> `find ./ -name '*.tcx' -type f -mtime =30 -delete`
> ?
> 
> Something similar for the trashcan?
> 
> PS. trascan directories are created on first delete :-)

I think that would work really well and 30 days sounds appropriate to retain the log etc. files.
Regarding the Wastebasket, if the most recent log etc. files are kept in the original folder then the whole wastebasket can be emptied? I understand it is necessary to clear out both '/home/pi/.local/share/Trash/info/.' and '/home/pi/.local/share/Trash/files/.' folders. Hope that works OK.

MeanHat, 2021-03-13:
> Wauw Keith, what a great job you did; thanks for compliments. I revised the manual and adopted 99% of the remarks. Attached is 6.0.d. Section 5.1 is rewritten, remark on GUIlogfile added. [...]

@WouterJD - your manual looks fabulous. It's a massive piece of work and really comprehensive and (I think) very easy to read and use. Hopefully it will inspire even more people to use your FortiusAnt system with the confidence to install it. The range of installation options is huge ranging from top end laptop and graphics to the affordable RPi. You've kept so many old Tacx trainers out of landfill and hopefully many more to come (let alone kept so many more people fit during Covid)!

WouterJD, 2021-03-13:
OK. So manual ready; last updates for FortiusAntMain.sh and then Raspberry_Release6 will be published

WouterJD, 2021-03-13:
...
```
echo Trashcan emptied (keep 1 month)
find /home/pi/.local/share/Trash/ -name '*.log*'  -type f -mtime +30 -delete
find /home/pi/.local/share/Trash/ -name '*.json*' -type f -mtime +30 -delete
find /home/pi/.local/share/Trash/ -name '*.tcx*'  -type f -mtime +30 -delete

echo Log-files from previous session are deleted (keep two days)
find ./ -name '*.log'  -type f -mtime +2 -delete
find ./ -name '*.json' -type f -mtime +2 -delete
find ./ -name '*.tcx'  -type f -mtime +2 -delete

```
...

??

WouterJD, 2021-03-13:
Pull Request now created, ready for merge.

Tonite I bought another Tacx Fortius and got it together with the Tacx Catalogue from 2012. This catalogue provided a great insight in the 2012 product offering, the types of brakes and their power curves - I have added it to the manual. Pieces fell together, check the new sections!

MeanHat, 2021-03-14:
Sounds great. The additional catalogue information in the manual is very helpful. I've shared some additional catalogue information in case it is of use regarding models and features. Probably not a huge help but I thought you might want it if you have not already seen it.
[IMG_20210314_0001.pdf](https://github.com/WouterJD/FortiusANT/files/6136445/IMG_20210314_0001.pdf)

jujroy, 2021-03-15:
Hello gang,

I am a Windows 10 user but being curious, I decided to test Release 6 on my Fortius.  I noticed that the power is now displayed on the Dash Board during the calibration process.  Love it.  After warm up and stabilisation, the resistance was around 1300 and power displayed 58 Watts.  During my workout, I noticed a a few warnings on the DOS page :  **Dongle.read : x characters skipped** (X being between 4 and 10 chararcters).  With version 5.2, my Pedal Stroke Analysis is always fairly circular.  With Version 6, often the bottom half of the circle was choppy, not as smooth as V5.2.  The overall feeling was the same as V5.2.  Thanks again for the great work.

WouterJD, 2021-03-15:
The dongle read errors are a nasty issue; the ANTdongle has to warm-up. It seems old data from previous sessions. Suggestions welcome. 

Pedal Stroke Analysis has been improved. There were some issues in the code, should be better now;  meaning "according the data received".
At the time of writing it was argued whether this was possible at all, since the USB-device must be read at a high frequency (20ms) to have enough samples. It is the main reason that multi-processing is implemented (also to get experience with it in python). It's about the max the device can do. Perhaps it's also dependant on the (quality of the) trainer (and cabling); one trainer may be better than the other.

Try to pedal smoothly and see whether it works. Try to ride irregular and check effect; bouncy ride or power left/no power right. On my Tacx Fortius the curve follows my way of cycling, curious how your experiences are.

MeanHat, 2021-03-16:
@WouterJD I ran a further test on RPi0 with the latest release (fixing T1902) using scripts: 1, 2, 3, 5, 6, 7 and 9. I found the following:

1. 3_InstallNodeJs.sh (and 4) left the terminal window in ~/Downloads rather than Home. Probably not a problem.
2. There was no requirementsNoGUI.txt file in ~/FortiusANT/pythoncode. I created one without wx in it
3. When I ran from reboot there was error reported from /raspberry/FortiusAntMain.sh in line 6 relating to logfile removal: unexpected token `('.  Should it read '('? I commented out the lines relating to log file and wastebasket deletion
4. On next reboot error reported in FortiusAntCommand.py line 63 with import wx error. I tried commenting out lines 63 and 64 but this resulted in error importing wx in FortiusAntBody.py line 212. I also tried changing Constants.py with GUI set from True to False but this produced different errors.

It seems that FA is trying to install/use wx. Does this help? Please let me know if you would like me to test further.

WouterJD, 2021-03-17:
Raspberry released today, so I think this one can be closed.
If there is any question, just let me know!

WouterJD, 2021-03-17:
1. Agree; I could save pwd and restore at end. Manuals says to execute from file-manager and then it's not relevant.
2. Should be there now https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/requirementsNoGUI.txt
3. I removed () because they are not within ''
4. Cannot reproduce; Check constants.py
```
UseGui              = True	    # Can be modified to force no-GUI
try:
    from wx import EVT_CLOSE    # Just checking presence
except:
    UseGui          = False  	# no wx, no GUI
```

I can run with `UseGui = False`; which should go automatic when no wx is there.
If wx is there, but faulty - then you will have to disable with `UseGui = False`

closed 2021-03-17 by WouterJD