WouterJD/FortiusANT #275: Porblem with TACX flow on OSX Guages dont work. BigSur 11.2

equdosBD, 2021-03-07:
Config is Tacx Flow mag Brake.
OSX BigSUR 11.2
Head is Flow Upgrade T1932 USB HEAD White & Blue
Notes: works in ZWIFT the tkinter guages dont work.
I will post again after trying on my win10  build mean time I am thinking this is an OSX issue as the TKINTER ui has white text on osx and is unreadable unless you click on the buttons which you cant see until you do this.

Question: Do I need to connect the TACX Cadence sensor (a jack plug) ? Seems to work in Zwift without it and Zwift can use an ANT+ Cad. Sensor anyway, so do I really need the old flow mag sensor ? HRM is configured directly with zwift. All dongles are ANT+ and recognized using lsusb as Dynastream. 
I have a dual dongle setup

I am running the python code via Pycharm and also directly from the command line using Python 3.92, note I was on 3.7 no change or bugs encountered with upgrade.

Other notes (general comment on file structure): some of the module/file names seem to coexist/conflict with standard python lib names. It may be prudent to change the name of eg. constants.py to FortiusConstants.py etc.

Great project!

equdosBD, 2021-03-08:
Full Log attached

[FortiusAnt.2021-03-08 10-55-37.log](https://github.com/WouterJD/FortiusANT/files/6098218/FortiusAnt.2021-03-08.10-55-37.log)

WouterJD, 2021-03-08:
Hi @equdosBD thanks for hints
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

equdosBD, 2021-03-08:
Why is this closed?  I have now tested on Windows 10. The hardware kit is listed in my first comments. On Windows 10 ALL works fine, so this appears to be a GUI issue for OSX.

equdosBD, 2021-03-08:
> Hi @equdosBD thanks for hints
> Welcome to the FortiusANT community
>
> I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
> Please tell me what [...]

equdosBD is a post grad software engineer and electronics engineer and hard core cyclist.
Currently location Sydney Australia, regularly in London, France and Netherlands.

This app is smooth on Windows and not so smooth on osx,...

equdosBD, 2021-03-08:
I am willing to help on this project. If the OSX side needs some work let me know.  I have questions about debugging this app in real time.

WouterJD, 2021-03-09:
> Why is this closed?

I don't know; perhaps you pushed the wrong button some days ago :-)

WouterJD, 2021-03-09:
FortiusAnt appears to be running well on Mac OS; I do not have a Mac so cannot judge.
Your idea's appreciated.

JoostBart, 2021-03-10:
To me the same type of issues are familiar, with Big Sur and Vortex T1960. Sometimes it works and sometimes connections seem to be lost, and gauges are frozen. When the Gauges are frozen in Swift, they are also frozen in the GUI of FortiusAnt. 
So I hypothesized that the connection was not stable. Is that possible? I am not a software engineer...

equdosBD, 2021-03-19:
I will let you know as soon as I have got RT debugging working. On big sur no gauge action whatsoever whilst everything
seems to be fine inside ZWIFT.

Personally, I don't need the gauges except for testing.

The app needs to be split into release and debug builds as it has a lot of crap in it that doesn't do anything
except help with debugging which is not reqd. in release builds.

The project seems to be targeting too many platforms at once and there are too many variants of old trainers as well.

 I have used it for a week solidly on a T1932 I/F with a TACX Flow , power is grossly over exaggerated.
On my last race I was warned (C grade) I was out of category because my power values were so high. That is not possible
as I am far from B grade and only at the bottom of C grade.

There appears to be no devops builds using the gitcode either.

All these issues will make it far more difficult to nail down bugs because the field is so wide fixing1 problem may create 10 others.

.....

As to your issue. The interface is USB to ANT+ and then ANT+ back to App eg. ZWIFT.

Most ANT+ dongles are rubbish and have very poor range and a really susceptible to interference from WIFI and particularly BT as
they use BTLP as the radio layer. It is possible you have issues with ANT+. A typical symptom of ANT+ radio issues are dropouts, sudden changes in speed and power readings and frozen power readings. Make sure you have no iphone or Apple tablet nearby with Bluetooth switched on.

WouterJD, 2021-03-19:
Hi @equdosBD I'm sorry to hear that the code is full of crap.
Yes you are right that there is no Devops environment.

I would suggest you simply remove this instable piece of software from your system and buy solid and tested Tacx systems.

I expect you will not mind that I close the issue.

closed 2021-03-19 by WouterJD