WouterJD/FortiusANT #276: FortiusAnt steps on the brake every 13/13 sec.

BikerVibor, 2021-03-08:
FortiusAnt briefly steps on the brake every 12 to 13 seconds. The speed also remains very low. Can someone indicate what I am doing wrong or what is going on? There is no speed when I am connected to use Rouvy or Zwift. I pedal drowsy and then go about 5 km / hour. See also the screenshot and the logfile.
What can i do about this?

[image]
[FortiusAnt.2021-03-08 23-10-55.log](https://github.com/WouterJD/FortiusANT/files/6104539/FortiusAnt.2021-03-08.23-10-55.log)

TacxBiker, 2021-03-09:
The same error comes from either no power or power switch off.
Also this problem can occur if the cadance sensor fails.

BikerVibor, 2021-03-09:
Thanks for the suggestions. But both work OK. 'Calibrate before start' is working. So the power is on.  The cadance given feels OK. 
With the adjustment of the wheel pressure I get to a maximum of 5.2 after roll time (7.0 would be nice according to the manual). That also seems pretty OK. Too bad, unfortunately I cannot get any further now.

WouterJD, 2021-03-09:
The logfiles shows the following

```
23:10:55,662: FortiusANT started
23:10:55,662:     Restart=False debug=15
23:10:55,662: -a
23:10:55,662: -A
23:10:55,662: -c False
23:10:55,662: -C 0/0
23:10:55,662: -d 15 (0b1111)
23:10:55,662: -D None
23:10:55,662: -g
23:10:55,663: -H 0
23:10:55,663: -n
23:10:55,663: -p 1.0
23:10:55,664: -P
23:10:55,664: -r
23:10:55,664: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
23:10:55,665: -t Motorbrake
23:10:55,665: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
23:10:55,665: -x
```
and

```
23:10:57,215: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=0 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:10:57,756: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=2150 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:10:59,939: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=0 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,168: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,170: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,200: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,323: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,354: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,384: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,415: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,446: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,477: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,601: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,631: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,662: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,692: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,723: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,753: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,877: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,907: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,938: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,968: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:00,999: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,031: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,155: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,187: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,218: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,248: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,279: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:01,310: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
[... cut ...]
23:11:54,716: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:54,748: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:54,779: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932
23:11:54,810: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=100 CurrentRes=0 CurrentPower=0, pe=0 hdr=0x21303 Connected to Tacx Trainer T1932

```

WouterJD, 2021-03-09:
Why did you specify the -r flag, setting the resistance to 100?
That settings is used to create a power-curve and usually not specified.

It seems you have specified many options, I do not know whether you actually intend to do so.
When checking that everything works, default options would be a good start.

I would suggest to remove the JSON file, and start with default settings -a -g and then see what happens.

BuitingDavid, 2021-03-16:
Same problem here.  Only just installed the FortiusAnt software.  Dashboard is showing correctly, only the power is continuously variating.
@WouterJD what do you exactly mean with "I would suggest to remove the JSON file, and start with default settings -a -g and then see what happens."  How is that done exactly?

BikerVibor, 2021-03-16:
I'll think that in the .JSON file the adjustments of the dashboard are stored. i have deleted this file and started with a clean desk.
I now keep kicking through it every 12 seconds. It is therefore dangerous to put pressure on or to stand on the pedals.
Too bad I was hopeful that this would work.
Ther is stil a fault in the log witch repeats everey 12 seconds.

'Ant dongle nog available' and then it reconnects.

But the communication with Zwift and Rouvy looks fine. Only teh speed is very very low. 

[image]

@WouterJD Do you have an suggestions?

BikerVibor, 2021-03-16:
Ik begin nu toch wel te denken dat het de USB stickjes zijn. Ik heb ze niet gekocht als CYCPLUS, maar Fortius ant herkent ze wel als CYCPLUS.  In de eigenschappen staat Dynastream Innovations. Aan de log te zien is het ook net of de dongles het niet kunnen bij sloffen.
Ik heb ze gekocht als zijnde van BOTAO.

WouterJD, 2021-03-17:
Zie #61 er zijn veel problemen met de CYCPLUS; ik stel voor dit issue te sluiten

closed 2021-03-17 by WouterJD

StevenHH91, 2021-03-25:
Ik heb dit probleem ook en ik denk dat het aan de cadans sensor ligt. Ik ben momenteel niet in de buurt van de opstelling, dus ik kan nu geen logs uploaden. Wat ik wel kan zeggen is dat ik de nieuwste versie draai op een Raspberry Pi 4gb en na kalibratie dat bij een testrit de cadans op 0 blijft staan, terwijl snelheid en vermogen wel reageren op inspanning (maar de snelheid klopt niet). 
De sensor heeft bij mijn weten jarenlang op de plank gelegen en is verder niks geks mee gebeurd, en het magneetje aan de crank is maar een magneetje (toch?). Is er een manier om te testen bij welke van de twee onderdelen het probleem zit? 

MVG
Steven

TacxBiker, 2021-03-26:
you can try to use a multimeter to test the cadence sensor. Unplug the sensor and connect (manualy) it to the multimeter and spin the crank. Set the multimeter on diode (with a beep) or resistance. The meter will beep when the magnet is near the sensor or the resistance drops significantly

StevenHH91, 2021-03-26:
Thank you for the advice! I tried to do so, but our multimeter is toast (rusty battery) so I guess I'll ask around if I can borrow someone else's. Or accept my defeat, take my losses and just cycle outside.

WouterJD, 2021-03-26:
In de volgende versie van FortiusANT is een "ledje" op de GUI die laat zien of de sensor werkt.
Geïnspireerd door de leds op de Raspberry. Nog even... door de test straat😉

StevenHH91, 2021-03-30:
Ik houd me aanbevolen!