WouterJD/FortiusANT #277: Raspberry works as Smartphone/TV interface

decodeais, 2021-03-11:
I wanted to test BLE on the RPi4, but this was only available on my smartphone. I needed a larger display. I found am old TV with HDMI input and connected it to my smartphone via RPi4.  This little thing is strong enough to manage FortiusANT and streaming at the sane time!!
I installed srcpy on it and it worked with very low latency. No app on the smartphone needed . You have only to enable the Developers options and connect it to the USB-port of the RPi4.
https://snapcraft.io/install/scrcpy/raspbian
I had to start the adb-server on the Raspi first.

`adb start-server`
`scrcpy --max-size 1440`

WouterJD, 2021-03-17:
I think I miss your point, please provide a picture and some more info.

MeanHat, 2021-03-17:
@decodeais - I too run FortiusAnt on a RPi linked via Bluetooth to a smartphone (Android - Google Pixel) running Zwift viewed on a TV. Rather than an HDMI connection I cast the screen using a Chromecast on the back of the TV.

BikeBeppe64, 2021-03-17:
> I wanted to test BLE on the RPi4, but this was only available on my smartphone. I needed a larger display. I found am old TV with HDMI input and connected it to my smartphone via RPi4.  This little thing is strong enough to manage FortiusANT and streaming at the sane time!!
> I installed srcpy on it and it worked with very low latency. No app on the smartphone needed . You have only to enable the Developers options (USB debuggin)and connect it to the USB-port of the RPi4.
> https://snapcraft.io/install/scrcpy/raspbian
> I had to start the adb-server on the Raspi first.
> 
> `adb start-server`
> `scrcpy --max-size 1440`

[image]

How did you get it? Thanks

decodeais, 2021-03-19:
Sorry I don't remember that problem. I think I took  beta version. So the first try worked.

WouterJD, 2021-04-29:
Streaming from smartphone to TV is beyond the scope of the FortiusAnt documentation; I draw a line somewhere :-)
Same holds for smartphone -> Raspberry -> TV

Thanks for sharing anyway

closed 2021-04-29 by WouterJD

MeanHat, 2021-04-29:
@BikeBeppe64 - I use a Raspberry Pi3 running FortiusANT connecting via Bluetooth to an Android smart phone running Zwift. This displays on a TV using a Chromecast attached the the HMDI input on the TV. Works perfectly. Hope this helps.