WouterJD/FortiusANT #278: TCX file stores miles instead of KM's

WouterJD, 2021-03-12:
Hi @WouterJD thanks for getting back and sorry for the confusion yes I know that the unit is meters.
I have a Garmin speed sensor also and I was comparing the speed while cycling between Garmin and FortiusANT both were the same but when I look in the tcx the speed and distance is much higher in the FortiusANT tcx. I am using UltraEdit to view files. 
Last data from Garmin DistanceMeters 8046.72  and Fortius  DistanceMeters 20349  
I think that it has convert 8 miles to 12 km and 12 miles to 20km not sure if I am right but the converison works. Thanks again for your work. I am enjoying indoor cycling since the weather is bad here at the minute.

_Originally posted by @gn00n in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-797323275_

gn00n, 2021-03-13:
Hi @WouterJD I attached todays data with log files if you want to see. Let me know if there is anyting else I can do.
[278.zip](https://github.com/WouterJD/FortiusANT/files/6135298/278.zip)

WouterJD, 2021-03-13:
Thanks @gn00n for posting the files. I do believe their content. Please explain the relation between the two files, what you expect and what you experience.

gn00n, 2021-03-14:
Hi @WouterJD FortiusANT.2021-03-13 17-15-11.tcx is from Fortius v5.2 and activity_6423922720.tcx is from my Garmin watch paired with and Garmin speed and cadence sensors. During the ride the screen displayed 20kph on Fortius and 13 mph on my garmin which is compariable. When I load the tcx from Fortius to Strava the speed is showen as 20mph and if I load the Garmin file it shows 13 mph which matches what I would have seen on both screen during the ride.

WouterJD, 2021-03-14:
Thanks, I will look into it shortly

WouterJD, 2021-03-17:
I owed you a check on this subject

The first trackpoint in FortiusAnt is;
```
					<Trackpoint>
						 <Time>2021-03-13T17:15:41.747086Z</Time>
						 <AltitudeMeters>0.0</AltitudeMeters>
						 <DistanceMeters>1.5920513178894502</DistanceMeters>
						 <HeartRateBpm>
							  <Value>0</Value>
						 </HeartRateBpm>
						 <Cadence>0</Cadence>
						 <Extensions>
							  <ns3:TPX>
									<ns3:Watts>6</ns3:Watts>
									<ns3:Speed>5.447265625</ns3:Speed>
							  </ns3:TPX>
						 </Extensions>
					</Trackpoint>
```

and the first in the Garmin is:
```
          <Trackpoint>
            <Time>2021-03-13T17:15:55.000Z</Time>
            <DistanceMeters>6.440000057220459</DistanceMeters>
            <HeartRateBpm>
              <Value>89</Value>
            </HeartRateBpm>
            <Cadence>48</Cadence>
            <Extensions>
              <ns3:TPX>
                <ns3:Speed>6.438000202178955</ns3:Speed>
              </ns3:TPX>
            </Extensions>
          </Trackpoint>
```

Here you see that both devices store (amongst others) : **Timestamp** and **Distance** (from start)

To calculate speed, each trackpoint must be compared with the previous where speed = delta-distance/delta-time.

If I put both TCX'es next to each other:

Time | \| | fSpeed (smoothed) | gSpeed (smoothed) | g/f | \|
-- | -- | -- | -- | -- | --
2021-03-13   17:15:56 | \| | 29,6 km/uur | 23,7 km/uur | 1,25 | \|
2021-03-13   17:15:58 | \| | 30,6 km/uur | 23,7 km/uur | 1,29 | \|
2021-03-13   17:15:58 | \| | 30,6 km/uur | 12,2 km/uur | 2,52 | \|
2021-03-13   17:15:59 | \| | 29,8 km/uur | 12,2 km/uur | 2,45 | \|
2021-03-13   17:15:59 | \| | 29,8 km/uur | 33,7 km/uur | 0,89 | \|
2021-03-13   17:16:00 | \| | 29,8 km/uur | 33,7 km/uur | 0,89 | \|
2021-03-13   17:16:00 | \| | 29,8 km/uur | 15,0 km/uur | 1,99 | \|
2021-03-13   17:16:01 | \| | 29,8 km/uur | 24,0 km/uur | 1,24 | \|
2021-03-13   17:16:02 | \| | 29,8 km/uur | 23,8 km/uur | 1,26 | \|

the ratio fluctuates, but there is no steady mile/km ratio between the two.
So I have no explanation for you right away.

**Sources**
The distance from FortiusAnt is calculated by taking the delta-time and speed.
Your Garmin takes the speed from the Speed Cadence Sensor.

I agree that both should result in approx the same values; rounding issues could be the most disturbing party here.
I am curious in your analysis, please find attached xls

[#278 TCX in miles.xlsx](https://github.com/WouterJD/FortiusANT/files/6159648/278.TCX.in.miles.xlsx)

WouterJD, 2021-03-17:
**Sources (2)**

FortiusAnt calculates the speed from the actual power
Garmin takes the speed from the wheel

This is the same approach as Zwift does: power / grade is calculated into speed

By definition, the speeds will be different.
The idea that it's miles/km put me on the wrong leg :-)

gn00n, 2021-03-18:
Thanks for explaining I understand now.

closed 2021-03-18 by WouterJD