WouterJD/FortiusANT #281: Details on FE-C implementation of Tacx Neo 2T appreciated

OevreFlataeker, 2021-03-14:
I am currently developing an emulator to make my "dumb" hometrainer "smart" - very similar to this project. I am already up and running by emulating FTMS, CPS, CSCS, ... towards Zwift and it works like a charm (== I have a working solution that I use for everyday training). Now I want to make my hometrainer also usable with the Tacx App. I understand Tacx uses a couple of proprietary BLE services like:

public static final UUID BOOT_CHARACTERISTIC = UUID.fromString("669aac01-0c08-969e-e211-86ad5062675f");
public static final UUID BOOT_SERVICE = UUID.fromString("669aa501-0c08-969e-e211-86ad5062675f");
public static final UUID FEC_BRAKE_SERVICE = UUID.fromString("6E40FEC1-B5A3-F393-E0A9-E50E24DCCA9E");
public static final UUID FEC_RX = UUID.fromString("6E40FEC2-B5A3-F393-E0A9-E50E24DCCA9E");
public static final UUID FEC_TX = UUID.fromString("6E40FEC3-B5A3-F393-E0A9-E50E24DCCA9E");

The thing is I managed to fully fake my emulator to be considered a Tacx Neo2T towards the app but the app is not interacting at all with my emu. I see it subscribed to notifications from FTMS but I see no I/O on the Nordic Uart Service which is the 6E40FEC1-B5A3-F393-E0A9-E50E24DCCA9E in the end. I expected the app to subscribe to the FEX_RX service and I would then be able to provide all the relevant ANT+ pages containing speed, power, .. data and receiving changed simulation parameters via the FEC_TX. But currently it's just sitting there like a "fat duck" and doing nothing. I am delivering FTMS measurements but those are not picked up via the Tacx App. So far I have not implemented 669aa501-0c08-969e-e211-86ad5062675f (BOOT_SERVICE) and couldn't get DFU implemented (00001530-1212-EFDE-1523-785FEABCD123). I am not sure whether those are relevant.

Can someone provide me a sniffed BLE dump of the interaction between the Tacx App and the trainer so I can better understand what the app is looking for and what it is doing after connecting to the trainer to get data flowing?

Also a simple Service/Characteristic dump is appreciated at that moment too, as I don't even have this. I have dumps from a Tacx Flow with the Flow brake service 669aa505-0c08-969e-e211-86ad5062675f and the sub characteristics 

public static final UUID ANT_C0_BURST_INPUT = UUID.fromString("669aab21-0c08-969e-e211-86ad5062675f");
public static final UUID ANT_CHANNEL0_INPUT = UUID.fromString("669aab01-0c08-969e-e211-86ad5062675f");
public static final UUID ANT_CHANNEL0_OUTPUT = UUID.fromString("669aab02-0c08-969e-e211-86ad5062675f");

but this does not seem to be needed by the Neo 2T.

As said: My emu is already working fine for Zwift emulating a FTMS trainer. I have all the calculations, I/O, ... working and am happy so far. But I'd like to go the extra-mile and implement FE-C as the FE-C implementation in Zwift seems to be better than with FTMS and I also want to use the Tacx App.

WouterJD, 2021-03-14:
@OevreFlataeker 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

> my "dumb" hometrainer

What trainer do you have?

OevreFlataeker, 2021-03-14:
Dear Wouter, well, as said it seems I tried to do something similar which you did on a greater scale. I developed a FTMS service emulator based on a Nordic nRF52 Development kit and interfaced it to my Aldi hometrainer ("Crane E.15")  with a Wahoo cadence sensor. The emu exposes CSCS, CSP and FTMS services for other clients, e.g. the Zwift app, i.e. I also connected my board to the PCB of the home trainer to trigger the resistance -/+ buttons electronically to turn the data from Zwift back into a change of resistance in the trainer. Now I'd also like to emulate FE-C for the Tacx app to be usable and the app recognizes my emu as a Neo 2T but doesn't try to read my data.

WouterJD, 2021-03-17:
Let's keep it open; perhaps somebody can share some data.
If not I will close the issue.

OevreFlataeker, 2021-03-17:
Thank you. I presume in your solution this topic is not relevant as you only offer FTMS to the apps not FE-C?

WouterJD, 2021-03-17:
We sure do: FE-C is ANT and FTMS is BLE. Both implemented.

OevreFlataeker, 2021-03-17:
Right, sorry, let me be more precise: 

public static final UUID FEC_BRAKE_SERVICE = UUID.fromString("6E40FEC1-B5A3-F393-E0A9-E50E24DCCA9E");

Do you implement the protocol as described in 
https://github.com/jedla22/BleTrainerControl/blob/master/How-to%20FE-C%20over%20BLE%20v1_0_0.pdf
?

As said above I offer this service to the apps but the Tacx app but the app is not picking up on it so I wonder what might be the reason. If you know what I am refering to and have info like "OK, after the initial service enumeration the app first does A and B, then subscribes to characteristic C and sends data D" this would be very helpful for me to troubleshoot. I basically see that the app recognizes my device as a proper trainer but it's not doing anything with it, i.e. not subscribing to e.g. the ...FEC2... characteristic and not sending anything to the ...FEC3... one.

WouterJD, 2021-03-17:
I recall something in this area. @marcoveeneman created the BLE-implementation.
We DO NOT implement FE-C over bluetooth, I believe for a good reason. This may well be why your FE-C is not recognized.

Check https://github.com/WouterJD/FortiusANT/blob/master/doc/bluetooth.md

OevreFlataeker, 2021-03-17:
As far as I heard iOS/iPadOS does not support ANT/ANT+, so the only way off communicating would be BLE with the various services like CSCS, CPS, FTMS or the FE-C over BLE. Zwift on i*OS works very nicely via FTMS but does not make use of the specific features of the Neo 2 like road feel etc. 

So I wonder: In the case of i*OS clients, do you offer your BLE FTMS to the outside (like I do) and speak ANT towards the device? Because those specific features of the Neo would then not be used? (I understand you would then forward/proxy the BLE FTMS messages from the outside to the trainer via BLE as well or - potentially - via ANT if possible, but you would never receive those special messages initially then from the client as long as the client doesn't see the FE-C over BLE services from you?

OevreFlataeker, 2021-03-17:
All my analysis is based on black box testing and observations. Well - actually not even black box, as I don't even have the "box" (the trainer). Even just a basic BLE service enumeration with all details of a Neo would help somewhat. A sniff/dump of the initial communication towards FE-C over BLE would be perfect so I could understand the interactions on that protocol better. FTMS is all set an done. I implemented everything and there is no open issue here.

WouterJD, 2021-03-17:
The manual should tell you quite some; please read.

ANT/USB to the device (depending on type) - ANT/BLE to the CTP

closed 2021-04-29 by WouterJD