WouterJD/FortiusANT #282: USB ANT not recognised

vincentvanheer, 2021-03-15:
Hello Wouter,

Thanks for your great job. Basic is now working for me.
I have a windows 10 pro fresh and fully up to date install  specific for FortiusAnt. My TacxFortius is recognised and my head unit T1932 has no issue.

But I unfortunately can't get my Bluetooth dongle recognised. I specifically bought two Asus BT-400. I tried leaving it as default and let it broadcast so I can use an app on a tablet or smartphone. No result.
I tried replacing the drivers with Zadig but same issue. FortiusAnt interface keeps saying "No (free) ant-dongle found"

Is there maybe something I could do to get those dongle recognised ?

Best regards ans thanks in advance for your precious help.
Vincent

WouterJD, 2021-03-15:
Hi @vincentvanheer 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

If you want to use BLE, specify -b on the command line.
FortiusAnt now looks for an ANT-dongle (and cannot find it) and does not use bluetooth (because it's not required).

Succes, like to hear from you

vincentvanheer, 2021-03-16:
Hi Wouter,

I didn't want to be too long on Github, so here is my answer.

I ride on a good old Tacx fortius T1941 with steerer frame and T1932 head unit.
I actually did receive the whole lot from someone who was about to thow it away as it was supposed to only work with Window 98 originally...

As far as my IT background is concerned, I'm a network and system engineer. I'm thus quite used to drivers, CLI and so on.
System used is installed on a HP Elitebook 820 G2 which runs very latest Windows 10 pro I specificaaly installed for this purpose

I tried specifying -b when launching Fortiusant but no better result... at least I still see the same message.
Here's what I use to get the trainer to work and get values displayed on my computer, launch calibration,..  FortiusAnt.exe -a -g -H0 -A

so interaction between computer and head unit is fine. I actually did my first training yesterday "manually"

I don't plan on using BLE specifically. Whatever can be used is good to know as long as I can use a CTP.

Thank you in advance
Best regards

Vincent

WouterJD, 2021-03-16:
You are confusing me:

- You bought two Bluetooth dongles
- You say that FortiusAnt does not recognize ANTdongle, that's obvious since there are no such dongles in your computer
- You start FortiusAnt without -b flag, so FortiusAnt will not look for Bluetooth

Since you have two BLE's, use them - start FortiusAnt with -b flag and do not forget to follow the instructions in the manual to install NodeJs!

Succes!

vincentvanheer, 2021-03-16:
I was confused by the message I got in FortiusAnt. I actually expected to at least see a message when connecting the BLE dongle... and not this message which led me to think something was wrong. My bad.

Should I start with -b flag only ?
NodeJs has been installed properly.

I'll try and start Zwift with both dongles in and keep my fingers crossed one link is enabled.

Do you maybe advise buying ANT dongles instead of those Asus BT-400 ? For the "price" of 0 euros I got the trainer, I can afford it ;)

Thanks for your informations

WouterJD, 2021-03-17:
The issue is that FortiusAnt does not connect to a BLEdongle, but calls NodeJs which connects to the dongle.
In fact, I do not even know whether there is a BLEdongle.

FOr the time being, stay with -b and the BLEdongles. It works fine.
I close for now,

closed 2021-03-17 by WouterJD

vincentvanheer, 2021-03-18:
Hi Wouter, problem indeed solved. I'm not very familiar with Bluetooth and I kind of forgot to wait for the magic to happen. I eventually saw a windows firewall popup asking whether I wanted to allow trafic for node.js on public and private networks. Do you see what I mean ? It might be worth telling it in the procedure. Seeing nothing for a moment might lead to think things are not working.
By the way thank you very much for your work and your precious help.
Oh and second "advise", wouldn't it be nice to advise people who have not yet bought their dongle to buy ANT+ instead of BLE ? I noticed yesterday that most CTP support ANT... not BLE. I wanted to give Goldencheetah a try as I'm more RPM/Indoor cycling training fan rather than roadie, but no luck. ;(

WouterJD, 2021-03-18:
Firewall; good addition.

ANT/BLE; an additional remark in the manual .
The market seems to move towards BLE; some CTP's support BLE before ANT in a new release.
BLE is cheaper
ANT is hard to get and the available dongles - from CYCPLUS - causing a lot of trouble.

Thanks for suggestions