WouterJD/FortiusANT #283: T1902 head unit identified as Anchor Chips AN2131 EZUSB Microcontroller

MeanHat, 2021-03-15:
Hi @fritz-hh @WouterJD @totalreverse @GewoonGijs - I hope you don't mind me mentioning you but I see that you have all been involved in getting a T1902 head unit to work with FortiusANT. I realise that firmware has to be loaded after every power-up but the head unit is being recognised as 0547:2131 (Anchor Chips, Inc. : An2131 EZUSB Microcontroller. It looks like the fxload routines at least require the head unit to be identified as a Tacx USB device. I'd like to get it to work with a T1901 trainer (which looks like a permitted combination from @totalreverse's excellent ttyT1941 document.
Have you seen this before and do you have any suggestions?

WouterJD, 2021-03-15:
The T1902 code is old antifier code, **I do not remember** I had any question about it.
As you see the code states @fritz-hh and @yegorvin, this is from the very early FortiusAnt days. (Thanks to comment, I did not recall).

Try to modify:
`         dev = usb.core.find(idVendor=idVendor_Tacx, idProduct=hu)`      # find trainer USB device
into
`         dev = usb.core.find(idVendor=0x0547, idProduct=0x2131 )`      # find trainer USB device
and see whether the device is found, later code can be cleaned-up

with the codes required and then catch at:
`            if **True or** hu == hu1902:`
and see whether code loads.

totalreverse, 2021-03-15:
I heard, that there are some rare models showing the default AN2131Vendor/Device ID. 
Maybe a very early model? Do you have a serial# on the backside? The first two decimals are typically the production year. 

Just try Wouters patch and let's see what happens.

WouterJD, 2021-03-16:
@totalreverse thanks for following and assisting where applicable👍

MeanHat, 2021-03-16:
@WouterJD and @totalreverse - thank you very much for the updates and helpful information. I'll test that shortly.

In the meantime, the serial number is 0715410 so 2007 in that case. Rest of product label reads I-MAGIC TRAINER T1902, P000308.

MeanHat, 2021-03-16:
@WouterJD and @totalreverse - now tested and works fine! Thanks so much for your help.
I changed one line (366) only in usbTrainer.py to:
`dev = usb.core.find(idVendor=0x0547, idProduct=0x2131 )`
and FortiusAnt identified the head unit as T1902, loaded the firmware and ran as normal (tested with BLE & Zwift).

It looks like this is a pretty rare head unit (I liked @totalreverse's "T1902 rare model" descriptor) - would you consider adding the Vendor and Product to a future code release?

Looks like we can probably close this issue?

WouterJD, 2021-03-16:
Aha this is a great and dirty trick - but it works
Since 1902 is the first in 
`        for hu in [hu1902, hu1904, hu1932, hu1942, hue6be_nfw]:`
it's recognized, of course no other type will work anymore

Rasbperry_release6 contains support for your trainer; please try

MeanHat, 2021-03-16:
> Aha this is a great and dirty trick - but it works
> Since 1902 is the first in
> ` for hu in [hu1902, hu1904, hu1932, hu1942, hue6be_nfw]:`
> it's recognized, of course no other type will work anymore
> 
> Rasbperry_release6 contains support for your trainer; please try

Will do @WouterJD

closed 2021-03-17 by WouterJD

MeanHat, 2021-03-17:
Many thanks @WouterJD @totalreverse

MeanHat, 2021-03-31:
Hi @totalreverse - you have written the most comprehensive Tacx trainer guide I have seen. Do you have any data on how many "Wheel On" magnetic brake and "Wheel On" motor brake trainers Tacx produced? @WouterJD and I are interested in understanding how many trainers there are out there.