WouterJD/FortiusANT #286: Issues with a Vortex...

RichBeardsley, 2021-03-18:
Hi All,

First of all, thanks for everyone involved with this, amazing to see the skills behind it in getting old kit working how we think Tacx should of done it, I was very frustrated a few years back when they suggested they couldn't update the i-Vortex to be smart or even broadcast the power over normal ANT+...clearly they could of. But I guess their tactics worked as I bought a Neo eventually :-/

Anyway, My wife is using the hand me downs now and I'm having a number of issues with my i-Vortex (T1961 & T1962 - latest firmware with 2.5 & 2.7 I think) - the data being collected by FortiusANT is very laggy or gets stuck with static values, or I get nothing at all.

I've installed the software on a Mac and also on a raspberryPi and I'm getting the same issues. Its probable that I'm doing the wrong things on both but at least I'm consistent.

The flags I'm using are basic starter values, just -a, -g and -t Vortex so not sure where I'm going wrong here. 

I have two ANT dongles and I've tried with them individually attached, and also with the head unit on/off but still the same problems.

If I use the trainer without any ANT+ connections then the head unit is displaying what you'd expect so it seems at least the brake unit is sending the correct data.

I'm also a little confused as to the target power, if I'm using Zwift then I assume it just tells the trainer a certain target gradient? or if in a workout then a target power? what relation to this is the 100W target you see in the GUI? is that relevant once Zwift takes over control?

Thanks
Richard in Nottingham, UK.

WouterJD, 2021-03-19:
Hi @RichBeardsley welcome in the FortiusAnt community.

100W is the start-up requested power from FortiusAnt and this means that no commands from Zwift are received.
Are you sure you paired in Zwift and with the correct device?

Specific pairing information is not in the manual because it would require to decribe it for all CTPs, although Zwift is (one of) the largests. FortiusAnt broadcasts a FE-C (Controllable Fitness Equipment) to pair with, and also a Power-meter and Speed-and-Cadence sensor which can be used to pair with a Garmin Edge.

Please check there first and let me know if further help is needed.

RichBeardsley, 2021-03-19:
Hi Wouter

Yes Zwift is pairing fine, but I'm trying to eliminate that by just using the FortiusANT GUI with no CTP attached, thats where  I'm getting the frozen power/cadence numbers. 

The two ANT+ dongles I have are both Dynostream, one is the original TACX device the other is one off Amazon if thats relevant?

Thanks for the help

WouterJD, 2021-03-22:
If you want to use without CTP attached, use -m or -M for manual mode.
See manual for more info

WouterJD, 2021-03-22:
ANT dongles are OK as long is no CYCPLUS is mentioned.
See #61

RichBeardsley, 2021-03-23:
Thanks Wouter,

So I'm assuming here that you've not seen the frozen or very slow to update numbers coming out of FortiusANT - I'll keep picking at it to see if I can get to the bottom of it.

Thanks again!

WouterJD, 2021-03-24:
FortiusANT broadcasts a FE-C device on ANT and this means that power/speed/cadence info is broadcasted every .25 seconds. To be exact: FortiusAnt sends it to the ANTdongle and believes it will be transmitted. Then Zwift must pick-up and -when aired- I have not heard about any issues there untill now.

Thusti, 2021-04-02:
Hi @RichBeardsley, I believe to have come across a very similar issue or most likely the exact same issue now that I just started using a Tacx Genius with FortiusAnt. I set up a RPI 4B with FortiusAnt which connects to the Genius through the Tacx ANT+ dongle and broadcasts through bluetooth for connecting with my smartphone or tablet.
I also had very laggy or stuck values that are not updating as they should which made it pretty useless. Sometimes I would get 20 seconds of stuck power values without any update at all. I installed on a Windows 10 computer and I had similar issues, although slightly better than the RPI.

I looked into this a little deeper through enabling the debug logging of FortiusAnt and I noticed that the data is actually correctly coming through the ANT+ dongle so I had the impression that there seemed to be an issue with parsing or decomposing the data.

While debugging the python code itself, I found out that there is an issue in the main ANT loop (Tacx2DongleSub).
Most of the time at line 1002 in FortiusAntBody.py (`TacxTrainer.Refresh(QuarterSecond, usbTrainer.modeResistance)`), The "trainer refresh" code actually flushes every incoming ANT+ data update message before reading in the actual ANT messages, every 0.25 seconds, later in the loop. In a lot of cases the data update messages are just not coming through.
I did a quick test by adding a sleep of 0.2 seconds after the trainer refresh to make sure there are new ANT data update messages available when actually parsing them and this fixed the whole stuck non updating values. However, this should probably not be the solution for this issue. I hope @WouterJD can help by looking into this because I don't really know what a good solution for this would be.

Nevertheless, thank you @WouterJD  and everyone involved for making this actually possible to still use my old Tacx Genius and make it "smart". 👍

RichBeardsley, 2021-04-04:
Hi @Thusti

thanks for that, I’ll try it...

RichBeardsley, 2021-04-05:
So following what @Thusti mentioned I’ve added a 0.25 second delay in to the loop and it indeed does now seem to work ok.

I’ll do some testing.

thanks @Thusti for looking in to it!

WouterJD, 2021-04-05:
I will look into it shortly

WouterJD, 2021-04-14:
Hi @Thusti sorry for responding late, but I've been busy with other developments.
Could you help me in referring WHERE the ANT-messages are flushed?

What I know is that only the first time messages are flushed, because they appear to be present from a previous session.

Tell me how the status is at the moment...

Thusti, 2021-04-14:
Hey, no worries. For the moment I have not set up my Fortius Genius anymore with the RPI 4 so I can't debug right now. 
However if I remember correctly, I had the impression that this line flushed the incoming messages each loop: https://github.com/WouterJD/FortiusANT/blob/38ac73ce0739077712bcd7944c2e56adb885f9ff/pythoncode/usbTrainer.py#L1948

WouterJD, 2021-04-15:
see antDongle.py
def Write(self, messages, receive=True, drop=True, flush=True):

I have changed the default-behaviour into flush=False, you are right: sometimes messages may be lost. Sorry for that.
It will be published shortly.

closed 2021-04-29 by WouterJD