WouterJD/FortiusANT #289: Problem connecting Fortius T1942 to FortiusAnt

yancart, 2021-03-23:
Hi, Ive now bought two genuine Garmin USB Ant+ dongles and I'm still having issues connecting to the Tacx T1942.

So my USB ports are setup as follows, does this look ok?
[image]

and when I start FortiusAnt I get the following:-
[image]

Is there anything that doesnt right?
Thanks

WouterJD, 2021-03-23:
Connecting the tacx is not related to ANT.
Did you follow instructions in manual?

yancart, 2021-03-24:
Yes followed the instructions.
Last week the Tax and CYCPLUS dongles were connecting but I was getting dropouts in the connection to Zwift.
I have now just changed the dongles to genuine Garmin ANT+ dongles, which as you say say shouldnt affect the Tacx connection as this is just a wired a USB link?
Assume the Fortius_Unint/Windriver1230 driver setup is correct?
Thanks

WouterJD, 2021-03-25:
Do you still get the same message "No Tacx USB trainer found"?
That is NOT related to the ANT+dongles.

I cannot verify whether the T1942 setup is correct because I do not have that headunit.

Recommended:
- Use a clean PC without Tacx software
- Follow manual 3.6.1	Windows – prepackaged drivers

FortiusAnt will detect the uninitialized headunit, load the software and then detect the T1942
See also 5.8.1	Unable to load firmware" on 1942 head unit

closed 2021-04-29 by WouterJD