WouterJD/FortiusANT #290: Tacx Genius connection

vincentvanheer, 2021-03-26:
Hi Wouter, I recently discovered your work recently and gave it a try on an old Fortius I received for free. This is working great, thanks a lot!! , and I therefore decided to search throughout the web for a second roller so I can train with a friend.
I eventually found a Genius T2021 brake with a T2018 ANT+ antenna.

Before doing anything, I was wondering what and if I'm supposed to buy anything or not. Can it work out of the box with Fortiusant without a second dongle or should I also have a second dongle like for my fortius ? In the case I were to need one more dongle, do I need specific tacx dongle or any ANT+ dongle will be able to communicate with the proprietary Tacx dongle ?
The aim is to run on Trainerroad or Golden Cheetah.

Best regards and thank you in advance

rcerrie, 2021-03-26:
> 
> 
> Hi Wouter, I recently discovered your work recently and gave it a try on an old Fortius I received for free. This is working great, thanks a lot!! , and I therefore decided to search throughout the web for a second roller so I can train with a friend.
> I eventually found a Genius T2021 brake with a T2018 ANT+ antenna.
> 
> Before doing anything, I was wondering what and if I'm supposed to buy anything or not. Can it work out of the box with Fortiusant without a second dongle or should I also have a second dongle like for my fortius ? In the case I were to need one more dongle, do I need specific tacx dongle or any ANT+ dongle will be able to communicate with the proprietary Tacx dongle ?
> The aim is to run on Trainerroad or Golden Cheetah.
> 
> Best regards and thank you in advance
> Vincent

I have had a Fortius for a while, which I sold to a friend. Got a Genius T2021 (without T2018 antenna) Initially I thought a single dongle will work since the Genius is Ant+ enabled, but I ended up using my same 2 dongle setup as on my Fortius without any problems. I am currently running Zwift. It was really a seamless transition from the one trainer to the other. Obviously you just have to select in the FortiusAnt menu the Genius model. I cannot see that there will be any issues running Trainer road or Golden Cheetah.
Jaco

vincentvanheer, 2021-03-26:
Dag Jaco,

Thanks for the infos, I'll buy two Ant+ dongles as my Fortius setup is with Bluetooth LE... so it's totally incompatible.

Have fun riding.

Best regards
Vincent

WouterJD, 2021-03-26:
Two additions:
- Avoid CYCPLUS see #61 
- You will have two tacxes; in the same or different locations? I never tried two tacxes on the same PC for example

vincentvanheer, 2021-03-26:
Thanks I did indeed read it several times so I didn't plan on buying Cycplus.

I'll indeed have two Tacxes in same room. One in Bluetooth LE, one in Ant+... but on different computers ! I'm a sys admin it shouldn't be too hard to find a second decent computer for a spot price.

I was wondering, could I use the t2018 dongle in combination with a noname dongle or should I best buy two dongles directly ?

Thanks for your efforts and advises, it's really appreciated !

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

WouterJD, 2021-03-29:
Two different dongles in same computer is no issue; for the advanced user even better cause you can address them separately

WouterJD, 2021-03-29:
Hi @vincentvanheer 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used!
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

vincentvanheer, 2021-03-30:
Great, I'll give this a try then ! Thanks ... and I saw your last mail, I won't present myself to the community a second time ;) I'm the one who received the Fortius recently and had it working with BLE dongles after having some issues connecting.

WouterJD, 2021-03-30:
> I won't present myself to the community a second time

Sorry for asking twice :-)

vincentvanheer, 2021-03-31:
No problem at all, your community is so wide I'm not surprised you can't remind of everyone...

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

closed 2021-04-29 by WouterJD

snitschcore, 2025-01-04:
May i ask for what i exactly need 2 ant+ dongles.
My genius got the antenna with it, so i just need one?
Both are used on the laptop or is one connected to the Trainer?
As i understood, i buy now 2 ant+ budget dongles and connect over fortiusant to my Laptop/Pc?

WouterJD, 2025-01-04:
@snitschcore Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what Genius bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

The [documentation ](https://github.com/WouterJD/FortiusANT/tree/master/doc)gives you detailed information.
Generally, on a PC two applications run: FortiusAnt and a CTP (Zwift, Trainer Road, ...) and hence you need two dongles; if the genius antenna functions as an ANT-dongle in the PC to communicate with the Genius itself you need one.

Read https://github.com/WouterJD/FortiusANT/issues/61 for more info on dongles.