WouterJD/FortiusANT #291: Connectivity issues RPi (BLE) and Zwift

kjoglum, 2021-03-29:
Using the latest FortiusANT code on a RPi 3b+ (built-in BLE) and Zwift on mac (without BLE dongle), I keep having issues related to "no signal" during device pairing. Often I need several reboots before my trainer gets connected successfully with signal.

I get the following log error from FortiusANT when unable to get connection/signal:

`17:46:57,534: ---------- BLE-devices are activated ----------
/home/pi/FortiusANT/node/fitness-machine-service/fitness-machine-control-point-characteristic.js:215
    this.indicate(response);
         ^

TypeError: this.indicate is not a function
    at FitnessMachineControlPointCharacteristic.onWriteRequest (/home/pi/FortiusANT/node/fitness-machine-service/fitness-machine-control-point-characteristic.js:215:10)
    at FitnessMachineControlPointCharacteristic.emit (events.js:315:20)
    at Gatt.handleWriteRequestOrCommand (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/gatt.js:922:26)
    at Gatt.handleRequest (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/gatt.js:344:23)
    at Gatt.onAclStreamData (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/gatt.js:274:8)
    at AclStream.emit (events.js:327:22)
    at AclStream.push (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/bindings.js:194:21)
    at Hci.emit (events.js:315:20)
    at Hci.onSocketData (/home/pi/FortiusANT/node/node_modules/bleno/lib/hci-socket/hci.js:595:14)
`

MeanHat, 2021-03-29:
Hi @kjoglum - in case it helps I loaded FortiusANT onto a RPI3B+ this morning and it operated fine with BLE on an Android device running Zwift. I also tested it with an RPi0 today (BLE and Android Zwift) on 2 different trainers - again without problems. I wonder if the issue could be mac related?

kjoglum, 2021-03-29:
For sure, it is likely related to mac connectivity. However, also raised as an issue here due to the FortiusANT error reported each time I am not able to connect/get signal.

ReinerMi, 2021-04-02:
I encountered a similar problem with a Raspberry Pi 3b: The raspi BLE Signal was not detected by an iMac (2019) neither an iPhone SE 2nd Gen.
The Raspberry Pi 3b supports Bluetooth LE 4.1, the Raspberry Pi 4 already BT 5.0. Maybe a Rasp 4 will talk to my iMac?

kjoglum, 2021-04-03:
> I encountered a similar problem with a Raspberry Pi 3b: The raspi BLE Signal was not detected by an iMac (2019) neither an iPhone SE 2nd Gen.
> The Raspberry Pi 3b supports Bluetooth LE 4.1, the Raspberry Pi 4 already BT 5.0. Maybe a Rasp 4 will talk to my iMac?

For me, a workaround is to use the Zwift Companion app on my iphone, having the Companion app set up in bridge mode. Then RPi 3b+/FortiusANT BLE is connected without problems using Zwift on my iMac.

ReinerMi, 2021-04-05:
Finally I managed my problems by re-installing the software on my RPi 3b! 
Now everything works fine with both bluetooth (on iMac and iPhone) as well with ANT+ via dongles. I left „-b“ in the settings while erasing „-D-1“ and now everything is fine and I am very happy...