WouterJD/FortiusANT #292: Genius Power in Erg mode

KillaBee2k5, 2021-03-31:
Hi everyone,

When I am doing a workout in Zwift in Erg mode, and I keep my cadence stable, my power is always about 10 watts over what is asked by Zwift.

My calibration is fresh so that doesn't seem to be the issue.

Is anyone else seeing this issue with the Genius or another trainer?

Is there any way to fix this with the settings?

Thanks,

simonperneel, 2021-04-01:
Hi,

I have the same issue as you described. My power is also always bit higher than the requested power in ERG mode.  I am also curious about a solution for this.

KillaBee2k5, 2021-04-02:
I looked at the code a little bit and the only thing that I could see that could be making it act differently in erg mode than sim mode is that the flywheel weight is set at 10kg in erg mode. The comments say that it's to make it more responsive. I tried putting the bike + rider weight as in sim mode, but it is indeed a lot less responsive. The watts seemed to be closer to the target watts though, but it took way too long to adjust during wattage changes.

simonperneel, 2021-04-12:
If it works, it works, I guess 🙂, but maybe it is not good practice. I also had a look into the code. The only thing I can think of why the target power is wrong, is because the GearboxReduction is not set properly for the Genius. Maybe it would work with just multiplying the TargetPowerProvided with 1, but I did not test it yet. https://github.com/WouterJD/FortiusANT/blob/7385cd6229e5a57276c687eb3083ccb14b780bad/pythoncode/usbTrainer.py#L481-L487

simonperneel, 2021-04-13:
I tried it today, it did not work out.

WouterJD, 2021-04-14:
Another simple explanation may be that the genius itself does not do exactly what is asked.

KillaBee2k5, 2021-04-14:
If that is the reason, then the -10 I put in the SetPower method for the Genius when in power mode seems to do the trick for me. I'll keep fiddling with the value,minus 10 might be a little aggressive. I'll try with -8 or -9, but even with -10, the number seems to be much closer to the number asked by Zwift, so I'm satisfied!

WouterJD, 2021-04-14:
The -p command line flag has the intention to adjust this; assuming the delta is a factor in stead of a constant.
10/100Watt would be -p 0.9 and you could experiment with -p 0.9 ... 1.1, depending whether plus/minus is required

KillaBee2k5, 2021-04-14:
I thought about that, but it doesn't seem to be a factor. It's pretty constant at 10 watts over whether it's asking for 100 or 200 watts. I will give it a try, but as I said, it did no seem to be a percentage of the target watts, that's why I went another direction.

Thanks for helping!

WouterJD, 2021-04-15:
I understand. Thanks for sharing.
I suggest we do not add another parameter and/or change the -p meaning.
Since your issue is now solved, can issue be closed?

KillaBee2k5, 2021-04-15:
Yes I will close it! Thanks a lot!

closed 2021-04-15 by KillaBee2k5

simonperneel, 2021-04-16:
Could you just tell on which line of code you  put the - 10 watts? I would like to try it also. Thanks!

KillaBee2k5, 2021-04-16:
https://github.com/WouterJD/FortiusANT/blob/38ac73ce0739077712bcd7944c2e56adb885f9ff/pythoncode/antDongle.py#L1931

This is where I reduce the Target value by 10.