WouterJD/FortiusANT #294: help BLUETOOTH communicatiom

mancio1977, 2021-04-06:
Hi my friend, thank You for this wonderfull project.

Here is Marco from Italia near Garda Lake.
I have Tacx trainer i-magic 1904
on windows 10 with usb ANT+
and FortiusANT seems to work well.

I set Zwift on my smatphone
with bluethoot connection on
but while FortiusANT is working
I cannot see anything
seeking by Zwift in the smartphone.

Any suggestion to make it working
only with smartphone?
Or have I to buy one more usb ANT+ for thr PC?

thanks in advance,
Marco

BikeBeppe64, 2021-04-06:
Ciao Marco, per usare FortiusAnt nel PC & Zwift nello smartphone, c'è bisogno di un'ulteriore ANT+ dongle da aggiungere alla porta USB di ricarica, a meno che tu non abbia un Sam***g, allora non ne hai bisogno in quanto ANT+ nativo (al 90% dei modelli).
Io faccio così per usare Rouvy con  uno Xiaomi 9 lite e non ho problemi

mancio1977, 2021-04-06:
Grazie per la risposta!!
Ho uno smartphone Sam***g, quindi come attivi ANT+ nativo?
per favore aiutami, ti ringrazio in anticipo

BikeBeppe64, 2021-04-06:
Cerca su Google " attivare ant+ Sam***g"

WouterJD, 2021-04-13:
Cari amici, sono contento che vi cavate assieme. Adesso occupato con nuovi sviluppi.
A presto al Garda Lake (previsto luglio)

closed 2021-04-26 by WouterJD