WouterJD/FortiusANT #295: Unable to communicate with Fortius Trainer

gilles-562, 2021-04-06:
I have the Fortius Antifier v5.0 on my PC and I am not able to communicate with my Tacx Fortius Trainer.  The trainer is connected by a USB cable.
Some idea to correct my problem ?

Thanks

BikeBeppe64, 2021-04-06:
I suggest to read the manual, many helps from it

gilles-562, 2021-04-07:
Thanks,
I installed the application but there was a bad file location.
Will the Zwift application fully control the Tacx Fortius Trainer ?
A friend of mine has the Tacx Vortex and he need to adjust manually the resistance on the Trainer.

WouterJD, 2021-04-14:
HI @GewoonGijs 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

If the USB-trainer is not find, either the Tacx USB driver is in the way and/or the libusb drivers are not correctly installed.
So first check that, refer to manual how.

What do you mean with "bad file location"?

And yes, Zwift will fully control the Tacx Fortius Trainer; no manual action required.
(Although you can use the head unit for virtual gear shifting)

FortiusAnt can also be used for your friend's Vortex.

gilles-562, 2021-04-21:
I didn't place the USB Driver in the good directory.  I changed that and it is OK now.
What do you mean by virtual gear shifting ?
Is it possible to use the mechanical gear shifting with Zwift and Rouvy applications ?

WouterJD, 2021-04-21:
Sure. BUt with virtual shifting you can maintain a high wheelspeed which is advantageous for some trainers.
I assume issue can be closed; if not please reopen

closed 2021-04-21 by WouterJD