WouterJD/FortiusANT #300: Sufferfest doesnt detect Ant+ device if started after FortiusAnt

jjmasc, 2021-04-15:
Hi every one.
I’ve received my ant+ Dongle and tested : Trainer Fortius + Fortius Ant on a laptop ( W10, 2 ANt dongle ) + SufferFest.

When i start in a logic order : power on fortius, AntFortius in automatic mode with  calibration (the shell window gives the id of two devices in broadcast mode : Power and FE-C). Then i launch  Sufferfest which does not detect any ANT device .
When i start in a reverse order : first Sufferfest : detection ANT+ OK, no active device ; Then  i launch FortiusAnt , SufferFest detects the 2 devices during calibration procedure.
If i close Sufferfest and reopen it, the detection is OK !!!
Just a little disturbing -). Any idea to correct that? 

Nevertheless after that everything seems OK : i tested 3 different workout, the ergmode and level mode. Virtual gearbox. Everything works fine. 

Sufferfest displays an other device:  SufferSpeed , that gives a more realistic speed and allows to ride with a high wheel speed on the trainer as recommended by @Wouter in doc file. I suppose that it is  that  speed which is saved on Strava. I’lll see tomorrow !!

WouterJD, 2021-04-22:
When I start Trainer Road and then FortiusAnt; Trainer Road detects the devices.
When I then stop Trainer Road and restart; Trainer Road detects the devices as well.

Different CTP's have different  pairing sequences, so I cannot say for sure.
Note that, only in operational mode (Start pressed, not calibrating); the FortiusAnt ANT+ loop is active.

Please let me know if I can help you with more information.

jjmasc, 2021-04-23:
HI Wouter
Is seems like FortiusAnt locked the dongle ANT+.
The error display in Sufferferst is " no Ant+" when starting FortiusAnt and then Sufferfest.
If starting first Sufferfest the display is "Ant+ OK", HR device is OK, other device :  "No device found"
Then starting FortiusAnt , Sufferfest detects the power and the Fe-C devices of FortiusAnt.

"Note that, only in operational mode (Start pressed, not calibrating); the FortiusAnt ANT+ loop is active."
        As i sayed in the first message, Sufferfest find the power devices two or three times during the calibration .

Just a litle difficulty. I am using FortiusAnt Two days a week for HIT sessions. Great job

WouterJD, 2021-04-23:
Each application needs it's own ANT dongle.

jjmasc, 2021-04-23:
Yes I know that.
This is why i don't understand why Sufferfest doesnt find ANT+ when run in second.
I will try another procedure
Insert usb and one ANT dongle, start FortiusAnt, wait for the endof calibration
Insert the second Ant Dongle and  start Sufferfest

WouterJD, 2021-04-26:
Does not ring a bell...
Verify that no other applications are running; Garmin Connect Express for example.

WouterJD, 2021-04-29:
See also #301

jjmasc, 2021-04-29:
OK Thanks. No other application is launched and TTS4 was never isntalled on the W10 laptop
I read the issue. 
It seems the last post is about the same pb (with TR)
In my case i never had an error message from fortius ant.

WouterJD, 2021-04-29:
Just tried (see #301) but have no idea at this moment

closed 2021-06-07 by WouterJD