WouterJD/FortiusANT #301: Hi Wouter

Timbertacxfortius, 2021-04-17:
Basic, basic, basic question. Sorry for, perhaps, overthinking!
I am so relieved to see that there is a way to rejuvenate my Tacx Fortius. 
I would really like to take on a Trainerroad account, but would love to be able to clarify the sequence that I “think” is correct!
I have a Tacx Fortius which I have used without major issues for the past 12 years.
It has a T1932 head unit, connected by a usb to a PC running Win XP (service pack 3). Never had problems, also not  connected to the internet.
I have a second PC with Win 10 installed. It sits right next to the older PC. But only one monitor.
I have 2x ANT+ dongles.
I use an AccuRate Hr strap which works fine with my TTS.
I also have a gen 6 iPad, if needs be.
So, does this procedure sound right?
First, download and install Fortius ANT+ onto my Win XP computer from GitHub, Quick start Windows executable version........
Then:
1) I install both dongles into my XP computer.
2) Start up my Fortius TTS on this computer, as is being done at present.
3) Ride for a while to warm up the trainer/bike using Fortius TTS and then calibrate.
4) Connect my second PC (Win 10) to a TR or Zwift, Sufferfest etc. account. This would  require a separate monitor so that I can see TR and Fortius?( Is it necessary to view both the Fortius TSS and the TR App. I am guessing not?) Start TR training programme which will control the resistance etc. and watch my training session on the monitor.

5) OR.....connect iPad to TR etc.
6) Then: my XP PC will talk to my Fortius through the USB connection, then one dongle will talk to the second dongle which will pass on that information to my iPad which is connected via App to TR. 
Does that sound right?
I just want to have this clear in my mind before I commit any scarce $$$$ to an online training programme.
Thank you in advance.
I’m in New Zealand, by the way.

WouterJD, 2021-04-18:
Hi Timbert - I presume.

Did you read the manual, this should answer the questions. 
Refer to 2. Background information
Refer to 3.2.4. Hardware, option 2. Two computers.

Summarized
1. Remove TTS and USB drivers from the PC where you start FortiusAnt
2. Windows XP seems to work (I hear), manual is Windows 10 based so there may be differences
3. One computer (Windows XP) - one ANT dongle - FortiusAnt
4. Other computer (one monitor) - one ANT dongle - Trainer Road

FortiusAnt uses one dongle - in your case to transmit to a CTP, e.g. Trainer Road.
If your want to use your iPad, that iPad must have ANT capabilities to receive info from FortiusAnt

Succes!

Timbertacxfortius, 2021-04-18:
Thank you! I’ll let you know how I get on in a few weeks.
Cheers
Timber

MeanHat, 2021-04-18:
Hi @Timbertacxfortius - further to @WouterJD 's comments I thought I would let you know that I could not get XP to work. There was a problem with Python or one of the other key requirements (I can't remember exactly from last year) as the latest version(s) would not work with XP and I had to use an earlier version. I resorted to running FortiusANT and CTP on a single Win10 machine and then a Raspberry Pi for FortiusANT and Won 10 for the CTP.
Hopefully you will have more success.
Cheers MH

Timbertacxfortius, 2021-04-18:
Thank you again, Wouter. I know that you must be a busy man!
One thing that I may be misunderstanding.
I assumed that my Tacx Fortius software had to be installed on the same
computer as FortiusAnt. This I can’t do as it is incompatible with the Win
10 on my 2nd machine.
So I can download FortiusAnt onto this 2nd Win 10 machine without TSS or
any Tacx software? And then use the two Ant+ usb sticks?
If that is the case, then problem solved!
Thanks again
Timber

oldgearian, 2021-04-20:
Timber
Please note that like you I am learning my way through FortiusAnt and am not an IT expert. I am sure you will receive reply from Wouter or others with much more experience. These are my findings from reading up as much as I can and from my own experiences.

I now also have the Tacx Fortius Motor Brake with T1932, on windows XP stand alone machine, no internet. I started running this on the Tacx Fortius software as you have probably been doing mostly using the Catalyst program from within it. I liked the Catalyst programme as it was easy to adjust slope/resistance through the headset. (Headset also accepted my polar HR)

Initially I used to run the Tacx Fortius software on XP laptop and using a separate windows 10 laptop connected to Zwift using speed and cadence sensors and my polar HR. This was fine and allowed me to have both software programmes running and visible on two monitors. However it was all wheel speed based, no inclines felt and certainly not controlled by Zwift but allowed me to realise I was happy cycling indoors and on Zwift.

I then purchased my two ant+dongles and downloaded FortiusAnt onto the Windows 10 laptop. My initial thoughts were, I think like yours, that somehow FortiusAnt had to be on the same laptop as the Tacx Fortius software. It does not have to be and I think from what I had read most advice was not to have both software systems running on the same pc due to driver issues?

My own advice, for what it is worth, would be not to do anything to your windows XP machine. You have TTS running there very happily and, I think, it is best to ensure you maintain that machine as it is and can always fall back to using it whenever you like.

download FortiusAnt onto windows 10 machine. (As you say you only have one monitor so not really much point trying to run across two machines but only see one screen)
Open FortiusAnt and Zwift etc (I have not used TR or sufferfest yet so can’t comment on how they work)
Windows 10 allows you to open 2 windows within one screen so I tend to run FortiusAnt on one half of screen and Zwift on the other half. This way I can see what gearing I am using within FortiusAnt.

WOUTER or others- is it possible that FortiusAnt display screen can be created to take on the format of a whole screen which can then be maximised, minimised or moved and made to share the screen with another programme?

TIMBER and any other Fortius Motor Brake users - please could you let me know the size of the surge protection fuse you have in your PowerBack unit. It came to me secondhand with 500MA fuse but I blew 5 quite quickly before upping to 1.5amp which seems ok but worried I may do damage to electronics if this is just too high.

Timbertacxfortius, 2021-04-20:
Thank you so much for your message. Yes, I can see that my basic
misunderstanding was that FortiusAnt had to run through the Tacx software.
Now that I realise that it doesn’t I’ll take  your advice and continue with
the plan to download FortiusAnt onto my windows 10 computer and leave my XP
computer alone, for those times I may want to revert to TTS. Thanks for the
split screen advice. I don’t think I have a seperate power surge protection
on my Fortius powerpack, is there an internal unit?
Cheers,
Timber

WouterJD, 2021-04-21:
> WOUTER or others- is it possible that FortiusAnt display screen can be created to take on the format of a whole screen which can then be maximised, minimised or moved and made to share the screen with another programme?

Whole screen / resize: this would require to dynamically resize all controls on the screen, which was not implemented untill now.
The size of the screen = the size of the FortiusAnt.jpg in the pythoncode folder; 900x600. Try to resize, see whether a smaller screen remains workable and let me know.
Minimise is enabled.
You can also run in console mode (remove -g from the command line) and then you have a minimal screen.

WouterJD, 2021-04-21:
> My initial thoughts were, I think like yours

Perhaps the manual has grown too big; please provide feedback where the description can be improved

WouterJD, 2021-04-21:
> Fuse ... before upping to 1.5amp

Since a motorbrake can work to 1000Watt, 500mA seems pretty low.

WouterJD, 2021-04-22:
PS. I have done a quick test on resizing.
This would mean a lot of work, not only resizing the window but also all controls including the font-size.
It is technically possible, but would result in unreadable controls.
This suggestion will be left for future improvement.

Option is to use a raspberry pi0 with a TFT screen, it frees your computer screen.

MeanHat, 2021-04-22:
Hi @Timbertacxfortius and @oldgearian - I can definitely recommend the Raspberry Pi Zero solution with a mini TFT screen. It's a great, very stable solution.

Timbertacxfortius, 2021-04-23:
Hi Wouter.

I’m still having issues with starting FortiusANT!

I managed to download FortiusANT.exe, installed and verified 2USB ANT+
dongles . (Dynastream Innovations).They show up as active/working in Device
Manager, as does the Tacx 1932 device. I then opened FortiusANT.exe from
File Explorer. 22,536Kb.

The .exe and GUI opened with the FortiusANT GUI saying :

Connected to Tacx Trainer T1932

No (free) ANT-dongle found.

Heart rate expected from Tacx trainer.

Pic #1 zip shows the .exe screen at this point.

Locate HW button does not seem to activate. Start etc greyed out.

I turned off “auto HW” in Settings and the GUI showed:

Tacx trainer

ANT+ dongle

ANT+ Heart rate monitor.

I then clicked Locate HW.......then......

Fortius Antifier v6.1 (not responding)

Pic#2 zip shows the command screen at this point.

Are you able to make any suggestions  about why this is not working?

Also tried to download/open manual.doc. (FortiusANTUserManual 6.0)

This sent me to page “showing 1 changed file”

Was unable to open anything here.

Thanks for your efforts!

Cheers
Timber NZ

WouterJD, 2021-04-26:
HI @Timbertacxfortius you write a lot of text.
Sorry for not remembering the whole trail...

I expect you use Windows 10.
What other ANT-software is running when you start FortiusAnt; mind software that remains running, such as Garmin Connect Express.
How did you install the libusb?

If no ANTdongle is available, either libusb is not installed or other software grabs the dongles.

Timbertacxfortius, 2021-04-27:
FortiusANT v6.1

Open FortiusANT.

Pedal kick to start. Powers wheel at 20kph for about 90 secs

When complete, Power display 100w this i can adjust down to 31w with the
head unit

If I ride, all the dials activate ok

In Device Manager, Tacx T1932 and the two ANT USB-m sticks are working
properly.

Open TR app.

In TR:  The Instruction is to Pair Devices. “ No devices Found”. “ANT+
Off”. No wireless signal detected.

If I turn off FortiusANT, then TR app displays “ANT+ ON.

If I launch FortiusANT then, the command screen displays “failed to execute
script FortiusANT”.  The GUI does not respond.

It appears that both apps cannot be on at the same time.

WouterJD, 2021-04-29:
Strange, first time I hear this.
It's just as if Trainer Road wants to use one dongle and not the other.

You can force FortiusAnt to use one specific dongle with the -D command line flag, specifying the deviceID of one dongle.

First try:
- put one dongle in the system; try Trainer Road
- put other dongle in the system; try Trainer Road
Does TR work in both cases?

WouterJD, 2021-04-29:
I have tried:
- run FortiusAnt
- install TrainerRoad, add ANT-dongle and check:

I cannot reproduce your problem.

[image]

planetSammy, 2021-05-28:
Hello, i have been using the original antifier software for a few years. First with i-magic brake T1901 and head unit T1904(green unit). After some connection problems (cannot read from Trainer) i changed the head unit to T1932. 
I ride on RGT-Cycling and after swapping the head unit, i had the feeling that the configuration especially on downhill is wrong.
Yesterday, first time using your FortiusAnt Software with i-magic T1901 brake and T1932 (Blue) head unit. After calibration everything works fine...but i had the feeling that is rather too easy and the software displayed too much Watt...i asked someone on RGT-Cycling to tell me his current Watt on a climb and it looked quite similar. 
Now my final question, can i use this current configuration or must i change the brake unit? I have the i-Flow brake in my Garage, also a T1901 Model. Anything else to Note in your Software?

Best regards Sammy

WouterJD, 2021-06-07:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

> Now my final question, can i use this current configuration or must i change the brake unit? I have the i-Flow brake in my Garage, also a T1901 Model. Anything else to Note in your Software?

T1901 (magnetic brake) is supported; with both headunits. See manual for info.
Also the power curve is extensively described.
If you have more questions, please raise an issue with a clear description and attach logfiles.

closed 2021-06-07 by WouterJD

WouterJD, 2021-06-07:
In addition: antifier is the base for FortiusAnt, but FortiusAnt works differently.

There are two headunits: T1901 and T1931
There are two brakes: Magnetic and Motor

There are three modes:
a- T1901 + Magnetic
b- T1932 + Magnetic
c- T1932 + Motor

Note that a and b operate very different, which may certainly affect the feel.
I do not have a T1901 headunit, not a magnetic brake; all implementations are done based on participants measurements and guidance.