WouterJD/FortiusANT #302: npm node installation Problem

TheBlacksmith86, 2021-04-24:
Hello everyone,
I just started the adventure to get everything running - but what an advernture it was so far.
I run an iFlow with a T1932 head unit (white/blue one) and it connects fine to FortiusAnt.
But the command line shows:
HTTPConnectionPool(host='localhost', port=9999
all the time. 
I saw that there was a bunch of errors while installing npm with bleno stuff.

Anyone able to help? I can attach the logfile if needed.

WouterJD, 2021-04-26:
Hi @TheBlacksmith86 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

The error you report indeed indicates that NodeJs is not running.
There is no hard link between FortiusAnt and NodeJs; so from FortiusAnt no more explicit error-messages can be given.

You say that "there was a bunch of errors while installing npm"; so that's where we should start.

TheBlacksmith86, 2021-04-26:
Hey. Quick introduction here too. Sorry for the rush - reread my post and it seems kinda rude.
So my name is Mathias and I'm from Nürnberg Germany. I have an IFlow T 1932 unit, which I have bougth used.
I came along your nice project here, because I already spend a week figuring out how to unlock the TTS Software.
Nice job!

So the last few lines of the node installation log are - since in my opinion the interesting stuff happens here.

5454 warn optional SKIPPING OPTIONAL DEPENDENCY: xpc-connect@^2.0.0 (node_modules\@abandonware\bleno\node_modules\xpc-connect):
5455 warn notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connect@2.0.0: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
5456 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Valid OS:    darwin
5456 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Valid Arch:  any
5456 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Actual OS:   win32
5456 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Actual Arch: x64
5457 warn fortiusant@0.0.1 No repository field.
5458 warn optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@0.5.3-7 (node_modules\@abandonware\bluetooth-hci-socket):
5459 warn optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@0.5.3-7 install: `node-pre-gyp install --fallback-to-build`
5459 warn optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
5460 verbose optional SKIPPING OPTIONAL DEPENDENCY:
5460 verbose optional Failed at the @abandonware/bluetooth-hci-socket@0.5.3-7 install script.
5460 verbose optional SKIPPING OPTIONAL DEPENDENCY: This is probably not a problem with npm. There is likely additional logging output above.
5461 verbose stack Error: @abandonware/bleno@0.5.1-4 install: `node-gyp rebuild`
5461 verbose stack Exit status 1
5461 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
5461 verbose stack     at EventEmitter.emit (events.js:315:20)
5461 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
5461 verbose stack     at ChildProcess.emit (events.js:315:20)
5461 verbose stack     at maybeClose (internal/child_process.js:1048:16)
5461 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
5462 verbose pkgid @abandonware/bleno@0.5.1-4
5463 verbose cwd C:\github\WouterJD\FortiusANT\node
5464 verbose Windows_NT 6.3.9600
5465 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "--unsafe-perm" "--verbose"
5466 verbose node v14.16.1
5467 verbose npm  v6.14.12
5468 error code ELIFECYCLE
5469 error errno 1
5470 error @abandonware/bleno@0.5.1-4 install: `node-gyp rebuild`
5470 error Exit status 1
5471 error Failed at the @abandonware/bleno@0.5.1-4 install script.
5471 error This is probably not a problem with npm. There is likely additional logging output above.
5472 verbose exit [ 1, true ]

WouterJD, 2021-04-29:
If you give below commands, what do you get

```
C:\>node -v
v14.15.2

C:\>npm -v
6.14.9

```
This is on Windows 10 desktop.

TheBlacksmith86, 2021-05-02:
I get following info:

C:\>node -v
v14.16.1

C:\>npm -v
6.14.12

I'm running it on Windows 8.2

WouterJD, 2021-05-17:
I remember I could not install on Windows 7 and moved to Windows 10.
Windows 10 is validated and working, Windows 8 I do not know

closed 2021-06-07 by WouterJD