WouterJD/FortiusANT #304: Resistance drops

hasise, 2021-04-29:
Hi

I have an issue with resistance drops, normally everything works just fine in the start but after a while the resistance goes
down and the watts in Zwift goes up. This can go on for a while until there is a hill then the resistance is back. So the brake
is not totally dead. Unfortunately I don't have a way of checking the power in any other way.
I don't know if this is the trainer or something else so I would like to find out if there is a communication problem.
What is the best way to debug this?

T1932 Fortius connected to Flow Multiplayer T1901

WouterJD, 2021-04-29:
Hi @hasise If you run FortiusAnt with -d32 (or -d255) a json-file is created.
This file contains "all" data and you could investigate the power-drop.

In supportfiles you find "FortiusANT JSON Analysis.xlsx" which you can use to analyse.
Or drop teh json-file as attachment and I will look into it

hasise, 2021-05-01:
Just realized that the log files etc are deleted when starting. Since FortiusANT starts automatically my log files are gone after reboot today.
Commented out the rm of log files temporary and will give it a go.

hasise, 2021-05-03:
Could not import json data, got an error about an additional character.
[FortiusAnt.zip](https://github.com/WouterJD/FortiusANT/files/6413029/FortiusAnt.zip)

hasise, 2021-05-05:
Todays ride started with rather low resistance, around 16 minutes into the ride the resistance came back at least
for some time. The resistance changes up and down but it is not always as high as it should be.

[logFiles.zip](https://github.com/WouterJD/FortiusANT/files/6429091/logFiles.zip)

[Ride.zip](https://github.com/WouterJD/FortiusANT/files/6429136/Ride.zip)

tkathulevej, 2021-05-16:
Hi,

I think I've seen the same problem. - Or at least something close to it.

I have 4 Tacx Fortius trainers (...they come cheap by now ;-). In the beginning when using Fortius ANT I experienced some strange behaviour, like resistance suddenly steeping up and down without notice. This could happen at any time during sessions but troubles normally only stayed on for short period of time. Unfortunately these problems were periodic, and it was very difficult to reproduce same behaviour from trial to trial. And op top of all, after a while these issues kind of 'went away by themselves'. - And I was of course quite confused about where start looking for errors.

However one problem that has stayed pesistant is the trainer suddenly loosing virtually all resistance, and at the same time it seems to lock itself reporting back a very high wattage, which stays fxed no matter pedaling or not. In my case this type of problem never starts earlier than 15-30 minutes into a session, and I've only been able so 'resolve' from such a situation by fully shutting down my session (in BigRing VR which I use) and restart Fortius ANT completely. However it's seems that the problemonly arises, when I'm pedaling fairly hard, i.e. in order of 250w plus, which I'm only barely capable of.

But then recently and interesting thing happened: I sold one of the trainers to a friend, who used the trainer with the original Tacx 4.3 software, and after short while he returned back telling that in every session he rode the program locked up with no restance and high wattage reported after around 20 minutes... :-( :-(

Well although I would say from all my experience, that this problem is somehow related to the trainer itself, it's also very unlikely, that I would have 4 trainers broken in the same way. - So now I'm VERY confused. And any good suggestions regards this are very, very welcome.... 

I myself will now start doing a systematic 'changing-parts-one-by-one' investigation of the problem. Unfortunately I'm now more convinced that this is somehow related which pedaling at fairly high wattages, since my friend reported to have been runing sessions at constant power around 280w...

hasise, 2021-05-18:
@WouterJD Any help analyzing the logs is apricated. 
Regards, Hans Simonsen

WouterJD, 2021-05-18:
Sorry for not reacting lately. Pretty busy with outside work.
I will verify the logfiles shortly

hasise, 2021-05-18:
"Busy outside work"? having a life? 😂

WouterJD, 2021-05-18:
OK guys, sorry again for keeping you waiting.

FortiusAnt logs "all" activities on the interfaces, resulting in the *.log files. These are not very readable.
On special request, *.tcx files are created for those who use FortiusAnt as hometrainer and still want to upload results in strava.

The *.json file is for analysis, because at the log-moment "all" internal data is written and graphs can be made.,
The json file can be uploaded into excel [FortiusAnt.2021-05-05 14-46-11.xlsx](https://github.com/WouterJD/FortiusANT/files/6503031/FortiusAnt.2021-05-05.14-46-11.xlsx) for analysis.

[image]
This graph shows the TargetGrade (as received from CTP) and the calculated TargetPower and TargetResistance.

WouterJD, 2021-05-18:
[image]
The second graph shows the actual cadence, speed and power.

WouterJD, 2021-05-18:
You have a T1901 brake which does not have a lineair behaviour; resistance can be set in distinct steps only: 1900, 2030, 2150, 2300, 2400, 2550, 2700, 2900, 3070, 3200, 3350, 3460, 3600, 3750 (regardless what these figures mean).

Where a motorbrake can follow positive and negative slopes, this is harder for the magnetic brake you have.

WouterJD, 2021-05-18:
CTP's operate in two modes: ERGmode (target power) and resistance mode (target slope).
ERGmode is OK for the magnetic brake (no negative value will ever be requested).
Resistance mode might be more difficult.

Using the -G parameter influences the behaviour
-G50 causes FortiusAnt to take 50% of the slope, causing a more moderate TargetPower
-G50/50 idem + reduce negative slopes with 50% again

If you specify -G50/0 then negative slopes will be reduced to 0%.

WouterJD, 2021-05-18:
Recommendations:
- read manual section 2.6
- experiment with -G parameter
- keep wheelspeed high (do not use bike gears) and use the digital gearbox to reduce target resistance
- Oh yeah; download newest version, which is a bit easier for me to know that we're using same code

Please let me know whether this helps.

WouterJD, 2021-05-18:
> "Busy outside work"? having a life? 😂

https://www.facebook.com/WJ.Dubbeldam

closed 2021-06-07 by WouterJD