WouterJD/FortiusANT #307: Ceinture Cardio "Coospo"

regis65, 2021-05-09:
Bonjour,
Depuis que j ai mis la nouvelle version de  
Fortisant + (6.1),la ceinture cardio Coospo  n apparer pas sur la page j ai juste les vitesses avant et l arrière mais pas mes pulsations cardiaque.
Comment réparer cela.
Sportivement 
Regis

BikeBeppe64, 2021-05-09:
Maybe this help, Raspberry PI 4 

[image]

 Bluetooth

regis65, 2021-05-09:
J ai oublié de dire que mon Fortisant + tourne sous windows 10.

WouterJD, 2021-05-17:
Regis, could you provide some more information?

regis65, 2021-05-17:
C est une ceinture Cardio coospo et je n arrive pas a le faire apparaître sur la page de lancement,même la page me dit de mauvais Watts

WouterJD, 2021-05-18:
Regis c'est le vrai minimum d'informations pour comprendre ce que tu veuz faire.
S'il vous plait explique mieux pur comprendre.

"la ceinture cardio Coospo n apparer pas sur la page": quel page de quel application sur quel systeme?
:-)

closed 2021-06-07 by WouterJD