WouterJD/FortiusANT #308: connecting BT HR monitor to the application

Hasoth, 2021-05-12:
Hi,

is it possible to connect BR HR moniotor to the application? I have old but quite expensive and still working fine HR strap, but ZWIFT is not recognizing that.... that would be a shame to spend money to buy a new one (laptop and android phones are working fine... endomondo too)

MeanHat, 2021-05-13:
Hi @Hasoth, yes, it is possible to connect a Bluetooth HR monitor to a device running FortiusANT if it is set up to use BLE rather than ANT. See the manual in the WIKI (https://github.com/WouterJD/FortiusANT/blob/master/doc/FortiusANTUserManual.pdf) for more information

Hasoth, 2021-05-13:
@MeanHat do you maybe know if I need to specify the unit? Standard windows .exe compiled file uses those parameters (looks like it should pick the first possible BT unit):

[image]

My device is Polar iWL, do I understand I should use -HBTHENUM at the end? (should I download python and compile my own version just for that parameter?) 

[image]

MeanHat, 2021-05-13:
HI @Hasoth, -H0 should work fine. The other alternative that I use is to connect the BLE HRM directly to the Cycle Training Program (in my case Zwift).

Hasoth, 2021-05-13:
-H0 does not work somehow... ZWIFT does not recognize the my BLE HRM either. I do not understand why because Endomondo (now MapMyRun) recognizes that, I've used by mand for running in the past. (device is visible on win10 and on android)

@WouterJD are you able to tell how the app recognize that the device is HRM or there's any BLE HRM available at all?

WouterJD, 2021-05-17:
FortiusAnt has three options:
- use the HR signal from the Tacx when you have paired the HRM to the Tacx. FortiusAnt then publishes the ANT+HRM device to be received by Zwift  (default operation, -H not defined);
- pair with HRM and display the result; no HRM is transmitted (-H0 or -H<id>)
- no HRM at all (-H-1)

FortiusAnt does not pair with a BLE-heartrate monitor

Hasoth, 2021-05-17:
@WouterJD Thanks a lot for the reply and clarification!

I'll check if there's any option to pair BLE-HRM with my old tacx (fortius T1940 with full blue T1942 head unit), but since the software for it doesn't work properly on win10 and I guess the head unit is not able to do the connection on it's own, I'll just use endomondo on the other phone to control HR when needed or by ANT+ HR band.

WouterJD, 2021-05-18:
OK succes. I assume issue can be closed, otherwise just let me know.

closed 2021-05-18 by WouterJD